SpinalHDL/SpinalHDL #6: Describe testbench from within Spinal

pmudry, 2016-04-06:
This would be a very nice feature. One would be able to describe time variation with a relationship to a signal and this would automatically generate a TB for simulation. For instance:

```
val a := some multibit signal as input 
val b := some other signal related to a through a component

tb{
    a := a + 1 every 10 uS
    assert ( something about b that changes according to a)
} with clock @ 5 MHz
```

Just an idea you know

Dolu1990, 2016-04-06:
Actualy, only a vhdl testbench without stimulus is automaticly generated. I don't have explore this way so much to keep focus on RTL, but it's true that it could be very usefull to have a way to simply test the generated hardware, which is realy borring in vhdl.

Probably solution like cocotb are very good one (Drive DUT from software world)
Having something like it in Scala could be greate, and then open the door to implement DSL (as your proposal)

Dolu1990, 2018-01-03:
♪ lolilol ♫
http://spinalhdl.github.io/SpinalDoc/spinal/sim/example/single_clock_fifo/

closed 2018-01-03 by Dolu1990