SpinalHDL/SpinalHDL #8: Cross platform toplevel pin mapping definition

Dolu1990, 2016-05-09:
For exemple, Xilinx and Altera have their own file format to specify mapping between design pins and package pins (+ specify pin technologies, current, terminaison, ...)

It could be great to have a scala DSL to specify them without having to worry about the specific syntax of the target EDA.

pmudry, 2016-09-23:
I agree, would be a great feature ! :+1:

numero-744, 2022-05-30:
Do you have a plan for this? A design idea? Do you want to add this in the `io` bundle or externally? I think the second would be clearer/cleaner and easier to implement.

What about a class to express constraints in general?

Here is a draft:

```scala
val constraints = new Constraints(new myTop) {
    // The clocking constraint would be inferred from the clock domain if it specifies a frequency

    // Default voltage first
    voltage(myTop.io).set50
    // Specific voltage, erasing the previous one just for one pin
    voltage(myTop.io.myPin).set33
}
```

Dolu1990, 2022-05-31:
I'm thinking more about adding the information to the io signals via the SpinalTag utilities.
Then we could analyse all the io of a given toplevel to generate the proper constraint file.

A bit like the qsysify tool does.

Readon, 2022-06-09:
> I'm thinking more about adding the information to the io signals via the SpinalTag utilities. Then we could analyse all the io of a given toplevel to generate the proper constraint file.
> 
> A bit like the qsysify tool does.

How about IP-XACT support mentioned #621?

Dolu1990, 2022-06-10:
Hmm, IP-XACT would be a good thing in itself, but it isn't related right ? i mean, this issue is more about the pin mapping of the FPGA i think.

Dolu1990, 2022-06-10:
While IP-XACT is more about interface definition ?

Readon, 2022-06-10:
yes， in my opinion, if the names can be mapped to unified standards by IP-XACT, then it's not necessary to map again through other mechanism.