SpinalHDL/SpinalHDL #14: Universal reset

RandomReaper, 2016-06-01:
Reset generation for synchronous blocks should be configurable since support in target architecture will vary.

I think this 3 cases (and combinations) should be supported:
* asynchronous reset (like it is at this time)
```vhdl
...
signal test : std_ulogic;
...
process(reset, clock)
    if reset = '1' then
        test <= '0';
    elsif rising_edge(clock) then
        test <= not test;
    end if;
end process;
```
* synchronous reset
```vhdl
...
signal test : std_ulogic;
...
process(clock)
    if rising_edge(clock) then
        test <= not test;
        if reset = '1' then
            test <= '0';
        end if;
    end if;
end process;
```

* no reset (can save a lot of routing resource in FPGA).
```vhdl
...
signal test : std_ulogic := '0';
...
process(clock)
    if rising_edge(clock) then
        test <= not test;
    end if;
end process;
```

Dolu1990, 2016-06-01:
Currently syncronous and asyncronous reset are already supported. But it's not the case for the "no reset one" (value loaded by the FPGA bitstream, right ?).

I agree, it should be added. I will think about it and how doing that well.

RandomReaper, 2016-06-02:
On 01.06.2016 16:34, Dolu1990 wrote:
> Currently syncronous and asyncronous reset are already supported. But it's not the case for the "no reset one" (value loaded by the FPGA bitstream, right ?).

Did you mean you've got a way for the synchronous reset (without writing more spinal code)?

Sure, the "no-reset" case is for RAM based FPGA, I have only tested on small designs,
but it seems it can save 5-15% of the used resources.

> I agree, it should be added. I will think about it and how doing that well.
:+1:

Dolu1990, 2016-06-02:
> Did you mean you've got a way for the synchronous reset (without writing more spinal code)?
Yes, when you define a new ClockDomain, you can define some additional configuration :
http://spinalhdl.github.io/SpinalDoc/spinal/core/clock_domain/#configuration

About the "no-reset" Yes i agree, i also add similar things (area reduction) and also better FMax.
I'm working on that on the rework_SyncNode branch, that should not be difficult.

Dolu1990, 2016-06-02:
It's nearly done (FPGA bitstream reset), juste have some issue with one particular testbench, it should be done tomorrow.

Dolu1990, 2016-06-02:
Done !

closed 2016-06-02 by Dolu1990