SpinalHDL/SpinalHDL #16: Unconstrained BitVector (feature request)

RandomReaper, 2016-06-02:
In ```VHDL``` we can use unconstrained arrays, see for instance my [synchronous fifo](https://github.com/RandomReaper/pim-vhdl/blob/master/hdl/rtl/fifo/fifo.vhd).

The array size are determined by the connected signals.

Is there a way to create an unconstrained BitVector?
Here is one of your examples:
```scala
class Counter(width : Int) extend Component{
  val io = new Bundle{
    val clear = in Bool
    val value = out UInt(width bit)
  }
  val register = Reg(UInt(width bit)) init(0)
  register := register + 1
  when(io.clear){
    register := 0
  }
  io.value := register
}
```
As a lazy programmer I would like to write:
```scala
class Counter extend Component{
  val io = new Bundle{
    val clear = in Bool
    val value = out UInt
  }
  val register = Reg(UInt) init(0)
  register := register + 1
  when(io.clear){
    register := 0
  }
  io.value := register
}
```

Dolu1990, 2016-06-02:
There already some unconstrained bit vector support. Maybe not exactly the same than VHDL one. There is an example :

```scala
object PlayUnconstrained {
  class Counter extends Component{
    val io = new Bundle{
      val load = slave  Flow(UInt)
      val value = out UInt
    }
    val register = Reg(UInt()) init(0)
    register := register + 1
    when(io.load.valid){
      register := io.load.payload
    }
    io.value := register
  }

  class TopLevel extends Component {
    val io = new Bundle{
      val load = slave Flow(UInt(8 bits))
      val value = out UInt
    }

    val counter = new Counter
    counter.io.load <> io.load
    counter.io.value <> io.value
  }

  def main(args: Array[String]): Unit = {
    SpinalVhdl(new TopLevel)
  }
}
```

From a graph perspective, nodes width inferation look for inputs to get a width. Currently you can't infer the width of something by looking consumers. 
Is that possible in VHDL ?

RandomReaper, 2016-06-02:
I don't really get your question, but here is how I'm used to write ```VHDL```:
```vhdl
library ieee;
	use ieee.std_logic_1164.all;
	use ieee.numeric_std.all;

entity sub is
	port
	(
		a	: in	std_ulogic_vector;
		b	: out	std_ulogic_vector
	);
end sub;

architecture rtl of sub is
begin
	b <= a;
end rtl;

library ieee;
	use ieee.std_logic_1164.all;
	use ieee.numeric_std.all;

entity top is
end top;

architecture rtl of top is
	signal foo : std_ulogic_vector(7 downto 0) := x"55";
	signal bar : std_ulogic_vector(foo'range);
begin
i_sub: entity work.sub
port map
(
	a => foo,
	b => bar
);
end rtl;
```

RandomReaper, 2016-06-02:
@Dolu1990,

I even don't know what I tested earlier, but spinal does what I want:
```scala
package MyCode

import spinal.core._
import spinal.lib._

class Sub extends Component {
  val io = new Bundle {
    val a = in UInt
    val b = out UInt
  }
  io.b := io.a
}

class Top extends Component {
  val io = new Bundle {
    val a = in UInt(7 bit)
    val b = out UInt(7 bit)
  }
  val o = new Sub
  o.io.a <> io.a
  o.io.b <> io.b
  
}

object Sub {
  def main(args: Array[String]) {
    SpinalVhdl(new Top)
  }
}
```

Thank you for your support!

Dolu1990, 2016-06-02:
Greate : D
About the question "Is that possible in VHDL ?" 
To reformulate, is it possible in VHDL to infer width by using the one of a consumer (output) ? (not an input)
Like in your example :

```scala
class Counter extend Component{
  val io = new Bundle{
    val clear = in Bool
    val value = out UInt
  }
  val register = Reg(UInt) init(0)
  register := register + 1
  when(io.clear){
    register := 0
  }
  io.value := register
}
```

RandomReaper, 2016-06-02:
IMHO ```VHDL``` does that.

Changing only that in the preceding example:

```diff
 architecture rtl of top is
 	signal foo : std_ulogic_vector(7 downto 0) := x"55";
-	signal bar : std_ulogic_vector(foo'range);
+	signal bar : std_ulogic_vector(8 downto 0);
 begin
 i_sub: entity work.sub
 port map
```

Will result in an error at the ```b <= a;``` instruction. So we can guess that the VHDL can constrain the size from it's outputs...

RandomReaper, 2016-06-02:
In fact I even use that in my [width_changer](https://github.com/RandomReaper/pim-vhdl/blob/master/hdl/rtl/width_changer/width_changer.vhd), a minimal depth fifo which can be make a signal wider or narrower.

I get the sizes from the connected signals and then choose which architecture is required.

RandomReaper, 2016-06-02:
I remember what went wrong, this one do not generate the code (fails with : ```Can't infer width on...```:
```scala
package MyCode

import spinal.core._
import spinal.lib._

class Sub extends Component {
  val io = new Bundle {
    val a = in UInt
    val b = out UInt
  }
  io.b := io.a
}

object Sub {
  def main(args: Array[String]) {
    SpinalVhdl(new Sub)
  }
}
```

Dolu1990, 2016-06-02:
While the scala front end (Spinal) support unconstrained bit vector, some potential backend will not support it (Verilog). It's why it's solved by Spinal completly. Also by solving it in Spinal, many additional checks could be done more deeply, like assignement checking width,  combinatorial loops checking. Many things was implemented on this assumption.

I will probably come friday to talk about that.

Dolu1990, 2016-06-02:
Hoo about the width width_changer, there is something similar in spinal.lib, You can use any Stream[Fragment[Bits]] to get any Stream[T] :

```scala
object PlayWidthChanger {
  case class RGB() extends Bundle{
    val r,g,b = UInt(8 bits)
  }
  
  class TopLevel extends Component {
    val cmd = slave (Stream Fragment Bits(8 bits))
    val rsp = master(cmd.toStreamOf(RGB()))
  }

  def main(args: Array[String]): Unit = {
    SpinalVhdl(new TopLevel)
  }
}
```

The spinal.lib is probably the next things that need to be extended with many utilies.

RandomReaper, 2016-06-02:
I think we should resolve when using with verilog, and not when generating vhdl.
It is still possible to make (fake) constraints for the spinal additional checks.

Dolu1990, 2018-01-03:
As the SpinalHDL generated hardware is a netlist (no loop, no 'lenght, no generate statements, no generics), it can't keep unconstrained bitvectors (which are kind of generics) into the generated RTL.  All the hardware elaboration is done in Scala.
As soon your design need a for loop to iterate over a vector width or something related, this paradigm will become incompatible with this requested feature.

closed 2018-01-03 by Dolu1990