SpinalHDL/SpinalHDL #24: BSD

sirinath, 2016-06-27:
Is it possible to have a more permissive license like BSD, MIT, ASL 2.0, BSD would be ideal as https://chisel.eecs.berkeley.edu/ uses it.

Dolu1990, 2016-06-27:
Hi,

In which practical cases the current license is constraining ?
Currently the LGPL applyed to Spinal HDL could be resume in following staments :

asb, 2016-07-16:
The interaction of the (L)GPL with hardware designs is not fully understood. In particular, the concept of static and dynamic linking doesn't really transfer. If you would like the Spinal core to remain copyleft, perhaps licensing spinal.lib under BSD/MIT/Apache would make things clearer to the end user?

Dolu1990, 2016-07-16:
@asb Hmm its probably right. Some other people already sent me emails about this confusing point.

To be clear, the goal was not to make money with licensing. I personnaly don't want to stop people with all these borring license stuff.

Juste give me one week to think a little bit more about that.

All idea are welcome :D

sirinath, 2016-07-16:
Chances are not body will maintain an internal fork of the library hence it might be best to have a permissive license throughout. This will encourage usage and also contribute more.

svancau, 2016-07-20:
I would propose that the core could remain in LGPL as a library (in the SW sense), and the component library could be in a permissive licence (MIT/BSD/whatever).

In that sense, the execution of a Spinal generator would call the core (SW only, LGPL behavior understood). And there is no restriction on the emitted code whatsoever, it can be in the licence the user chooses to use for the VHDL/Verilog.

Dolu1990, 2016-07-20:
Yes that make sense, i'm currently changing the license of the spinal.lib.

Dolu1990, 2016-07-20:
spinal.lib is now under MIT => 0d6d36821d44cf6b598906a925790e490de7c952

closed 2016-07-22 by Dolu1990