SpinalHDL/SpinalHDL #28: Simulation hangs with Verilog backend

svancau, 2016-07-24:
Icarus Verilog with Cocotb hangs when the signal sckPrescaler goes to 0. 
Problem happens around line 30 of the Scala file in the prescaleArea.
[Code](https://github.com/svancau/spinal_play/blob/046775ae94c236d70a18469141d8d0dce259cd37/SPIMaster.scala)

Dolu1990, 2016-07-24:
I'm investigating now.
    dut.io_configIf_CPHA <= 1 if (mode & 1) else 0
    dut.io_configIf_CPOL <= 1 if (mode & 2) else 0
should be
    dut.io_configIf_CPHA <= (1 if (mode & 1) else 0)
    dut.io_configIf_CPOL <= (1 if (mode & 2) else 0)
But that not the main issue, I continue investigation

Dolu1990, 2016-07-25:
I realy think this is a IcarusVerilog 10.0 issue (I already had other kind of issues with 9.7)

So, iverilog is buggy because the following code work : 
```verilog
  always @ (zz_1 or zz_3)
  begin
    prescaleTick <= !zz_1 && zz_3;
  end
```

while the following one doesn't ...
```verilog
  always @ (zz_1 or zz_3)
  begin
    prescaleTick <= 0;
    if(!zz_1 && zz_3)begin
      prescaleTick <= 1;
    end
  end
```

(zz_1 and zz_3 are boolean wires)

assign prescaleTick = !zz_1 && zz_3;

closed 2016-07-25 by Dolu1990