SpinalHDL/SpinalHDL #32: Problem creating Pinsec

roman3017, 2016-09-05:
I have a problem creating Pinsec:

`$ cat MS6P.scala`
`import spinal.core._`
`import spinal.lib._`
`import spinal.lib.soc.pinsec._`
`object Pinsec{`
`  def main(args: Array[String]) {`
`    SpinalVerilog(new Pinsec(100 MHz))`
`  }`
`}`

`$ cat build.sbt`
`name := "SpinalBaseProject"`
`version := "1.0"`
`scalaVersion := "2.11.6"`
`EclipseKeys.withSource := true`
`libraryDependencies ++= Seq(`
`  "com.github.spinalhdl" % "spinalhdl-core_2.11" % "latest.release",`
`  "com.github.spinalhdl" % "spinalhdl-lib_2.11" % "latest.release"`
`)`

`$ scala -version`
`Scala code runner version 2.11.6 -- Copyright 2002-2013, LAMP/EPFL`

`$ sbt run`
`[info] Loading project definition from /.../project`
`[info] Set current project to SpinalBaseProject (in build file:/.../projects/SpinalBaseProject/)`
`[info] Compiling 2 Scala sources to /.../projects/SpinalBaseProject/target/scala-2.11/classes...`
`[error] /.../projects/SpinalBaseProject/MS6P.scala:6: too many arguments for constructor Pinsec: ()spinal.lib.soc.pinsec.Pinsec`
`[error]     SpinalVerilog(new Pinsec(100 MHz))`
`[error]                   ^`
`[error] one error found`
`[error] (compile:compileIncremental) Compilation failed`
`[error] Total time: 2 s, completed 4-Sep-2016 6:00:14 PM`

When I remove argument 100 MHz from Pinsec constructor I get more less expected runtime error:

`...`
`[error] (run-main-0) spinal.core.SpinalExit: `
`[error] `
`[error] `
`[error] You are trying to get the frequency of a ClockDomain that dosen't know it`
`...`

Can you please advice what I am doing wrong?

Dolu1990, 2016-09-05:
Hooo,
You are working on the 0.10.0 release, but the online documentation is based on the SpinalHDL master branche. Since the last release, Some stuffhave change in Pinsec. Thats the problem.
I will now release the 0.10.1  Then you code will work :)

Sorry for this desynchronisation.

Dolu1990, 2016-09-05:
Hi,
I have released the update under the 0.10.2 version.

You can update your build.sbt by changing "last.release" with"0.10.2"  as following :

```scala
libraryDependencies ++= Seq(
  "com.github.spinalhdl" % "spinalhdl-core_2.11" % "0.10.2",
  "com.github.spinalhdl" % "spinalhdl-lib_2.11" % "0.10.2"
)
```

Then following commands should work (tested on my side):

```sbt
sbt
reload
compile
run
```

Let's me know how things are goind : D

roman3017, 2016-09-05:
It works, thank you very much.

closed 2016-09-05 by roman3017