SpinalHDL/SpinalHDL #62: How to extend blackbox attribute support uint32_t parameter?

linuxbest, 2017-04-09:
I try to  implement a LUT5 as a blackbox with spinalhdl, but is not expect.

Thanks.

 case class LUT5(INIT : Int) extends BlackBox {
   val generic = new Generic {
     val INIT = LUT5.this.INIT
   }
   val io = new Bundle {
     val I0 = in Bool
     val I1 = in Bool
     val I2 = in Bool
     val I3 = in Bool
     val I4 = in Bool
     val O = out Bool
   }.setName("")
 }

.......
val lut = LUT5(0x96696996) // xor

generated code:

LUT5 #(
     .INIT(-1771476586)
 ) lUT5_1 (
     .I0(zz_26),
     .I1(zz_27),
     .I2(zz_28),
     .I3(zz_29),
     .I4(zz_30),
     .O(zz_14370)
   );

Dolu1990, 2017-04-09:
Hi,

So using

Dolu1990, 2017-04-09:
Hi,

So 0x96696996 -> -1771476586 is kind of correct, because in verilog (and also in scala), an Int are only 32 bits signed value.

Then if you want something unsigned you shoud use Hardware UInt or Hardware Bits :

```scala
  case class LUT5(INIT : Bits) extends BlackBox {
    val generic = new Generic {
      val INIT = LUT5.this.INIT
    }
    val io = new Bundle {
      val I0 = in Bool
      val I1 = in Bool
      val I2 = in Bool
      val I3 = in Bool
      val I4 = in Bool
      val O = out Bool
    }

    noIoPrefix()
  }

    val lut = LUT5(B"x96696996")
```

But by trying it, i just discover that it's brocken in the current release. I have a fix, i'm releasing it in the next 30 mn.

Dolu1990, 2017-04-09:
It's released with the 0.10.12, you can get it by changing the "last.release" into  0.10.12  in the build.sbt.

closed 2017-04-09 by Dolu1990