SpinalHDL/SpinalHDL #68: Documentation Feature Request: Documentation for QSysify...

madscientist42, 2017-07-14:
While I realize it's not a full priority, this tool could be very powerful overall- with a comparable thing for each vendor's or Open Source project's macroblock assembly tool.

Not all potential prospective users can embrace SpinalHDL directly.  (I know, they should, but...)

Some are stuck with using Verilog or VHDL because of management edict, etc. while they CAN use it as a subsystem/macro building tool to generate valid RTL of the same two there.  (My current use...)

For some, they'll just absorb the Verilog or VHDL.  Some would like to be able to crank out a QSys capable set of files for what function macros they design with SpinalHDL.

I've done some experimentation with the VexRiscv CPU core and QSysify.  It's not obvious whether the issues I encountered are functionality needing to be added or told what to do.  Part of this is due to a lack of clear instructions on some of it's use.

Is it possible to tell QSysify to use something other than a Conduit for an interface?

- Some things are legitimately Conduits.
- Some things are actually a valid bus under Avalon Interconnect rules now.  (AXI4 counts as a valid bus)

The problem right now is that if you QSysify the VexRiscv as a full build-out, it Conduits everything other than the clocks and the resets.  Less than useful- and not clear if it's an omission on the user's part or a design thing needing to be worked out as a real feature enhancement.  You can specify axi4, ahb, amb as bus types in the latest versions of QSys for the TCL-spec files and it will understand and use it properly as to whichever it was you specified.

Clear basic documentation on driving this functionality would help a lot on this and let people help you grow it into even more powerful/capable abilities.

Dolu1990, 2017-07-15:
So currently the QSysify tool isn't able to detect AXI4 buses, the list of things supported is here :
https://github.com/SpinalHDL/SpinalHDL/blob/764193013f84cfe4f82d7d1f1739c4561ef65860/lib/src/main/scala/spinal/lib/eda/altera/QSys.scala#L13

Basicaly, the QSysify tool will try each of those "translator", and if none are matching, the conduit one will be used (last one in the list)

It's very easy to add support for new things.
Also some buses need additional tags. For instance to generate an interrupt sink, you have to add a InterruptReceiverTag to the corresponding pin, then the QSysify tool, in place to take it as a conduit will generate an QSys interrupt sink.

example : https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/VexRiscv/demo/VexRiscvAvalon.scala#L161

There is some more talk about the VexRiscvAvalon which is QSysified there : https://github.com/SpinalHDL/SpinalHDL/issues/67#issuecomment-315519498

Dolu1990, 2017-07-15:
Just, don't be afraid by the VexRiscvAvalon stuff, this is realy advanced hardware metaprogramming stuff, SpinalHDL could be used in much more regular ways.

The VexRiscvAvalon example is how you can rework theVexRiscv component afterward by using advanced feature of Scala/SpinalHDL.
Also it require the very last git head of SpinalHDL. (git clone ... / sbt publish-local)

An example of generate file are the following : 

[VexRiscvAvalon.zip](https://github.com/SpinalHDL/SpinalHDL/files/1150270/VexRiscvAvalon.zip)

madscientist42, 2017-07-19:
Not even close to afraid.  In the process of starting a board bring-up with it.  I'm keen on high-end metaprogramming/metadesign tools as well as practical HLS tools like CLaSH.  While I can do the lower-level HLS stuff...it goes slow for me because it's not what I've done for the last 30 years.  I'd rather reach for a much higher-level, programming-like construct or literally program it with Haskell or Scala. 

Only odd thing I'm seeing right now is that on the Cyclone IV (This is what I have in hand, not what we'll be implementing the project ultimately on...) I'm only seeing an FMax of 70 MHz on the SOC I've implemented against the QSys implementation I'm working with. 

I'm going to build out a Briey to this FPGA to get a better feel for what you've done with this and see if FMax being "off" is something to do with the Avalon interface as currently implemented or QSys.  I'm semi stuck on several fronts with using it or something I can use to glue things together with in a similar manner.  It must be in VHDL if at all possible (to simplify ASIC implementation down the line), no HLS if at all possible, and must use or Altera's tools or be usable in a manner not unlike QSys, etc. is used.

If nothing else, I'll look at what you did there and learn a little- because if you can re-work stuff like that...it's got even more potential than I thought...just have to figure out how to move it there for the client.  :D

madscientist42, 2017-07-19:
As for QSysify...I looked at the code there for doing an AXI4 translator piece.  I need to brush up on my Scala and look at it closer.  It looks straightforward enough, but I can't seem to wrap my head around where to find the similar hooks that the Avalon and APB emitters use to generate the chunks accordingly.

Part of that is my need to get this going cleanly as quickly as I can for the client.  Part of it is that my Scala's rather rusty...  X-D

Dolu1990, 2017-07-19:
You can find some FPGA project which instantiate the Briey SoC there (DE0-Nano is Cyclone IV): https://drive.google.com/drive/folders/0B-CqLXDTaMbKZGdJZlZ5THAxRTQ?usp=sharing

Be carefull, if you change the frequancy, you need to regenerate the Briey SoC with corresponding  setup, as the SDRAM controller use the clock frquancy to apply some timings

I have done some test to know the FMax of the Briey SoC (default synthesis setings, best grade FPGA) :
  Artix 7        -> 230 Mhz 3551 LUT 3612 FF 
  Cyclone V  -> 126 Mhz 2,608 ALMs
  Cyclone IV -> 117 Mhz 5,196 LUT 3,784 FF 
  Cyclone II  -> 102 Mhz 5,321 LUT 3,787 FF 

Maybe the avalon version of the VexRiscv need some pipelining  buffer before behing connected to the interconnect (i think in particular to the DBus read response) to get a good FMax

Let's me know how things are going ^^

madscientist42, 2017-07-19:
> CqLXDTaMbKZGdJZlZ5THAxRTQ?usp=sharing

Will do.  I'll keep the change in mind.  Thanks for the advice.  It's
appreciated.

Dolu1990, 2017-08-08:
http://spinalhdl.github.io/SpinalDoc/spinal/lib/eda/qsysify/

closed 2017-08-08 by Dolu1990