SpinalHDL/SpinalHDL #69: VHDL generation does not appear to produce vaild results with Attributes added.

madscientist42, 2017-07-17:
From a generator pass of Briey:

> architecture arch of VexRiscv is
> attribute verilator public : boolean;

Here's the complaint from Quartus:

>Error (10500): VHDL syntax error at bt_soc2.vhd(5525) near text "public";  expecting "of"

Another oops from generated VHDL:

> signal writeBack_RegFilePlugin_regFileWrite_valid : std_logic;
> attribute verilator public of writeBack_RegFilePlugin_regFileWrite_valid: signal is true;

...and the complaint from Quartus:

> Error (10500): VHDL syntax error at bt_soc2.vhd(5685) near text "public";  expecting "of"

Looks like something's injecting "public" in the mix there and at least Quartus doesn't like it.

Dolu1990, 2017-07-17:
Hooo funny, 
It is not a bug, but just the fact that the "verilator public" attribut used for sim is only for Verilog. I will findout a way to not generate them when the backend is the VHDL one.

madscientist42, 2017-07-17:
Got it.  Misfeature.  X-D

I'll limp along for now in Verilog because at this stage I can get away
with that with this project- I'll just need the "right" thing with VHDL
down the line.  Thanks!

Dolu1990, 2017-07-17:
It's now fixed, you just need to pull the last SpinalHDL version and do a sbt publish-local
I tested it on Quartus 15, it successfuly compile things.

closed 2017-07-17 by Dolu1990