SpinalHDL/SpinalHDL #74: Octal SPI support

fathomtreelimited, 2017-08-25:
Hi!

Would you consider supporting octal SPI controllers - with a view to support Macronix OctaFLASH and OctaRAM products.

http://www.macronix.com/en-us/about/news/Pages/Macronix-Unveils-New-Generation-Low-Pin-Count-OctaBus-Memory-Subsystem-Solution-for-IoT-and-Automotive-Electronics-Markets.aspx

Thanks! :-)

Dolu1990, 2017-08-25:
Hi,

I'm currently working (background task) on some peripherals (i2c/spi/sio) so normaly the SIO controller should fit, but i haven't tested it yet, i will do it soon 

https://github.com/SpinalHDL/SpinalHDL/blob/master/lib/src/main/scala/spinal/lib/com/spi/SpiMaster.scala#L64 => WIP

fathomtreelimited, 2017-08-25:
Neat!  Look forward to seeing i2c/spi/sio develop.  Cypress HyperBus is another OctalSPI like bus.
http://www.cypress.com/file/213356/download
With 13-pins and you have 333MB/sec memory and FLASH on a single 8x6 IC.
http://www.cypress.com/file/322936/download

closed 2017-09-05 by fathomtreelimited