SpinalHDL/SpinalHDL #79: B(0) returns a Bit vector of 0 bits

cristianmacario, 2017-09-22:
Hi Dolu, Hi SpinalHDL development team.
Many thanks for the great job you have done. I am currently playing with SpinalHDL and it is amazing!!

I think I found a little issue. B(0) returns a Bits of 0 bits. I think it should return a Bits of 1 bit.
Found in master branch, SHA ID a6b3b9

Thanks and regards,

Cristian

Dolu1990, 2017-09-22:
Hi,
Why should it be 1 bits ?  the value zero is "nothing" "non" "nada", it doesn't need any space to be encoded.  B(0) mean a binary value loaded with the integer value zero.
If you want to exprime a single bit literal value you have multiple choose :
B"0"   which explicitly exprime a binary value  (ex : B"10101010")
Else there True and False to Bool.

Basicaly SpinalHDL integrate a compilation phase which is able to remove all zero wire signals. So having a zero bits value isn't realy a problem.

Dolu1990, 2017-09-22:
Ahhh yeah i think this is it,
B(0) isn't the same than B"0", as the first is to give an Bits literal loaded with an integer, while the second one is to get a Bits literal loaded with an explicit binary value

Seem ok ?

cristianmacario, 2017-09-27:
Hi Dolu,
sorry for the delayed response.
"nothing" "non" "nada" doesn't need any space to be encoded -> agreed
Zero is none of them, it is a number which represents "nothing" as a quantity.
Zero, IMHO, has the same dignity and rights of other numbers and carries the "zero" information, that's why I would expect B(0) to return a 1-bit bit vector; 0-bits do not carry any information.

Anyway, now I think I understand your point: the width of B(n) is the smallest width sufficient to host all the 1 present in the binary representation of n. Since the binary representation of 0 doesn't contain any bit at 1, B(0) returns a bit vector of 0 bits. Is this what you mean?

I think it is a matter of taste. I'm not saying that B(0) returning bit vector of 0 bits is wrong. Personally, I would expect it returning at least 1 bit, since 0 bits can't carry any information.
Seeing B(0) returning a bit vector of 0 bits was looking strange to me, that's the reason why I opened this ticket.

Dolu1990, 2017-09-28:
The thing is that there is many case where via components parameters, you endup with unecessary logic if you aren't doing it that way. For instance a burst counter could be to count from zero to parameterX, and when this parameterX is zero, then you don't need any bit for the counter and the U(parameterX)

Zero width wire support is realy something that can be usefull to avoid defining useless logic.

Then if you want to explicitly define a Bits value of 1 bit set to zero, you can use :
B"0"
B(0, 1 bits)

Does that make sense ?

Dolu1990, 2017-09-28:
**Anyway, now I think I understand your point: the width of B(n) is the smallest width sufficient to host all the 1 present in the binary representation of n. Since the binary representation of 0 doesn't contain any bit at 1, B(0) returns a bit vector of 0 bits. Is this what you mean?**

Yes kind of, it is also the beaviour of the log2up function, which return how many bits i need to encode N states. Very often in VHDL/Verilog, this function is implemented to have a bias and return 1 bits for both 0 and 1 values. 

Imagine you have an array of signals, but this array only contain one element, then how many bit you need to access it => zero bits,    =>    myArray(U(0))  is fine, because currently U(0) return zero bits.

cristianmacario, 2017-09-29:
It makes sense, thanks for the clarification.
Furthermore I agree that in many cases you want to express the width of a bit vector explicitly.
Also, it is good that log2up(0) returns 0.
I don't have any further comment from my side. I think the ticket can be closed if you also don't have any additional comment/observation.
Cheers

closed 2017-09-29 by Dolu1990