SpinalHDL/SpinalHDL #80: Proposal: what about automatic cast from Bits(1 bits) to Bool?

cristianmacario, 2017-09-22:
This would allow stuff like this:
val tmp = Bool
tmp := B(1)

and other stuff.

Moreover I think a bitVector of 1 bit and a bool are fundamentally the same thing (I might overlook some aspects, please correct me if this is not true).

Snoopy87, 2017-09-22:
I think it is not a good idea to allow people writing this kind of thing `tmp := B(1)`
B() is used to create Bits and not Bool. I think it is more clear to write `tmp := True`

cristianmacario, 2017-09-22:
What about having another operator?
As we have B(), S(), U(), having another one for Bool type.
I am picking 'L()' just as an example:
val tmp = Bool
tmp := L(1)

Of course L(2) would give an error.
Digital designers like to use zeros and ones, they are not so used to True and False...

cristianmacario, 2017-09-22:
I think you can ignore my proposal.
If one doesn't like True and False, he can make his own aliases in his own library.
Something like
object Bl0 extends Bool { this := True }
object Bl1 extend Bool { this := False }

and use his own aliases
val tmp = Bl0

Dolu1990, 2017-09-22:
@cristianmacario 
If you want a signals which is initialized with a literal value you can do :

```scala
val a = True
when(something){
   a := False
}

val b = U(42, 8 bits)
when(something){
   b := 66
}
```

cristianmacario, 2017-09-27:
@Dolu1990 
Hi, thanks for your reply. As said, please ignore my proposal, thinking better about it, I realized it doesn't make sense.

For the sake of clarity, I recognize I was not clear in my first post. There are basically 2 things:

-  from the view of a digital designer, a bit vector of 1 bit and a Bool are the same thing (a wire in a circuit), so why not having an automatic cast, so that we can assign a Bool from a 1-bit Bits? Thinking more about it, that's probably a bad idea, better to keep the advantages of a strongly typed language

- digital designers are not used to True and False values, they are used to work with 0s and 1s. If I have a Bool variable I want to assign it with 0 and 1 or something similar. B(0) and B(1) would look nice to me and, introducing the above mentioned automatic cast, would work. That's another bad idea; better to keep B() for bit vectors and keep bit vectors separate from Bool. If somebody wants to use 0- and 1-like notations, he can define aliases for True and False.

Dolu1990, 2017-09-28:
Yeah, the idea is realy to keep the strongly typed language to avoid as much of possible bugs.
Then in SpinalHDL   Bool is realy an True False value, while Bits is an raw data value without much signification.

VHDL with std_logic and boolean is a little bit "broken"
myStdLogic <=   myUInt > 5;     -- This line is VHDL-wrong,

cristianmacario, 2017-09-29:
That's clear, thanks
I don't have any further comment, ticket can be closed from my side.

closed 2017-09-29 by Dolu1990