SpinalHDL/SpinalHDL #81: Synthesis error in FSM with Synplify

oligu, 2017-09-29:
Synthesizing a FSM with Synplify yields an error message as can be seen in the line starting with `@E` in the log file below. The SpinalHDL code used is the one of the Style A example on the [State machine page of the  SpinlDoc](http://spinalhdl.github.io/SpinalDoc/spinal/lib/fsm/). The line causing the `Type binding failed` error is the first when statement in the fsm_stateReg  case:
```VHDL
case fsm_stateReg is
      when pkg_enum.fsm_stateA =>
```
Synthesis Error Log
=============
```bash
#Build: Synplify Pro G-2012.09M-SP1 , Build 013R, Feb 15 2013
#install: C:\eda\Microsemi\Libero_v11.0\Synopsys\synplify_G201209MSP1
#OS: Windows 7 6.1
#Hostname: WE5370

#Implementation: synthesis

$ Start of Compile
#Fri Sep 29 14:19:04 2017

Synopsys VHDL Compiler, version comp201209rcp1, Build 245R, built Feb 20 2013
@N|Running in 64-bit mode
Copyright (C) 1994-2012 Synopsys, Inc. This software the associated documentation are confidential and proprietary to Synopsys, Inc. Your use or disclosure of this software subject to the terms and conditions of a written license agreement between you, or your company, and Synopsys, Inc.

@N: CD720 :"C:\eda\Microsemi\Libero_v11.0\Synopsys\synplify_G201209MSP1\lib\vhd\std.vhd":123:18:123:21|Setting time resolution to ns
@N:"E:\projects\CubeSatCOM\dev\implementation\spinal\SpinalBaseProject\FSM.vhd":408:7:408:9|Top entity is set to FSM.
VHDL syntax check successful!
@N: CD630 :"E:\projects\CubeSatCOM\dev\implementation\spinal\SpinalBaseProject\FSM.vhd":408:7:408:9|Synthesizing work.fsm.arch 
@N: CD233 :"E:\projects\CubeSatCOM\dev\implementation\spinal\SpinalBaseProject\FSM.vhd":10:26:10:27|Using sequential encoding for type fsm_enumdefinition
@E: CD400 :"E:\projects\CubeSatCOM\dev\implementation\spinal\SpinalBaseProject\FSM.vhd":442:11:442:18|Type binding failed
@END
Process took 0h:00m:01s realtime, 0h:00m:01s cputime
# Fri Sep 29 14:19:05 2017

###########################################################]
```
VHDL source file
===========
```VHDL
-- Generator : SpinalHDL v0.10.15    git head : 28b16cb90aa742a1126bc60a447a61f1d211f6fc
-- Date      : 29/09/2017, 14:17:03
-- Component : FSM

library IEEE;
use IEEE.STD_LOGIC_1164.ALL;
use IEEE.NUMERIC_STD.all;

package pkg_enum is
  type fsm_enumDefinition is (boot,fsm_stateA,fsm_stateB,fsm_stateC);
  type fsm_enumDefinition_debug is (XXX,boot,fsm_stateA,fsm_stateB,fsm_stateC);

  function pkg_mux (sel : std_logic;one : fsm_enumDefinition;zero : fsm_enumDefinition) return fsm_enumDefinition;
  function pkg_toStdLogicVector_native (value : fsm_enumDefinition) return std_logic_vector;
  function pkg_tofsm_enumDefinition_native (value : std_logic_vector(1 downto 0)) return fsm_enumDefinition;
end pkg_enum;

package body pkg_enum is
  function pkg_mux (sel : std_logic;one : fsm_enumDefinition;zero : fsm_enumDefinition) return fsm_enumDefinition is
  begin
    if sel = '1' then
      return one;
    else
      return zero;
    end if;
  end pkg_mux;

  function pkg_tofsm_enumDefinition_native (value : std_logic_vector(1 downto 0)) return fsm_enumDefinition is
  begin
    case value is
      when "00" => return boot;
      when "01" => return fsm_stateA;
      when "10" => return fsm_stateB;
[... 400 lines cut ...]
        fsm_stateNext <= pkg_enum.fsm_stateA;
    end case;
  end process;

  process(clk,reset)
  begin
    if reset = '1' then
      fsm_counter <= pkg_unsigned("00000000");
      fsm_stateReg <= pkg_enum.boot;
    elsif rising_edge(clk) then
      case fsm_stateReg is
        when pkg_enum.fsm_stateA =>
        when pkg_enum.fsm_stateB =>
          fsm_counter <= (fsm_counter + pkg_unsigned("00000001"));
        when others =>
      end case;
      if ((not zz_1) and zz_2) = '1' then
        fsm_counter <= pkg_unsigned("00000000");
      end if;
      fsm_stateReg <= fsm_stateNext;
    end if;
  end process;

end arch;
```

Dolu1990, 2017-09-29:
It is maybe a specific issue with symplify ?
Maybe he is confused because an enumeration also defined is the enum package also have same elements name ?
Could you give a try by manualy removing the    type fsm_enumDefinition_debug is (XXX,boot,fsm_stateA,fsm_stateB,fsm_stateC);    definition and all related things ?

oligu, 2017-09-29:
As the synthesis works with ISE, it seems to be a Symplify issue.
Removing that line didn't change anything.

Dolu1990, 2017-10-02:
Fixed in 64c7e467773ed81324b09373e8449060d98e390a

closed 2017-10-02 by Dolu1990