SpinalHDL/SpinalHDL #84: Unassigned, initialized register generates a Latch

cristianmacario, 2017-11-09:
When we have an initialized register declared but never assigned, a latch is generated (at least by the Verilog backend, I am not a VHDL expert...).
For instance
```scala
val fake_flop = Reg(Bool) init (False)
```
Generates the following Verilog code:
```verilog
  reg  fake_flop;
  wire _1;
  assign _1 = 1'b0;
  always @ (posedge clk or posedge reset)
  begin
    if (reset) begin
      fake_flop <= _1;
    end else begin
    end
  end
```
Which describes a latch!
This causes the following 2 issues:
1. If we declare a Reg, we want to se a Flop in our netlist, not a latch (at least this is what I would expect)
2. Reset signal is connected to the "enable" input of the Latch: this is a misuse of the reset signal and might cause issues at backend

I see the following 3 ways to solve this issue (there might be more):
1. Unassigned Registers give an error, even if they are initialized, just as it happens for Unassigned Uninitialized registers
2. Unassigned Initialized Registers are converted into a constant
3. Unassigned Initialized Registers generate Verilog code like the following:
```verilog
  reg  fake_flop;
  wire _1;
  assign _1 = 1'b0;
  always @ (posedge clk or posedge reset)
  begin
    if (reset) begin
      fake_flop <= _1;
    end else begin
      fake_flop <= _1;
    end
  end
```

For completeness: I am working with the following git head: a6b3b9b9

Dolu1990, 2017-11-09:
Hi,

From my experience (=> not exhaustive), the synthesis tool will report it as a latch in early phases, but then it will be symplified as a constant in futher phases.

Anyway, it is probably not true for all synthesis tools, generate warnings, and also i think that your proposal 1) is a sensitive one.
I will implement proposal 1) in the reworkDev branch to see how the actual code base react and get a pratical feedback ^^

Thanks for the issue !

cristianmacario, 2017-11-09:
Hi Dolu,
many thanks for the quick response and for fixing the issue!
Cheers!

Dolu1990, 2017-11-09:
Ok so, i added a fix on the reworkDev branch as this :

- All register of the design which  are vital (directly or indirectly drive an output) are checked
- The check is that all bits of the register should be assigned at least one time.
- Because in some case, because of the component parameterization, the case could happen (no data assignement but with an Init), you can define your register as following "RegInit(False).unsetRegIfNoAssignement" Then if there is no assignement, it will be converted into a conbinatorial signal/constant during compilation. (This is now used twice in my code base)

Sound fair ?

cristianmacario, 2017-11-09:
You are as fast as the speed of light!
Sounds perfect!
Thanks!

Dolu1990, 2017-11-10:
Renaming of unsetRegIfNoAssignement into allowUnsetRegToAvoidLatch

closed 2017-11-10 by Dolu1990

cristianmacario, 2017-11-14:
Hi,
sorry to bother you again with this ticket, but I think I found a small issue.
It looks like the unsetRegIfNoAssignement does not always work.
I generated VexRiscv (vexRiscv.demo.GenFull) and I see the following signals being generated as latches (instead of being converted into constants)
- CsrPlugin_exceptionPortCtrl_exceptionValidsRegs_0
- CsrPlugin_exceptionPortCtrl_exceptionValidsRegs_1

Also these signals have a latch structure
- CsrPlugin_misa_base
- CsrPlugin_misa_extensions

but this is less critical, since they are never read.

Signal CsrPlugin_mtvec, instead, is correctly converted into a constant.

Dolu1990, 2017-11-14:
Hi,
Do not worry, it's good that you find those issues.
So basicaly the bug was that the transformation phase was iterating over the graph while modifying it, which produced some checks skip.

I have submit a fix into the master branch. I checked the VexRiscv generated RTL and in my case that was fine. But let's me know if you have any other issue.

So git clone the master branch. cd into it, and do an       'sbt clean publish-local'   to get this  fix live on your machine. Just check that when you run your hardware generation the git hash code correspond to the git head :)

cristianmacario, 2017-11-14:
Looks fine to me now as well.
Thanks for the quick support!