SpinalHDL/SpinalHDL #85: Feature Proposal: allowing latches

cristianmacario, 2017-11-09:
Hi,
talking about latches (let's say today is Latch day ;) ), even if they are rarely used, they might be useful (for instance in describing clock gating cells).
I would propose to allow the intentional introduction of Latches, using tags, just as we do with combinatorial loops (with the "noCombinatorialLoopCheck" tag)
For instance, introducing a tag called "noLatchCheck" or something similar, we could define latches like follws:
```scala
  val d = Bool
  val e = Bool
  val l = Bool addTag(noLatchCheck)
  when (e) {
    l := d
  }
```
This would need to be translated in Verilog as:
```verilog
  wire e;
  wire d;
  reg l;
  always @ (e or d)
  begin
    if(e)begin
      l <= d;
    end
  end
```
Let me point out that in Verilog (I don't know about VHDL) latches should be assigned using the non-blocking assignment ("<=") instead of the blocking assignment ("=").
Just a proposal, let me know what you think.
Thanks!

Dolu1990, 2017-11-10:
Hi,

So until now, the idea was "If you want latches, you have to instanciate them through blackboxes"

But now it could be possible to integrate the feature in reworkDev branch, be before trying it, i would like to release the reworkDev branch into the master one, see if everything is fine, get some feedbacks, and later adding new feature of this order (latches).  

Until that time, is the blaboxing way enough ? 
Maybe those blackboxes could be integrated into devRework, and be instanciated via a wrapper. Then when the feature is nativly supported it would not need any changes in the user code base.

cristianmacario, 2017-11-10:
Sure, latches are not so essential and blackboxing is OK.
Future/long term feature is totally fine.

numero-744, 2022-05-30:
For the design I think a `Latch(T)` would feel more natural (like `Reg(T)`). With `LatchWhen(...)`, etc.

So your example would be:

```scala
val d = Bool
val e = Bool
val l = Latch(Bool)
when (e) {
    l := d
}
```

Or

```scala
val d = Bool
val e = Bool
val l = LatchWhen(d, e)
```

closed 2022-11-08 by Dolu1990