SpinalHDL/SpinalHDL #86: StreamArbiter locked should clear when fire assert.

linuxbest, 2017-11-09:
Hi, 

diff --git a/lib/src/main/scala/spinal/lib/Stream.scala b/lib/src/main/scala/spinal/lib/Stream.scala
index 54860c35..c9c3e41a 100644
--- a/lib/src/main/scala/spinal/lib/Stream.scala
+++ b/lib/src/main/scala/spinal/lib/Stream.scala
@@ -397,14 +397,14 @@ object StreamArbiter {
     def transactionLock(core: StreamArbiter[_]) = new Area {
       import core._
       locked setWhen(io.output.valid)
-      locked.clearWhen(io.output.ready)
+      locked.clearWhen(io.output.fire)
     }
 
     def fragmentLock(core: StreamArbiter[_]) = new Area {
       val realCore = core.asInstanceOf[StreamArbiter[Fragment[_]]]
       import realCore._
       locked setWhen(io.output.valid)
-      locked.clearWhen(io.output.ready && io.output.last)
+      locked.clearWhen(io.output.fire && io.output.last)
     }
   }
 }

Dolu1990, 2017-11-09:
Nice shot XD

Dolu1990, 2017-11-10:
Fixed in 862f96c248e52a5c81da103340e5189adc6a7789

closed 2017-11-10 by Dolu1990