SpinalHDL/SpinalHDL #87: [reworkDev] Initial statements outputed in Verilog RTL

cristianmacario, 2017-11-10:
Hi,
first of all, I don't know how stable is the reworkDev branch supposed to be, so if the below described issue is a known problem, please close this ticket.
I tried the reworkDev branch and I like very much how the new Verilog backend behaves, since it produces less verbose code. Nevertheless, I found a small issue: when a bit vector is assigned bit-wise, initial statements are produced in the outputed RTL. These statements are not synthesizable, so they are ignored by the synthesis.
Here is an example
```scala
  val io = new Bundle {
    val obits = out Bits(2 bit)
  }
  io.obits(0) := True
  io.obits(1) := False
```
This generates the following Verilog code
```verilog
  initial begin
    io_obits[0] = 1'b1;
    io_obits[1] = 1'b0;
  end
```
which, as I said, is behavioral code, not RTL.
A possible way to fix it would be to use the `always @*` statement instead of the `initial`, but this would generate issues in simulation, since the `always @*` is not evaluated at time zero. The `always_comb` statement solves this issue, but it is a SystemVerilog statement, not Verilog (I don't know whether you want to switch to SystemVerilog as RTL language). Probably the best way to solve it is to use 'assign' statements:
```verilog
  assign io_obits[0] = 1'b1;
  assign io_obits[1] = 1'b0;
```
I tried to identify the issue, and it seems related to line 347 of ComponentEmiterVerilog.scala (SHA ID 5dc53c5). Given my very low knowledge of Scala, I can't propose a fix.

Dolu1990, 2017-11-10:
Hi,

Basicaly, reworkDev will be soon merged into master and released as 0.11.0
So it's time to flush out bugs ^^

About backends, they are changed a lot compared to previous versions, also all the internal graph of SpinalHDL changed. Everything should be better now XD (i hope)

About the initial stuff, it isn't a bug, but they way i'm solving combinatorial signals which have no sensitivity to any other signal and have a "complex assignement" (when statement, multiple partial assignements, ...)

In VHDL i'm used to use a function to solve this kind of things. maybe here i could do the same ?
Anyway, i tested it on Quartus, and the initial stuff is working for synthesis too, also, i use the initial begin syntax to initialisate ROM.

Do you know a synthesis tool which isn't able to handle initial statements ?

Also  the assign xx = yy; solution isn't an universal solution (can't solve cases where the signals has some when statements)

And yes,  line 347 of ComponentEmiterVerilog.scala  is effectivly the line where this thing happend : D

cristianmacario, 2017-11-10:
Hi Dolu, thanks for the prompt response.

I don't think functions would help in this case: a function would anyway need to be called either from an `assign` or an `always` process, tell me if I am wrong.

I am an ASIC frontend digital engineer so I don't have a big experience with synthesis tools. I don't know whether Cadence and/or Synopsys tools have options to synthesize initial statements. What I know is that no ASIC company would do that. If I give RTL containing initial statements to the backend team, they shout at me very bad and they refuse to synthesize it. Synthesizing non-sythesizable statements can lead to serious problems (like non-equivalence between RTL and netlist).

I would use assign statements in the simple case. If we have when or case statements, a `generate` block can be used:
```verilog
  generate
    if (cond) begin
       ...
    end else begin
      ...
    end
  endgenerate
```
Another way would be to solve the driven value: if there is no sensitivity list, it means that all conditions are known constants, correct? But I don't know whether this is easy to implement.

Otherwise, there is still the `always_comb` option, but in this case the outputed RTL is SystemVerilog.

Dolu1990, 2017-11-10:
There is the funny/dirty way by using function (tested on Quartus) :
```verilog
  wire [1:0] a;
  function [1:0] zz_a;
    input dummy;
    reg [1:0] a;
    begin
      a[0] = 1'b1;
      //////if(xxx) ....   ... ... could be also inserted there
      a[1] = 1'b0;
      zz_a = a;
    end
  endfunction
  assign a = zz_a(0);
```
If this one look ok for you, i will implement it on reworkdev

You right about the backend stuff. It is nightmare ^^
I'm not an verilog expert (i learned VHDL). but i think the generate statement isn't enough for my purpose. i'm realy not sure it could support multiple if statements/switch cases ?

About solving the driven value, sure it could be a solution, but i realy would like to avoid it, as it would need to be able to resolve all possible expressions (lot of simple code) and would need a lot of testing.  for a corner case feature. That's why i realy try to solve it via the generated code.

always_comb isn't acceptable, as probably most people can't use systemverilog (because of their flow, policy, ...)

cristianmacario, 2017-11-10:
You are right, generate statements are nasty.
Moreover they cannot be nested (you can't have a generate inside another generate).

The idea of the function is great! I just had to make a few changes to it to avoid warnings.
```verilog
  wire [1:0] a;
  function automatic [1:0] zz_a (input dummy);
    begin
      zz_a[0] = 1'b1;
      // if (xxxx)
      zz_a[1] = 1'b0;
    end
  endfunction
  assign a = zz_a(1'b0);
```
I also took the freedom to slightly simplify it, to make it less verbose.
Let me know what you think.

Dolu1990, 2017-11-10:
which kind of warnings it avoid ?

Currently i can get this out of the generation : 

val a = (Bits(2 bits))
a(0) := True
a(1) := False

=>

```verilog
   reg [1:0] a;
  function automatic [1:0] zz_a(input dummy);
    begin
      zz_a[0] = 1'b1;
      zz_a[1] = 1'b0;
    end
  endfunction
  wire [1:0] zz_1;
  assign zz_1 = zz_a(1'b0);
  always @ (zz_1) a = zz_1;
```

- `a` is generated as a reg, because it was destinated to be in a always block originaly. It could be possible to generate it as a wire, but it would add complexity and cross dependancy into the backend, i realy try to avoid this kind of things, especialy for corners features :)
- Also i know it look stupid, but realy this kind of cases should not happend often at all, and also generate the same hardware.

Let's me know if it's ok ^^

cristianmacario, 2017-11-10:
I don't have the logfiles available anymore but, if I recall correctly it
was complaining about the function not being "automatic" and about the
width of the value passed to the function.

The generated rtl looks perfect to me.
Thanks for the support!!

closed 2017-11-10 by Dolu1990

Dolu1990, 2017-11-21:
It look like the automatic keyword has some issue with some flow (yosys), is it realy bad if i remove it XD ?

Dolu1990, 2017-11-21:
Ok, i will then remove it :)
About the ticket, it's allright, the person that findout this bug will reporting it ^^ (not me but somebody else)
Thanks anyway : D

cristianmacario, 2017-11-21:
Do you remember which kind of issue is the automatic keyword giving with yosys?
If I recall correctly, I ran some yosys synthesis on vexriscv generated with spinal 0.11.0 (which generates automatic functions) and I didn't have any problem. I will anyway retry tomorrow.

Dolu1990, 2017-11-21:
There is the issue : 
https://github.com/cliffordwolf/yosys/issues/457
=> 
Executing Verilog-2005 frontend.
Parsing Verilog input from `../../gen/src/verilog/J1Ico.v' to AST representation.
ERROR: Parser error in line ../../gen/src/verilog/J1Ico.v:5546: syntax error, unexpected '[', expecting ';' or '('

Just removing the automatic keyword make the synthesis passing :) 
And it appear only into the UART peripheral (that's why you had no issue with 0.11.0 ?)