SpinalHDL/SpinalHDL #89: BlackBox pararam assignment with width-check

cristianmacario, 2017-11-15:
Hello,
since Verilog parameters can have a width (in bits), I would like to be able to assign them as follows, to have a check on the width of the value assigned to them
```scala
// Definition
class bboxedm extends BlackBox {
  val generic = new Generic {
    val intparam = UInt(8 bit)
  }
  val io = new Bundle {
    // ...
  }.setName("")
}

// Instantiation
val bboxedm_inst = new bboedm
bboxedm_inst.generic.intparam := U(7, 8 bit)
// ...
```
Doing this, if I assign to the parameter a value of different size or a value which does not fit, I get an error.
What do you think?
Or is there already a way to check the parameter value width?
Thanks.

cristianmacario, 2017-11-15:
Well, please ignore my my proposal: I realized that if that was possible, we would be able to set parameters with nets -> bad

cristianmacario, 2017-11-15:
Now, let me rephrase my problem. Here is the BlackBox model of an hypothetical Verilog module:
```scala
case class bboxedm (io_width : Int, default_value : Int) extends BlackBox {
  val generic = new Generic {
    val io_width      = U(bboxedm.this.io_width,      32 bit)
    val default_value = B(bboxedm.this.default_value, bboxedm.this.io_width bit)
  }

  val io = new Bundle {
    val clk = in Bool
    val a   = in Bits(io_width bit)
    val z   = out Bits(io_width bit)
  }.setName("")
  mapClockDomain(clock=io.clk)
}
```
This can be instantiated as follows
```scala
val bboxedm_inst = bboxedm (io_width = 256, default_value = 2)
```

Now, my problem is: how can I pass a 256-bit value to the "default_value" parameter?
To solve this problem, I tried to use strings:

```scala
case class bboxedm (io_width : Int, default_value : String) extends BlackBox {
  val generic = new Generic {
    val io_width      = U(bboxedm.this.io_width,      32 bit)
    val default_value = B(bboxedm.this.io_width.toString.concat("'").concat(bboxedm.this.default_value))
  }

  val io = new Bundle {
    val clk = in Bool
    val a   = in Bits(io_width bit)
    val z   = out Bits(io_width bit)
  }.setName("")
  mapClockDomain(clock=io.clk)
}

// Instance
val bboxedm_inst = bboxedm (io_width = 256, default_value = "xFFFFFFFFFFFFFFFFFFFFFFF")
```

This way I was able to set a parameter with with a value which does not fit 32 bits.

Now, for the sake of consistency, I would like to pass also the io_width parameter as a string.
To do so, I would need to be able to convert a UInt to an Int, so that I can set the width of the IOs:
```scala
  val io = new Bundle {
    val clk = in Bool
    val a   = in Bits(generic.io_width.toInt() bit)
    val z   = out Bits(generic.io_width.toInt() bit)
  }.setName("")
```
Is there a way to cast UInt (and SInt) to Int?

Dolu1990, 2017-11-16:
Hi,

What's about : 

```scala
  case class bboxedm (io_width : Int, default_value : BigInt) extends BlackBox {
    val generic = new Generic {
      val io_width      = U(bboxedm.this.io_width, 32 bit)
      val default_value = U(bboxedm.this.default_value, bboxedm.this.io_width bits)
    }

    val io = new Bundle {
      val clk = in Bool
      val a   = in Bits(io_width bits)
      val z   = out Bits(io_width bits)
    }
    mapClockDomain(clock=io.clk)
    noIoPrefix()
  }

  // Instance

  class TopLevel extends Component {
    val bboxedm_inst = bboxedm (io_width = 256, default_value = BigInt("FFFFFFFFFFFFFFF",16))
  }
```

Dolu1990, 2017-11-16:
Also i'm thinking about adding this syntax to avoid the boilerplate : 

```scala
  case class bboxedm (io_width : Int, default_value : BigInt) extends BlackBox {
    addGeneric("io_width", U(io_width, 32 bits))
    addGeneric("default_value", U(default_value, io_width bits))

    val io = new Bundle {
      val clk = in Bool
      val a   = in Bits(io_width bits)
      val z   = out Bits(io_width bits)
    }
    mapClockDomain(clock=io.clk)
    noIoPrefix()
  }
```

cristianmacario, 2017-11-17:
Brilliant! I didn't know about the BigInt (I am a newbie to Scala), looks exactly what I need.
Also the addGeneric() looks like a nice enhancement.
Thanks!!

closed 2017-11-17 by Dolu1990