SpinalHDL/SpinalHDL #93: Default input values for Components?

kaofishy, 2017-11-21:
Hi there,

Is there a way to declare inputs in a Component to have default values? For instance something like
`val input = in UInt(8 bit) := 0`

Right now that statement compiles in Scala but gives me a hierarchy error during HDL elaboration.

Tony

kaofishy, 2017-11-21:
Equivalent VHDL:

`input : in std_logic := '0'`

Dolu1990, 2017-11-21:
Hi
Yes, there is :
https://github.com/SpinalHDL/SpinalHDL/blob/master/lib/src/main/scala/spinal/lib/Stream.scala#L570
This default is completly removed from the moment there is an assignement to this pin.

closed 2017-11-21 by Dolu1990

kaofishy, 2017-11-21:
Thanks! Worked perfectly.

Dolu1990, 2017-11-21:
Anyway thanks for the issue, it allowed to spot the fact that it worked fine for inputs, but not necessarly for outputs XD
I'm currently releasing the 0.11.3, where this feature will work in all cases