SpinalHDL/SpinalHDL #94: [Verilog Backend] Lint issues

cristianmacario, 2017-11-22:
Hi,
I am reporting some Linting issues of the Verilog backend.
These are neither functional issues nor syntax issues, but, since Linting is often used to evaluate the RTL quality, in my opinion, it would be good to solve them. Without solving those, most of the digital designers will not have confidence in the RTL generated by Spinal.

**Width mismatch**
sometimes (when we resize a signal) we can have assignments like this:
```verilog
wire [31:0] a;
wire b;
assign a = b;
```
this is usually considered as a lint issue, since there is a mismatch on the signal widths.
Would be nice to to have some zero padding
```verilog
assign a = {31'd0, b};
```
I guess the opposite scenario (`assign b = a` with `a` wider than `b`) cannot happen, right?

**Signal assigned by mutiple non-blocking assignments**
When we have source code like the following
```scala
val flop = Reg(Bool) init(False)
flop := io.a
when (io.e) {
  flop := io.d
}
```
the generated Verilog RTL looks like follows
```verilog
always @(posedge clk or negedge reset)
begin
  if (reset) begin
    flop <= 1'b0;
  end else begin
    flop <= io_a;
    if (io_e) begin
      flop <= io_d;
    end
  end
end
```
This kind of coding, where the same variable is assigned by multiple non-blocking assignments (`<=`), is considered as a bad practice, even if it is allowed by the Verilog language. The reason for this (as you probably already know) is that non-blocking assignments are executed in 2 steps: evaluation and update, this means that in normal situations they can be re-ordered; in other words
```verilog
always @(posedge clk)
begin
  a <= b;
  b <= a;
end
```
is equivalent to
```verilog
always @(posedge clk)
begin
  b <= a;
  a <= b;
end
```
Now, the Verilog standard states that non-blocking assignments assigning the same variable cannot be reordered and their order must be taken into account. So the generated Verilog describes the desired behaviour (other than being syntactically correct); nevertheless, this coding is considered as a bad practice. I would generate Verilog code as follows:
```verilog
always @*
begin
  flop_nxt = io_a;
  if (io_e) begin
    flop_nxt = io_d;
  end
end

always @(posedge clk or negedge reset)
begin
  if (reset) begin
    flop <= 1'b0;
  end else begin
    flop <= flop_nxt;
  end
end
```

Dolu1990, 2017-11-22:
**assign a = {31'd0, b};    => +1**
How to do sign extends ? like that : ?
input  [15:0] unextend;
output [31:0] extended;
assign extended = {{16{unextend[15]}}, unextend};

**I guess the opposite scenario (assign b = a with a wider than b) cannot happen, right?**
Right, it will not append as to resize down it do    x = y[3:0];

**Signal assigned by mutiple non-blocking assignments**
For this one i think that the good coding practice is more related to "mythology"

Let's take this simple example : 
```scala
val flop = Reg(Bool) init(False)
when (io.e) {
  flop := io.d
}
when (io.f) {
  flop := io.g
}
```

To emit this properly the good practice way, you also have to generate a clockEnable signal : 

```erilog
always @*
begin
  flop_enable = '0';
  flop_nxt = ???; //avoid latch stuff
  if (io_e) begin
    flop_nxt = io_d;
    flop_enable = '1';
  end
  if (io_f) begin
    flop_nxt = io_g;
    flop_enable = '1';
  end
end

always @(posedge clk or negedge reset)
begin
  if (reset) begin
    flop <= 1'b0;
  end else begin
    if(flop_enable) begin
        flop <= flop_nxt;
    end
  end
end
```

But it's so much typing O.o Does people working with Verilog realy have to do this kind of things ? Is it realy a bad practice ? i mean all synthesis/simulation tools are happy with the "bad practice" right ? Sometime linting is just excessive and more about religion, especialy in hardware design. Isn't it ?

cristianmacario, 2017-11-22:
In  my width extension example, I meant unsigned integers or bit vectors, since signed are very rarely used outside ALUs. Hope I didn't hurt your feelings too much ;)

Regarding the multiple non-blocking assignments,  I don't think you need a flop_enable:
```verilog
always @*
begin
  flop_nxt = flop
  if (io_e) begin
    flop_nxt = io_d;
  end
  if (io_f) begin
    flop_nxt = io_g;
  end
end

always @(posedge clk or negedge reset)
begin
  if (reset) begin
    flop <= 1'b0;
  end else begin
    flop <= flop_nxt;
  end
end
```
You can call it bad practice or you can call it religion, choose the one you prefer ;)
We both know the code is equivalent and I agree with you that

Dolu1990, 2017-11-22:
I submited the fix for the signals resizing ^^

About the flop_enable, so, if i'm a stupid synthesis tool, i will infer your last sample of code as a flipflop without clock enable. which isn't the same than adding a if(flop_enable) in the clocked process which will infer an explicit clock enable on the flipflop. (which is better for FPGA because it mean less lut based muxes)

But in practice i don't know the impact of your above sample of code :)

cristianmacario, 2017-11-22:
Sorry, i fired my message before finishing...
We both know the code is equivalent and I agree with you that it makes no difference.
But if you want Spinal to be adopted, you need to make happy the current generation of designers...
Consider it as 'marketing"

cristianmacario, 2017-11-22:
About the 2 coding styles, I think that for modern synthesis tools they are equivalent.
Talking about ASIC, for a single flop, no synthesis tool will add a clock gating cell for only one flop. A gating cell consumes area and power and it is not worth to spend one of them to gate a single flop.

I propose to do the following: considering the syntax without flop_enable, I will try to figure out what happens with ASIC backend tools (if flop is a signal of a 100 flops

cristianmacario, 2017-11-22:
Sorry again... Damn smartphone...

(if `flop` is a signal of 100 bits, would the backend tool be able to infer a clock gating cell?)

I invite you to do the same for FPGA, if you have the possibility. Then we share our findings.

cristianmacario, 2017-11-22:
Thanks for the "resizing" fix BTW

cristianmacario, 2017-11-23:
So, I talked with a synthesis expert: even with the code I proposed, ASIC tools can extract the "update_en" signal (or equation) and they can be instructed to insert (or not) clock gating cells.

Dolu1990, 2017-11-23:
Ok :)
I will check how it go with FPGA stuff. 
Anyway, it shouldn't be hard to change they way how register are generated ^^
Thanks for the linting stuff !

Dolu1990, 2017-11-23:
Is there a way to PM you ?

cristianmacario, 2017-11-23:
Sure, you can PM me ;)

I also ran yosys (amazing tool, isn't it?!) on the 2 versions of the attached verilog module.
[flop_en_check.v.txt](https://github.com/SpinalHDL/SpinalHDL/files/1499779/flop_en_check.v.txt)
Here is the very simple script I ran:
```
read_verilog
prep
write_verilog
```
and the outputed verilog in the 2 cases is the same, except some variable renaming.

cristianmacario, 2019-02-14:
Hi ljiaxi,
thanks for your reply, and sorry for the delayed response, your message slipped  into my backlog...
Thanks for the tip. Nevertheless, my main message is that Spinal should always generate Lint-clean code whatever the source code is. What do you think?

Dolu1990, 2019-02-14:
From my opinon, (but i haven't use lint tools much), the main issue of using lints as reference of how should be emited the verilog/vhdl, are : 
- Some rules probably come from a forgotten legacy
- The lints rules aren't the same for everyone, there is probably much of arbitrary in them

Then, i'm not against having special options to emits the code acordingly to some specific rules. But those specific rules realy need a good reason to be :)

cristianmacario, 2019-02-14:
I agree with both your points.
The fact is that Lint compliance (together with other variables) is often used by the ASIC industry to evaluate the quality of the code.
IMHO, it would be beneficial for the generated code to be compliant with the most common Lint rules.

tomverbeure, 2019-02-14:
> The fact is that Lint compliance (together with other variables) is often used by the ASIC industry to evaluate the quality of the code.
> IMHO, it would be beneficial for the generated code to be compliant with the most common Lint rules.

But code style that started this thread (signal assigned by multiple non-blocking assigments) reports an 'issue' that none of the linters in my career have ever reported as an issue... because they are perfectly valid and synthesizable RTL code. More: the 'fix' that was implemented make the generated Verilog IMO harder to understand because code that belonged together is now split between different processes.

IOW: it's pretty much impossible to generate Verilog code that will satisfy all linters and human...

closed 2019-06-28 by Dolu1990