SpinalHDL/SpinalHDL #107: The annoying extra `.` in `.elseWhen` and `.otherwise`

soronpo, 2018-01-03:
See the following example:
```scala
when(cond1){
  //
}.elsewhen(cond2){
  //
}.otherwise{
  //
}
```
I wish to write the above without the extra `.`
```scala
when(cond1){
  //
} elsewhen(cond2){
  //
} otherwise{
  //
}
```

I investigated the matter on the [scala contributors gitter channel](https://gitter.im/scala/contributors?at=5a4c49db5355812e5745b193) and it turns out this may be a Scalac parser bug.

Follow issue https://github.com/scala/bug/issues/8859 . If resolved, it may solve this issue for Spinal.
I also created a ticked for dotty, which has a similar issue. https://github.com/lampepfl/dotty/issues/3737

som-snytt, 2018-01-03:
Alternatively, `when()() orelse when()() otherwise {}`? Sorry if that's unhelpful.

Dolu1990, 2018-01-03:
Right this is anoying.
@soronpo Currently, you don't need to dot prefix the otherwise, only the elsewhen. :)
@som-snytt I just checked, look like the orelse when trick doesn't work (scala 2.11.6)

Thanks for the issues tracking on the scala compiler : D

PR #108 (2018-01-03): removing the need for `.` in `.elsewhen`
Fixes #107
closed 2018-01-03 by Dolu1990