SpinalHDL/SpinalHDL #114: SpinalSim doesn't work on windows 10

wifasoi, 2018-02-26:
Like the title, SpinalSim doesn't work on Windows 10 (StackTrace: https://pastebin.com/U7A46R1M). Now the code execute posix utilities(rm, mkdir, ecc..) to manipulate directories/files, but the commands windows 10 provide do not adhere to the posix standard. One example:

https://github.com/SpinalHDL/SpinalHDL/blob/80327c8b15eba0407a0e8603d253354ea5f3c9ca/core/src/main/scala/spinal/core/sim/SimBootstraps.scala#L278

In this case the flag -p is not supported by windows 10.
I think the correct thing to do is rewrite the file manipulation part with java/scala io library

Dolu1990, 2018-02-27:
Right, it should be done via some pure scala/java/apache libaries :)
It will be changed

Dolu1990, 2018-02-27:
It should be fixed in the dev branch by 805a4e95adbb4a02c04a7784642ee9aca148d769
Let's me know if you have issues on windows 10, as i can only test on windows 7

closed 2018-02-27 by Dolu1990

wifasoi, 2018-02-28:
The problem is still there: [StackTrace](https://pastebin.com/U8AHvmS0)
I'm running the regression test ( sbt "test-only *SpinalSim*") on 805a4e95adbb4a02c04a7784642ee9aca148d769

Dolu1990, 2018-02-28:
Sorry i missed it XD
I submited another fix on the dev branch, should be fine now, let's me know