SpinalHDL/SpinalHDL #120: Axi4Lite interface at top level

FpgaDude, 2018-04-30:
Hi,
I'm new to the SpinalHDL / Scala language... As one of my first project attempts with SpinalHDL, Im trying to create an AXI4lite peripheral with the AXI IO at the top level,  and i can't seem to get the syntax just right

here is what i have so far...

`class AxiPeriph(axiLite4Config : AxiLite4Config) extends Component  {
  val io = new Bundle {
    val M0 = master(AxiLite4WriteOnly(axiLite4Config))
  }
}

object MicronDmaVerilog {
  def main(args: Array[String]) {
    SpinalVerilog(new AxiPeriph(axiLite4Config = AxiLite4Config(
      addressWidth = 32, dataWidth = 64
    )))
  }
}`

I get a number of errors looking like : NO DRIVER ON (toplevel/io_aw_aw_valid : out Bool), defined at...

Any help is greatly appreciated

Dolu1990, 2018-04-30:
Hi,

SpinalHDL is checking that all parts of the design which are directly/indirectly connected to the outputs of the toplevel are drived by something. (Safety feature)

Currently your code is right, it's just that you leaved some output of the toplevel "floating" :)
so : 
```scala
io.M0.aw.valid := False
io.M0.aw.addr := 0
...
```

FpgaDude, 2018-04-30:
@Dolu1990 Thank you, that was the issue... SpinalHDL is Awesome by the way

closed 2018-04-30 by FpgaDude