SpinalHDL/SpinalHDL #125: Use of generic sighal names (_zz_...) when they don't need to be

tomverbeure, 2018-05-27:
IMO, one of the biggest issues with SpinalHDL is its tendency to use generic names for registers when there's really no reason to do so.

This really makes it a chore to analyze timing issues where they're no good reason for it to be that way.

For example, I have a small SOC for a hobby project that contains a VexRiscv:
https://github.com/tomverbeure/rv32soc/tree/vexriscv

When I run this design through Quartus, and run a timing analysis, I get a worst case timing path from register _zz_185[21] to the program memory.

Looking at the source code, _zz_185 is only used in the following two places:
https://github.com/tomverbeure/rv32soc/blob/2e194973097c6348a92b07cc519611327ed2b63c/vexriscv/VexRiscv_FullNoMmuNoCacheNoCsrNoDbg.v#L2081
and this:
https://github.com/tomverbeure/rv32soc/blob/2e194973097c6348a92b07cc519611327ed2b63c/vexriscv/VexRiscv_FullNoMmuNoCacheNoCsrNoDbg.v#L1062'

There is really no good reason to have this second assignment. The code should simply be:
```
    if((! decode_arbitration_isStuck))begin
      decode_INSTRUCTION <= _zz_73;
    end
```
(and, of course, _zz_73 should be fetch_INSTRUCTION as well.)

I totally understand an automated code generator will always result in cases where you end up with signal names that are auto-generated and meaningless. But the code generator should also go out of its way to only do this when it's strictly necessary, especially for clock registers, because those will show up in timing scripts, coverage reports etc.

Tom

Dolu1990, 2018-05-27:
It isn't a SpinalHDL issue, it is a VexRiscv issue. The thing that you are pointing in the issue, is coming from the dataflow pipeline generator of VexRiscv. It defined input and output of each stages with a proper name, but the registers which linked stages together were defined without names. 

I submited a fix ( 1752b5f184b38347bff4824a0e1235a70c6cae6c ) , give a try, it realy should be much better.

Thanks for pointing this lack of naming anyway :D

tomverbeure, 2018-05-28:
Excellent! That's a huge improvement!

This solves almost all cases of unnamed registers, except for the list below:
```
egrep "zz.*<=" VexRiscv_FullNoMmuNoCacheNoCsrNoDbg.v
      _zz_149 <= RegFilePlugin_regFile[decode_RegFilePlugin_regFileReadAddress1];
      _zz_150 <= RegFilePlugin_regFile[decode_RegFilePlugin_regFileReadAddress2];
      _zz_86 <= 1'b0;
      _zz_107 <= 1'b1;
      _zz_119 <= 1'b0;
        _zz_86 <= 1'b1;
        _zz_86 <= 1'b0;
      _zz_107 <= 1'b0;
      _zz_119 <= (_zz_47 && writeBack_arbitration_isFiring);
      _zz_87 <= iBus_rsp_inst;
    _zz_120 <= _zz_46[11 : 7];
    _zz_121 <= _zz_69;
```
I think the first two are probably worth fixing, since they're a meaningful member of the pipeline. The rest are probably nice to have.

I'm going to make a wild guess and speculate that the following two lines will need the same `setName` treatment?
https://github.com/SpinalHDL/VexRiscv/blob/1752b5f184b38347bff4824a0e1235a70c6cae6c/src/main/scala/vexriscv/plugin/RegFilePlugin.scala#L53-L54

Thanks!
Tom

Dolu1990, 2018-05-28:
About 
      _zz_149 <= RegFilePlugin_regFile[decode_RegFilePlugin_regFileReadAddress1];
      _zz_150 <= RegFilePlugin_regFile[decode_RegFilePlugin_regFileReadAddress2];
This one is due to the way how SpinalHDL emit the memories. Basicaly, it emit the memory ports on anonymus signals, which are then connected to the rs1Data,rs2Data signals. 
It should be possible to improve that by naming the anonymus signal with some memory/address name concatenation.

For others anonymus register, it basicaly when i'm using some handy SpinalHDL syntax, like : 

```
//Assign mySignal from a register which has a reset value 'False', which is set when xxx is true and clear when yyy is true. clear has the priority over set.
mySignal := (RegInit(False) setWhen(xxx) clearWhen(yyy))
```

Let's me know if one of those register appear in the critical path of your design, I haven't done much area/fmax profiling on Artix7, how is it going ?

tomverbeure, 2018-05-28:
Those registers don't appear in the critical path, so it's not a big deal.

I'm currently using an Altera Cyclone II EP2C5, but I'll eventually move to a Xilinx Spartan 3 and an Altera Cyclone 4.

closed 2018-05-28 by tomverbeure