SpinalHDL/SpinalHDL #127: Vec init(0)

zeroday-engineer, 2018-06-19:
I've been having a hard time finding the proper syntax/preferred method of initializing a Vec(Reg(Uint())), currently using the map function. Is init extended for arrays/multidimensional representations?

Dolu1990, 2018-06-19:
Hi,

```scala
  class TopLevel extends Component {
    val x = Vec(Reg(UInt(8 bits)) init(0), 4)
    val y = Vec(Reg(UInt(8 bits)), 4)
    y.foreach(_ init(0))
    
    //Just to avoid having reseted only flops
    x := y
    y := x
  }
```

Should be applicable to more dimension :)

zeroday-engineer, 2018-06-19:
WoW, that's simple. Thank you for your help. Is there a Discord channel or some other chat setup for general questions? I kinda feel bad about creating an issue every time I'm stuck :)

Dolu1990, 2018-06-19:
Sure, there is : 
https://gitter.im/SpinalHDL/SpinalHDL
No problems ^^

closed 2018-06-19 by zeroday-engineer