SpinalHDL/SpinalHDL #128: cdc check for multi-clock black box

asicdruide, 2018-06-28:
Hi,
we do ASIC designs with many (>300) clock domains.
Currently we develop in SystemVerilog.
We don't have a methodology to automatically detect clock-domain-crossings better than following the rule of "don't do mistakes".
To benefit from SpinalHDLs ability to assert on clock-domain-crossings we have the idea to, in the first step, convert the toplevel to SpinalHDL and import all existing sub-modules as  black boxes.
Some of the sub-modules have multiple clocks inputs. 
I know from the SpinalHDL black-box documentation the mapClockDomain() method, but i assume that this sets the domain membership for all inputs/outputs.

I'd like to know whether it's possible to assign ports of a black-box individually to clock domains?

I'm fully aware that a manual mapping is error prone but still better than what we have today (nothing).

If this is already possible please provide a pointer to an example/docu.

If this is not already possible please treat my question as a feature proposal because i think this area of clock-domain-crossing checks could be a good story to push SpinalHDL into old-styled design teams.

Thank you for your support!

Best regards,
Klaus

Dolu1990, 2018-06-28:
Hi,

That's funny because i think that a feature that was very recently added by Critian

You need SpinalHDL 1.1.6

```
class MyBlackBox(val cd : ClockDomain) extends BlackBox{
  val i = in(Bool).addTag(ClockDomainTag(cd))
  val o = out(Bool).addTag(ClockDomainTag(cd))
}
```

Then those signals will be assigned to the given clock domain. It is tested and work in regressions tests, but it's isn't extensivly tested, Anyway if you have issues with the feature just let's me know.

Then about blackboxing all the component you already did in SV to instanciate them in a SpinalHDL toplevel. Just, you have to know that blackboxing will not support multi dimentional array and SV 'interface'.

What kind if design are you doing to have that many clock domains ^^ ?

Best regards
Charles

asicdruide, 2018-06-29:
Hi Charles,
many thanks for the quick response. That's indeed funny that Critian just did what i had in mind.

That Blockboxes do not support SV interfaces is unfortunate. This would mean that we've to wrap the original SV interfaced modules with a level having normal ports and use this wrapper as blackbox. This limits the benefit of interfaces dramatically because you can't just add a new wire to an empty pipe without also touching the wrappers and toplevel. I've to think more about it (e.g. generating the interface definition from a bundle that SpinalHDL code becomes the master).

The design we're doing consist of one central part with 6 clocks from indivual PLLs (for power tweaking reasons), 4 DDR4 controllers having a few individual clock domains with enable, 8 cores having a digital clock, several phase shifted clocks for communication with an analog part, an 8GHz domain, some 1GHz domains, separate rx-and tx clocks from 12 serdes interfaces, clocks from analog part, several reference clocks, clocks for scan/mbist/jtag/i2c interfaces,...

Bottomline a lot of clock domains without any static check/assertion. Synthesis/STA just sets max-delays per default on all domain crossings.

I've looked at the way the big EDA companies approach the issue (extracting clock domain properties from the design (=automated reverse engineering) and evaluating those).
I'd prefer the abstract (forward) approach of handling this issue in the design domain. But this needs support from the design language that SV and VHDL can not provide.

Best regards,
Klaus

numero-744, 2022-05-30:
So as a conclusion, the feature has already been added but another feature of supporting Verilog interfaces has been suggested.

May we close this issue, and open another one if Verilog interfaces matter? If so, please mention https://github.com/SpinalHDL/SpinalHDL/pull/318 in the new issue as it would affect blackbox generation.