SpinalHDL/SpinalHDL #130: Issue warnings when BusSlaveFactory read multiple signals to the same address

marcoep, 2018-07-17:
Lets say I have multiple modules all of which I drive from the same bus. For all of them I use the same BusSlaveFactory. There should be a way to detect when reading multiple signals at the same address, as this is most commonly the case when mis-configuring base-addresses, addresses, or bitOffsets.

So for example this spinalHDL code:
```scala
val apbSlaveFactory = new Apb3SlaveFactory(io.apb3, 0)
val thermalCtrlDigitalBridge = thermalCtrlDigital.driveFrom(apbSlaveFactory, 0x00)
val thermalCtrlDigitalBridge_2 = thermalCtrlDigital_2.driveFrom(apbSlaveFactory, 0x04)
```
but  I have made the mistake that one thermalCtrlDigital actually reads from addresses base+0x00 to base+0x07! This then results in the following data model:
```
...
Address 0x4 :
  R[5:0] io_AIF_TCtrl_ref_current_DAC_o - ThermalControllerDigital: analog interface: Iset [5..0] 
  W[0:0]  - ThermalControllerDigital FSM control bit: start_locking 
  R[0:0]  - ThermalControllerDigital FSM control bit: start_locking 
  W[1:1]  - ThermalControllerDigital FSM control bit: start_ramp_heater 
  R[1:1]  - ThermalControllerDigital FSM control bit: start_ramp_heater 
  W[2:2]  - ThermalControllerDigital FSM control bit: stop_locking 
  R[2:2]  - ThermalControllerDigital FSM control bit: stop_locking 
  R[3:3] io_idle - ThermalControllerDigital FSM control bit: idle 
...
```
and in the following VHDL:
```VHDL
...
        io_apb3_PRDATA(5 downto 0) <= std_logic_vector(zz_6);
      when "00000100" =>
        if apbSlaveFactory_doWrite = '1' then
          zz_13 <= pkg_toStdLogic(true);
        end if;
        io_apb3_PRDATA(5 downto 0) <= zz_30;
        io_apb3_PRDATA(0 downto 0) <= pkg_toStdLogicVector(zz_7);
        io_apb3_PRDATA(1 downto 1) <= pkg_toStdLogicVector(zz_8);
        io_apb3_PRDATA(2 downto 2) <= pkg_toStdLogicVector(zz_9);
        io_apb3_PRDATA(3 downto 3) <= pkg_toStdLogicVector(zz_40);
      when "00000101" =>
...
```
which can hardly be any designers intention.

If somebody helps me setup a spinalHDL dev environment, I can try to implement an assertion.

Dolu1990, 2018-07-19:
Hi,

Right, this kind of things should be be checked just before this line : 
https://github.com/SpinalHDL/SpinalHDL/blob/master/lib/src/main/scala/spinal/lib/bus/misc/BusSlaveFactory.scala#L620

About setting the environnement, the SpinalHDL repository is like the Spinal template projects, it is SBT base,
then there is some sandbox/dummy scala file where you can try your things, like  https://github.com/SpinalHDL/SpinalHDL/blob/master/tester/src/main/scala/spinal/tester/PlayDev.scala :)
You can also add automated regression tests  like for instance https://github.com/SpinalHDL/SpinalHDL/blob/master/tester/src/test/scala/spinal/tester/scalatest/SpinalSimStreamFifoTester.scala#L16

Is that fine ?

marcoep, 2018-12-16:
Solved in PR #164

closed 2018-12-16 by marcoep