SpinalHDL/SpinalHDL #133: RFE: Change autocreated signal names from _zz_<number> to _zz_<number>_

tomverbeure, 2018-08-10:
Right now, autocreated signal names look this: `_zz_<number>`.

When there are many such signals, this makes it hard to search to the generated Verilog files with an editor, because a search for, say, `_zz_3` will also match `_zz_30`, `_zz_31` ...

This can be avoided if the signals are named `_zz_3_` and `_zz_30_` instead.

Tom

Snoopy87, 2018-08-10:
Normally, when you are looking for a specific word inside an editor, there is an option called "match whole word" or something like that. For instance, if you are using intellij there is an option named "Word". With this option you are going the get all _zz_3 without _zz_30, _zz_31, ...

tomverbeure, 2018-08-10:
Ok. But are there good reasons to not do this? With this proposal, it doesn’t even matter what your editor (or the keyboard jockey who uses it) can do.

Tom

Dolu1990, 2018-08-10:
For the VHDL generation, it can't be done with a _, as VHDL itself is very restrictive about the usage of _, doesn't allow doubl _, ending _ or starting XD
But for the verilog backend, i would say why not implementing it. Also some wave viewer doesn't have word filter to navigate through signals.

tomverbeure, 2018-08-23:
Here's one other potential (optional?) addition: strictly incrementing _zz_ numbers across different modules.

Even though I'm now getting more comfortable with SpinalHDL, I still find myself looking at the Verilog quite a bit when searching for combinatoric signals. 

One issue there is that multiple Verilog modules have the same _zz_1 signal. Since all modules are written to the same file, this makes it difficult to find the correct location in the text file.

It'd be much easier to have unique names across all modules by never resetting the counter.

Dolu1990, 2018-08-23:
I think that's is a realy good idea, so yes, +1
Will create somme big numbers zz_xx_, but that's just netlist stuff.

Dolu1990, 2018-08-25:
Hmm about the global zz identifier, then the issue is that when you do a single change, the whole netlist will change => not friendly for versionning tools / diff

tomverbeure, 2018-08-25:
That’s true. I personally don’t care a lot about the versioning part, but it also impacts waveform viewing.

One solution is that for each module, you could start with an integer multiple of, say, 1000 or 10000.

Or you could optionally write out multiple files, one for each module. 

In general, I’d like to have more flexibility in writing out verilog files. Right now, I don’t even know how to change the file name. :-)

Tom

tomverbeure, 2018-08-25:
Yes another option: use a different prefix per module and still start at 1.

Is there currently a way to set the prefix per module? That’s be probably my preference.

Dolu1990, 2018-08-27:
@tomverbeure Feature added in #133 for the verilog / system verilog backend :)

tomverbeure, 2018-08-27:
Thanks! I have a lot of things to try this week. :-)

tomverbeure, 2018-09-06:
The _ postfix is working fine. I've been using it for the past few days.

I'm seeing in practice now that the global increment or a per-module prefix would be really helpful, since I'm constantly going through the source code to identify critical paths in my design, and they're often include _zz_ paths.

Shall I close this issue and file a different one for that, or shall I simply leave this one open?

Dolu1990, 2018-09-06:
Cool :)
So let's close this issue, and open a new one for the increment ^^
Then, i don't know what should be the default setup (with or without global increment)

Dolu1990, 2018-09-06:
Also, which kind of design are your analysing ?

tomverbeure, 2018-09-06:
This design:
https://github.com/tomverbeure/mr1

All the cool kids are designing their own RISC-V, so it seemed like a good non-trivial project to test drive SpinalHDL and riscv-formal.

Tom

tomverbeure, 2018-09-10:
This RFE has been implemented and can be closed. As separate RFE has been filed.

Tom

closed 2018-09-10 by tomverbeure