SpinalHDL/SpinalHDL #137: Minimal support for assert, assume, cover

tomverbeure, 2018-08-19:
Right now, a SpinalHDL assert statement like this:

`assert(!(dBus.rsp.ready && input(MEMORY_ENABLE) && arbitration.isValid && arbitration.isStuck),"DBusSimplePlugin doesn't allow memory stage stall when read happend")
`

gets converted into something like this:
```verilog
    if (!(! (((dBus_rsp_ready && memory_MEMORY_ENABLE) && memory_arbitration_isValid) && memory_arbitration_isStuck))) begin
      $display("ERROR DBusSimplePlugin doesn't allow memory stage stall when read happend");
    end
```

This is useful enough for simulation, but it's not for formal verification.

Today, Symbiyosys makes it possible to write do formal verification with open source software.

Its default capability are limited and don't come close to the formal verification features of professional SystemVerilog tools, but they are sufficient to do formal proof of CPUs etc.

It'd be extremely useful to have support for these limited features in SpinalHDL.

All that's needed is the following:

* A section that is surrounded by ifdef FORMAL ... endif
* the ability to generate `assert`, `assume`, and `cover` statement
    These statement should be able to receive a Verilog expression just as is the case for, say, the expression of a port assignment
* the ability to call some new functions such as $past, $rise, ...

An example is here:
https://github.com/YosysHQ/SymbiYosys/blob/2fef25f93dd1cb5137a08e71f507e3eee8100fb1/docs/examples/quickstart/demo.sv#L14-L18

Ideally, I think that the implementation of functionality should be split into 2 parts: one part that supports the insertion of the assert, assume etc statements. And one part that makes it possible to surround an arbitrary piece of code (for example, an Area or a Component) with arbitrary ifdef ... endif macros.

Tom

Dolu1990, 2018-08-19:
Hi,

Yes it need improvements

- ifdef FORMAL -> As you did in the previous pull request that's quite easy too
- So adding assume/cover support is like adding clone of assert with another keyword, so that's easy
- $past, $rise, can be seen as new unary operators, it isn't complicated
- Can i assume that all assert/cover/assume statements are clocked ? (to avoid deltacycle glitch issues)

> And one part that makes it possible to surround an arbitrary piece of code (for example, an Area or a Component) with arbitrary ifdef ... endif macros.
This one is the most tricky, as the idea of SpinalHDL is to generate a netlist where all the elaboration is already solved and checked. So having the 'ifdef statement around a comonent can be allowed only if that component is a leaf which doesn't provide any thing back to the design.
Adding arbitrary ifdef macro around assert/.. statements is fine, as it isn't realy part of the netlist (leaf statements).

If all the above sound ok, then i can start implementing it ^^

tomverbeure, 2018-08-20:
> So adding assume/cover support is like adding clone of assert with another keyword, so that's easy

Yes, it's indeed very similar.

> Can i assume that all assert/cover/assume statements are clocked ? (to avoid deltacycle glitch issues)

It's not something you can universally assume.

Here's an example:
https://github.com/ZipCPU/zipcpu/blob/74f0aa9cca4817035124f21df8e25cdb926b88b6/rtl/core/idecode.v#L820-L835

Under the hood of Yosys, the RTL is literally synthesized to the SMT2 format, which is a standard representation of boolean equations. There is no simulation and there are no delta cycles. 

> This one is the most tricky, as the idea of SpinalHDL is to generate a netlist where all the elaboration is already solved and checked. 

Yes, this is probably a bit of stretch goal. 

> If all the above sound ok, then i can start implementing it ^^

Excellent!

Tom

Dolu1990, 2018-08-20:
> It's not something you can universally assume.
> 
> Here's an example:
>https://github.com/ZipCPU/zipcpu/blob/74f0aa9cca4817035124f21df8e25cdb926b88b6/rtl/core/idecode.v#L820-L835

I'm prudent about this example, sometime, human written code is done by some way to avoid extra verbosity. Is there realy a reason to have those assertion in a combinatorial process ? Maybe sometime it make sense with "assume" to avoid the formal checking to go futher ?

tomverbeure, 2018-08-21:
Let's look at this example: 
https://github.com/cliffordwolf/riscv-formal/blob/3a8fec7dd092362b293e4801eb11bcc704dc3a5c/checks/rvfi_insn_check.sv#L107-L154

I don't know if there's a strong reason why that assert is not clocked: it may be equally possible to do that in a clocked assert.

For sure, you could come up with examples where you don't want them to be combinatorial. (e.g. logic that simply doesn't have a clock)

But I realize that add doing this kind of stuff may not be trivial to plumb into SpinalHDL. It's better to have something that doesn't do everything than nothing at all.

Tom

Dolu1990, 2018-08-22:
I'm doing some progress on this issue, then as i can see, the assert keyword is a SystemVerilog one. So, i will add a SpinalSystemVerilog generation option which will then emite assertion with the assert keyword, and keep the SpinalVerilog generating the assertion with the if statements.

Make sense ?

tomverbeure, 2018-08-23:
Sounds good!

Dolu1990, 2018-08-24:
I submited a first implementation at eb885df07abdd9088cd80413958127bca1ce242e

```scala
object PlayAssertFormal extends App {
  class MyTopLevel extends Component{
    when(in Bool()){
      when(False){
        assert(True,"asd")
        assume(True)
        cover(True)

        formal(assert(True))
        formal(assume(True))
        formal(cover(True))
      }
    }
  }

  SpinalSystemVerilog(new MyTopLevel)
  SpinalVerilog(new MyTopLevel)
  SpinalVhdl(new MyTopLevel)
}
```

the SV generated is : 

```
module MyTopLevel (
      input   clk,
      input   reset);
  always @ (posedge clk) begin
    if(1'b0)begin
      if(1'b0)begin
`ifndef SYNTHESIS
        assert(1'b1) else begin
          $error("asd");
        end
`endif
`ifndef SYNTHESIS
        assume(1'b1);
`endif
`ifndef SYNTHESIS
        cover(1'b1);
`endif
`ifdef FORMAL
        assert(1'b1) else begin
          $error("");
        end
`endif
`ifdef FORMAL
        assume(1'b1);
`endif
`ifdef FORMAL
        cover(1'b1);
`endif
      end
    end
  end

endmodule
```

Let's me know is something isn't good or friendly or can be better / more usable.
I'm not experienced with FV ^^

Dolu1990, 2018-08-24:
One thing is : 
```
        assert(X) else begin
          $error(Y);  //Is the else begin $error  end an issue for formal tools ?
        end
```

tomverbeure, 2018-08-24:
I'll give it a try soon. Probably early next week.

How do I have to change my project so that it uses a local copy of SpinalHDL?

>           $error(Y);  //Is the else begin $error  end an issue for formal tools ?

I don't know. :-) But I'll check it out.

Tom

Dolu1990, 2018-08-24:
There is build.sbt which use a local version of SpinalHDL (should normaly work), but let me know if you have issue

```
lazy val root = (project in file(".")).
  settings(
    inThisBuild(List(
      organization := "com.github.spinalhdl",
      scalaVersion := "2.11.6",
      version      := "0.1.0-SNAPSHOT"
    )),
    name := "superproject"
  ).dependsOn(spinalHdl )
lazy val spinalHdl = RootProject(file("local/path/to/the/VexRiscv/sbt/project/spinalHdl"))

addCompilerPlugin("org.scala-lang.plugins" % "scala-continuations-plugin_2.11.6" % "1.0.2")
scalacOptions += "-P:continuations:enable"
fork := true
```

wifasoi, 2018-09-10:
If I may add some other "nice to have" feature
* support for the restrict statement
* support for initial statement (e.g. initial assume(input < 42) )
* $past has a quirk, the first clock cycle will always return zero.. so there is a need for a is_past_valid macro/function sometimes.. it will be cool if you can specify this with a function parameter ( like past(signal,3,check_valid=true) ).
* support for invariant, when you import already formally verified module, is useful on the child module to exchange assert with assumption and vice versa, to save computation time... ZipCPU does this with macro magic™️ [[blog post about it]](https://zipcpu.com/formal/2018/04/23/invariant.html)

And if i can add my two cent... it will be awesome to have all the formal proof on a separated class, so you can reuse/extend the formal proof of a component.

Next step is to have all this integrated with Symbiyosys as another simulation backend.. but this is way out of scope.

This is what i learnt from reading the ZipCPU's blogposts.. didn't try myself (so take my wish-list with a grain of salt)
I can hack together some code, but after I finish the Wishbone bus implementation.

dvc94ch, 2018-09-17:
Mmh sorry if I missed some docs again.

```scala
import spinal.core._
import spinal.lib._
import spinal.core.sim._
import spinal.core.internals._

class Strobe(resetValue: Int) extends Component {
  val io = new Bundle {
    val reset = in Bool
    val strobe = out Bool
  }
  val counter = Reg(UInt(log2Up(resetValue - 1) bits))

  when(counter > 0) {
    counter := counter - 1
  }

  when(io.reset) {
    counter := U(resetValue - 1)
  }

  io.strobe := counter === 0

  formal {
    val resetPast = Reg(Bool()) init(False)
    when(io.reset) {
      resetPast := True
    }
    cover(resetPast && io.strobe && !io.reset)
  }
}
```

only wraps the cover statement in `ifdef FORMAL. Is there a way to get it to wrap the entire block?

Dolu1990, 2018-09-26:
> Mmh sorry if I missed some docs again.
>
> ```scala
> import spinal.core._
> import spinal.lib._
> import spinal.core.sim._
> import spinal.core.internals._
>
> class Strobe(resetValue: Int) extends Component { [...]

Hmm this would be great, i agree. More generaly, the feature would be to add #ifdef blocks inside the SpinalHDL AST in a proper way. Currently its only applied to the cover statement.
The issue with having this kind of #ifdef construct as a statement into the AST which can host multiple inner statements, is that it break many of the netlist checks. If you have #ifdef stuff inside the netlist, you can't check combinatorial loop, latches throug them as they aren't resolved at SpinalHDL elaboration time.

I have to think about it.

Dolu1990, 2018-09-26:
From a SpinalHDL technical point of view, the best would be to generate two netlist, one with the format statement, and one without. But then it break the safty provided by verifying the same file than you use for synthesis.

wifasoi, 2018-09-27:
> #ifdef stuff inside the netlist, you can't check combinatorial loop, latches throug them as they aren't resolved at SpinalHDL elaboration time.

for formal verification purpose, you don't need to check any that... you can commit all the sins you want, it is not synthesizable code to begin with. The #ifdef is the figurative rug where you hide the dust under, or in this case, not synthesizable code. The #ifdef block is usaually used to make the verilog/vhdl code configurable... something that spinalhdl already does via scala. IMHO a proper implementation of the #ifdef block is not required (I know this statement will bite me in the ass someday, but is what I think), and is not required for this feature.

(disclaimer: I am not a developer, and i don't know how the SpinalHDL internals very well, but I home my jargon can convey some useful idea)

wifasoi, 2018-10-08:
I kwon that yosys can be used to setup problems (some [slides](http://www.clifford.at/papers/2016/yosys-synth-formal/slides.pdf#page=34) on the matter) for equivalence checking (with abc or other smt solver i think)... but I did not use it yet, so i can't comment on how much time it requires. IMHO a flag on the SpinalConfig is fine for me. For formal checking maybe is best to generate the synthetizable file along side the one with all the formal verification one (and use the same seed for both, if possible), for minimizing differences (like I said.. don't know how the spinalhdl core works, so it can be an unnecessary step)

tomverbeure, 2018-10-08:
I've tried formal equivalence checking with Yosys for a VexRiscv with the formal interface and without. It did not work out of the box: Yosys couldn't deal with the fact that the FFs had different names between 2 netlists. (Commercial tools have a separate preprocess step that tried to figure out equivalent FFs.)

It's possible that Yosys can do this with some additional commands (I'm not experienced enough with it to know).

Dolu1990, 2018-10-08:
I implemented that way of doing things, see https://github.com/SpinalHDL/SpinalHDL/blob/3e0099cd65dfc5ecf5f800619734a59770eec7bf/tester/src/main/scala/spinal/tester/PlayDev.scala#L1170 for a example
Now, i will add the $past $rise $fall. Is there some other ones ?

tomverbeure, 2018-10-08:
$changed has recently been added. It’s similar to $rise.

There are also $anyconst, $anyseq, and $initstate.

Dolu1990, 2018-10-09:
@tomverbeure @wifasoi 
I submited most of the $xxx 
There is an example : https://github.com/SpinalHDL/SpinalHDL/blob/dev/tester/src/main/scala/spinal/tester/PlayDev.scala#L1184

Let's me know if you have some idea of improvment :)

wifasoi, 2018-10-09:
ok i noticed this:
```scala
class MyTopLevel extends Component{
    val io = new Bundle{
        val count = out UInt(8 bits)
    }

    val co = Reg(UInt(8 bits))
    co := co +1
    io.count := co
    when(co === 10){
        co := 0
    }

    GenerationFlags.formal{
        assume(io.count < 10)
        assert(io.count >= 15,"test")
    }
}

```

```verilog
module MyTopLevel (
      output [7:0] io_count,
      input   clk,
      input   reset);
  reg [7:0] co;
  assign io_count = co;
  always @ (posedge clk) begin
    co <= (co + (8'b00000001));
    if((co == (8'b00001010)))begin
      co <= (8'b00000000);
    end
    assume((io_count < (8'b00001010)));
    assert(((8'b00001111) <= io_count)) else begin
      $error("test");
    end
  end
endmodule

```

symbiyosys (and yosys) doen't like the assert statement
```
SBY  1:38:20 [hello] base: MyTopLevel.sv:18: ERROR: syntax error, unexpected TOK_ELSE, expecting ';'
```
whre line 18 is the one with the assert

wifasoi, 2018-10-11:
> Let's me know if you have some idea of improvment :)

well, I have some.

Initial statements
---
support for initial statement (initial assume is the most important one) example:
```verilog
   module MyTopLevel (
        output [7:0] io_count,
        input   clk,
        input   reset);
    reg [7:0] co;
    assign io_count = co;
    initial assume( reset == 1);
    always @ (posedge clk or posedge reset) begin
      if (reset) begin
        co <= (8'b00000000);
      end else begin
        co <= (co + (8'b00000001));
        if((co == (8'b00001010)))begin
          co <= (8'b00000000);
        end
      end
    end

    always @ (posedge clk) begin
      assert(((8'b00001000) >= io_count));
    end

  endmodule
//this will fail becouse the counter will reach 10 and reset instead of 8 [8'b00001000]
```
`initial assume( reset == 1);` will make so that the verification starts on a reset state.
IMHO the best way to define something similar in Spinalhdl is:
```scala
initial {
  assume(reset ===1)
  assert(/* some statement that can't be true at the start*/)
}
```
but have initial_assume() and initial_assert() should be fine (and easier i think)

$initstate 
---
$initstate is not in the system verilog standard, but is a [yosys estention](https://twitter.com/oe1cxw/status/1034054538355990529) essentially $initstate substitute can be implemented in verilog with:
from zipCPU's [repo](https://github.com/ZipCPU/zipcpu/blob/ab82a886305ceec79b5516d7fc356f95a762c9fd/rtl/peripherals/wbwatchdog.v#L76):
```verilog
reg	f_past_valid;
initial	f_past_valid = 1'b0;
always @(posedge f_past_valid)
  f_past_valid <= 1'b1
```

assert
---
this:
``` scala
assert(io.count >= 15)
```
will generate this:
``` verilog
assert(((8'b00001111) <= io_count)) else begin
  $error("");
end
```
maybe is more appropriate (and yosys would be happy about it), to not generate the else begin when you don't specify it in the assert statement.
for now (and testing purpose) i can awk/sed/tr/manually remove it.

wifasoi, 2018-10-11:
another nice-to have feature is the ability to flip assert to assumption (and vice versa) to all used module other than the top one.. so you don't have to re-verify alll the verified module. This can be done via read_verilog in yosys... (via read verilog -assert-assumes  to all but the top level) so , like a said, more than a nice to have than a requirement.

Dolu1990, 2018-10-11:
Thanks for the feed back :)
So

For the 
```
    assert(((8'b00001111) <= io_count)) else begin
      $error("test");
    end
```
It is a yosys issue right ? I'm taking as reference https://www.doulos.com/knowhow/sysverilog/tutorial/assertions/.

Then for the rest, i have to improve things ^^

wifasoi, 2018-10-11:
yes, totally a yosys fault.

wifasoi, 2018-12-27:
Ok, found another issue:
past taks two parameters (the second one, if not present, is assumed to 1)
$past(exp,number_of_ticks)
for example:
$past(reset,5)
will return the value of the reset line 5 tick/clock_cycle into the past.

wifasoi, 2018-12-27:
another thing:
it seam that there is no way to generate assume/assert/... constructs in a asynchronous block (I mean inside "always (*)" block)

piegamesde, 2020-03-13:
I tried formally verifying `Stream::slowdown` because it's currently broken and here's a list of issues that I ran into:

- Initializing is hell. The prover just picks some random initial state and everything breaks. And I had a pretty rough time writing code that synthesizes to some Verilog that handles this
    - Give me something that synthesizes Verilog `initial`-Statements, please.
    - Every time a register or value is initialized (e.g. `RegInit`), add an `initial` reset into the Verilog code. The current way it is done is to synthesize some `if reset` code, but this is not enough, especially for synchronous resets.
- When synthesizing Verilog instead of SystemVerilog, all `cover` and `assume` statements silently don't get rendered, which is quite bad.
- I didn't see any formal guards à la `ifdef FORMAL` around any of the statements.
- It would be nice to have `$anyconst`, `$anyseq`, and `$initstate`. I thankfully only needed `$anyseq` for my proof which is quite trivial to emulate but still.

Dolu1990, 2020-03-13:
@piegamesde Thanks for the feedback :D

So i have to say that i'm personnaly not using formal stuff, so what i say now could be wrong and stupid XD

> Initializing is hell. The prover just picks some random initial state and everything breaks. 

Shouldn't the reset of the design avoid such cases ? Or syncronus reset is an issue ? or formal assertion shouldn't be cheked when the reset is active ? Does your checker does combinatorial assertions ?

> Every time a register or value is initialized (e.g. RegInit), add an initial reset into the Verilog code

But that would also tell the synthesis to infer the register as loaded by the FPGA bitstream right ?

> When synthesizing Verilog instead of SystemVerilog, all cover and assume statements silently don't get rendered, which is quite bad.
> I didn't see any formal guards à la ifdef FORMAL around any of the statements.

So, there is no Verilog cover/assume equivalents right ? Basicaly, in SpinalHDL, the idea isn't to use ifdef stuff in the generated netlist, but instead use : 

```
GenerationFlags.formal{
    Your  code which will only be generated if the SpinalConfig enable formal
}
```

https://spinalhdl.github.io/SpinalDoc-RTD/SpinalHDL/Other%20language%20features/formal.html

This resolve your two cases, right ?

> $anyconst, $anyseq

Those are more in the optic to write your formal testbench in SpinalHDL, more than using it in the dut right ?

Also, can they be used in a always block or in a assign statement ? In other words, is that a regular expression ?

piegamesde, 2020-03-13:
> So i have to say that i'm personnaly not using formal stuff

This is only my second proof as well, so I might doing something overly complicated because I don't know any better …

I'm using the formal properties files of ZipCPU, which sometimes use combinatorial check statements and `initial asserts`. I cannot generate code that fulfills them using Spinal and have to modify that file.

> But that would also tell the synthesis to infer the register as loaded by the FPGA bitstream right ?

Honestly, I don't know. Wouldn't this be favorable anyways for an initialized register?

> So, there is no Verilog cover/assume equivalents right ?

Yes and no. Yosys has [extended](https://symbiyosys.readthedocs.io/en/latest/verilog.html) it to include those statements. I am using `GenerationFlags.formal` and enabling formal output.

Regarding `anyconst` and `anyseq`: As written in the formal Verilog language extension, these are some kind of annotations to wires/registers. They tell the prover to treat them as some kind of "proof input".

Dolu1990, 2020-04-06:
@piegamesde Sorry for the delay, just ping me if i forgot to follow up things.

> Honestly, I don't know. Wouldn't this be favorable anyways for an initialized register?

Not realy, because some FPGA do not support it and would emit warnings / linting issues.

So, what is the usage of "initial asserts" ? what is the purpose ? Doesn't formal verification testbench start by doing a reset to the design ?

> anyconst, anyseq

So they are used to write the formal testbench, and also used in the netlist when you strip out parts of your design to speed up formal verification of some parts ? right ?

wifasoi, 2020-04-06:
If I may answer:
Initial *condition* it useful in formal verification for force an initial condition when you run a BMC verification. (Aka start from a condition and check afte n cycle you're in an acceptable state).
The workaround is to use $initstate like I wrote above.

Anyconst and anyseq are just custom Attribute (I usually add them manually with the .addAttribute) and tell the solver to put a constant value in the register(anyconst) or put a new value in the register for each tick (anyseq).
Anyconst and anyseq works only on registers.

Dolu1990, 2020-04-14:
@wifasoi thanks ^^
> Anyconst and anyseq are just custom Attribute (I usually add them manually with the .addAttribute)

Is see in the http://www.clifford.at/papers/2017/smtbmc-sby/slides.pdf anyconst syntax is : 
wire [7:0] cmd = $anyconst;

What are the equivalent attributes ?

wifasoi, 2020-04-14:
I'm basing on: https://symbiyosys.readthedocs.io/en/latest/verilog.html#unconstrained-variables
So maybe can be both?

piegamesde, 2020-04-14:
These are custom Verilog extensions, so no wonder that they're not 100% standardized. But I don't think Spinal needs to support both, one of the options should be enough.

jjj11x, 2020-04-28:
> @wifasoi thanks ^^
> 
> > Anyconst and anyseq are just custom Attribute (I usually add them manually with the .addAttribute)
> 
> Is see in the http://www.clifford.at/papers/2017/smtbmc-sby/slides.pdf anyconst syntax is :
> wire [7:0] cmd = $anyconst;
> 
> What are the equivalent attributes ?

The equivalent attributes are `(* anyseq *)` and `(* anyconst *)`, e.g:
~~~ verilog
(* anyseq *) logic data;
(* anyconst *) logic addr;
~~~

jjj11x, 2020-04-28:
> @piegamesde Sorry for the delay, just ping me if i forgot to follow up things.
> 
> > Honestly, I don't know. Wouldn't this be favorable anyways for an initialized register?
> 
> Not realy, because some FPGA do not support it and would emit warnings / linting issues.

For formal, it would be useful to support at least one of the following (can view the formal synthesis target as an FPGA that supports initial register values):

~~~ systemverilog
logic a = 1'b0;
~~~

or:

~~~ systemverilog
logic a;
initial a = 1'b0;
~~~

> 
> So, what is the usage of "initial asserts" ? what is the purpose ? Doesn't formal verification testbench start by doing a reset to the design ?

As wifasoi mentioned, the primary use would be either initial conditions (e.g. as shown above) or initial assumes (not asserts) such as:

~~~ systemverilog
(* anyseq *) logic a;
initial assume a = 1'b0;
~~~

will constrain `a` to be 0 on the first clock, but then allow it take other values afterwards. This could be useful for modeling devices. You are correct, that in most cases there would probably be a reset, e.g. something more like:
~~~ systemverilog
(* anyseq *) logic a;
always_ff @(posedge clk) begin
    if (rst) begin
        assume(a == 1'b0);
    end
end
~~~

jjj11x, 2020-04-28:
> another nice-to have feature is the ability to flip assert to assumption (and vice versa) to all used module other than the top one.. so you don't have to re-verify alll the verified module. This [...]

For future people reading this, this isn't generally true, gisselquist has since published a follow on article: https://zipcpu.com/formal/2018/12/18/skynet.html

Dolu1990, 2021-08-02:
Hi all,

I recently released a new set of feature : 
https://spinalhdl.github.io/SpinalDoc-RTD/SpinalHDL/Other%20language%20features/formal.html#clocked-assertion-without-reset

And below

My guess is that we could avoid the usage of initstate() with them, and that the example could be rewritten.

wifasoi, 2021-08-02:
Ok, I'll take some time this evening to modify the docs accordingly.

Dolu1990, 2021-08-02:
@wifasoi 

> My guess is that we could avoid the usage of initstate() with them, and that the example could be rewritten.

This is just a guess, i'm not using formal personnaly, it is just out of my competances so far ^^
If anything isn't right, please /all let's me know.

numero-744, 2022-06-03:
I think this issue can be closed as the feature has been added.

Bug issues can be opened if necessary.

Dolu1990, 2022-06-06:
Right ^^

closed 2022-06-06 by Dolu1990