SpinalHDL/SpinalHDL #138: Feature Proposal: Support for XIP mode

roman3017, 2018-08-20:
It would be nice to have XIP mode support for SPI/NOR flash.

Dolu1990, 2018-08-23:
Right, it would be great, would be realy usefull for SoC like Murax.
Also, i'm thinking about implementing a better SPI controller with selectable SPI mode.
Which SPI mode make sense  ?
- half duplex (of course)
- quad SPI
- is there some others ?

roman3017, 2018-08-24:
I think this document has modes nicely described:
[dm00227538.pdf](https://www.st.com/resource/en/application_note/dm00227538.pdf)
I hope to use dual mode with Murax on iCE40-hx8k_breakout_board.

Dolu1990, 2018-08-29:
Definitively something to add. I will work on it ^^

roman3017, 2018-09-01:
Thanks, that would be great.

Dolu1990, 2018-09-06:
What is your target flash ? (as instance)
I have implemented the low level spi stuff (up to quad DDR)

roman3017, 2018-09-06:
My first target is flash on iCE40-hx8k_breakout_board: [n25q032a13esc40f](https://www.micron.com/parts/nor-flash/serial-nor-flash/n25q032a13esc40f)

Dolu1990, 2018-09-15:
Got the XIP controlleur working in sim with the micron's flash simulation model.
Now have to implement hardware breakpoint + scripts to load flash + murax integration XD

roman3017, 2018-09-15:
Sounds great, I can see Apb3SpiDdrMasterCtrl tested. I plan to use JTAG/openocd to load elf image to memory once there is a way to integrate it to Murax.

Dolu1990, 2018-09-18:
I ported openocd to support the spi flash. Now have to create the Murax bootloader to configure the flash controller and that should be ok.

Dolu1990, 2018-09-18:
I got it working on the ice40-hx8k breakout board, FPGA boot on flash, then CPU boot on a small on chip rom which configure the SPI controller and then jump in the flash.

Still WIP, but end to end it's working.

roman3017, 2018-09-23:
I have tried XIP from dev branch but I may be doing something wrong since it is not working for me yet. 

This is what I did:

```sh
cd SpinalHDL/
git checkout dev
sbt clean compile publish-local
cd -

cd VexRisc/scripts/Murax/iCE40-hx8k_breakout_board/
git checkout dev
make compile
make prog
cd -

cd VexRiscvSocSoftware/
git checkout dev
make clean
make
cd -

cd openocd_risc/
git checkout spinalhdl/riscv_spinal_dev -b  spinalhdl/riscv_spinal_dev
make clean
./bootstrap
./configure --enable-legacy-ft2232_libftdi --disable-werror
make
src/openocd -f tcl/interface/ftdi/ft2232h_breakout.cfg -c "set MURAX_CPU0_YAML $PWD/../VexRiscv/cpu0.yaml" -f tcl/target/murax.cfg
```

The JTAG connects fine through openocd and I can connect gdb to the remote target and load the demo_xip.elf. But then I am not able to debug or run it:

```sh
/opt/riscv/bin/riscv64-unknown-elf-gdb demo_xip.elf 
GNU gdb (GDB) 8.0.50.20170724-git
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-pc-linux-gnu --target=riscv64-unknown-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from demo_xip.elf...done.
(gdb) target remote :3333
Remote debugging using :3333
0x80000000 in ?? ()
(gdb) load               
Loading section .init, size 0xb0 lma 0xe0040000
Loading section .text, size 0x380 lma 0xe00400b0
Start address 0xf001e000, load size 1072
Transfer rate: 174 KB/sec, 536 bytes/write.
(gdb) disas 
No function contains program counter for selected frame.
(gdb) step
Cannot find bounds of current function
(gdb) bt
#0  0xf001e000 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) si
0xf001e004 in ?? ()
(gdb) 
0xf001e008 in ?? ()
(gdb) disas
No function contains program counter for selected frame.
(gdb) si   
0xf001e00c in ?? ()
(gdb) 
0xf001e010 in ?? ()
(gdb) 
0xf001e014 in ?? ()
(gdb) 
0xf001e018 in ?? ()
(gdb) 
0xf001e01c in ?? ()
(gdb) cont
Continuing.
```

Probably I am missing something.

Dolu1990, 2018-09-23:
:D
murax_xip.cfg to run openocd

Dolu1990, 2018-09-23:
https://github.com/SpinalHDL/openocd_riscv/blob/riscv_spinal_dev/tcl/target/murax_xip.cfg#L22

It djoud work, but performance is todo

Dolu1990, 2018-09-23:
Hoo also don’t worry about 0xf001e0xx, it is a bootloader rom added to initialise the spi flash, it then jump at 0xE0040000

roman3017, 2018-09-23:
When using murax_xip.cfg I am getting:

```sh
(gdb) target remote :3333
Remote debugging using :3333
Ignoring packet error, continuing...
warning: unrecognized item "timeout" in "qSupported" response
Ignoring packet error, continuing...
Remote replied unexpectedly to 'vMustReplyEmpty': timeout
(gdb) load               
You can't do that when your target is `exec'
```
Probably have some more problems...

Dolu1990, 2018-09-23:
Also, the board jumpers have to be set in flash configuration mode, or  if you want to have the board in stam mode you have to reverse the miso/mosi pins in the toplevel.v, if you do, dont forget to also reverse the input output définition (i forgot it once XD)

roman3017, 2018-09-23:
I am using flash configuration. With RAM configuration I was not able to connect JTAG.

roman3017, 2018-09-23:
This is my openocd output:

```sh
src/openocd -f tcl/interface/ftdi/ft2232h_breakout.cfg -c "set MURAX_CPU0_YAML $PWD/../../atrix/VexRiscv/cpu0.yaml" -f tcl/target/murax_xip.cfg 
Open On-Chip Debugger 0.10.0+dev-00315-g2fbab1f3 (2018-09-22-18:10)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
/home/roman/projects/riscv/openocd_riscv/../../atrix/VexRiscv/cpu0.yaml
adapter speed: 800 kHz
adapter_nsrst_delay: 260
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
jtag_ntrst_delay: 250
Info : set servers polling period to 50ms
Error: libusb_get_string_descriptor_ascii() failed with LIBUSB_ERROR_INVALID_PARAM
Info : clock speed 800 kHz
Info : JTAG tap: fpga_spinal.bridge tap/device found: 0x10001fff (mfg: 0x7ff (<invalid>), part: 0x0001, ver: 0x1)
requesting target halt and executing a soft reset
```

Dolu1990, 2018-09-23:
Hmm weird. I can’t test itt again now, but one thing that i had issue with, inside virtual box, is when both jtag and board usb were mounted

Dolu1990, 2018-09-23:
Can you run openocd with -v3 ? Or-d3 , don’t remember which one

roman3017, 2018-09-23:
I will try debug/verbose next. This one does not show errors but it is not blinking yet:

```sh
telnet localhost 4444
Trying 127.0.0.1...
Connected to localhost.
Escape character is '^]'.
Open On-Chip Debugger
> reset halt
JTAG tap: fpga_spinal.bridge tap/device found: 0x10001fff (mfg: 0x7ff (<invalid>), part: 0x0001, ver: 0x1)
> load_image ..../demo_xip.elf
1072 bytes written at address 0xe0040000
downloaded 1072 bytes in 0.056478s (18.536 KiB/s)
> resume
```
```sh
src/openocd -f tcl/interface/ftdi/ft2232h_breakout.cfg -c "set MURAX_CPU0_YAML $PWD/../../VexRiscv/cpu0.yaml" -f tcl/target/murax_xip.cfg 
Open On-Chip Debugger 0.10.0+dev-00315-g2fbab1f3 (2018-09-22-18:10)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
cpu0.yaml
adapter speed: 800 kHz
adapter_nsrst_delay: 260
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
jtag_ntrst_delay: 250
Info : set servers polling period to 50ms
Error: libusb_get_string_descriptor_ascii() failed with LIBUSB_ERROR_INVALID_PARAM
Info : clock speed 800 kHz
Info : JTAG tap: fpga_spinal.bridge tap/device found: 0x10001fff (mfg: 0x7ff (<invalid>), part: 0x0001, ver: 0x1)
requesting target halt and executing a soft reset
Info : accepting 'telnet' connection on tcp/4444
Info : JTAG tap: fpga_spinal.bridge tap/device found: 0x10001fff (mfg: 0x7ff (<invalid>), part: 0x0001, ver: 0x1)
1072 bytes written at address 0xe0040000
downloaded 1072 bytes in 0.056478s (18.536 KiB/s)
```

Dolu1990, 2018-09-23:
Hoooo i think i get it now
I remember that the first try after running open ocd behave slow to load the flash, then second gdb run was fine. I have to figure out why, there is something weird there

roman3017, 2018-09-23:
This is -d3 openocd output when used with gdb:

```sh
src/openocd -f tcl/interface/ftdi/ft2232h_breakout.cfg -c "set MURAX_CPU0_YAML $PWD/../../VexRiscv/cpu0.yaml" -f tcl/target/murax_xip.cfg  -d3
Open On-Chip Debugger 0.10.0+dev-00315-g2fbab1f3 (2018-09-22-18:10)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 13 4 command.c:544 command_print(): debug_level: 3
Debug: 14 5 options.c:181 add_default_dirs(): bindir=/usr/local/bin
Debug: 15 5 options.c:182 add_default_dirs(): pkgdatadir=/usr/local/share/openocd
Debug: 17 5 options.c:184 add_default_dirs(): bin2data=../share/openocd
Debug: 19 5 configuration.c:42 add_script_search_dir(): adding /local/home/roman/projects/riscv/openocd_riscv/src/../share/openocd/site
Debug: 21 5 configuration.c:82 find_file(): found tcl/interface/ftdi/ft2232h_breakout.cfg
Debug: 22 57 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_interface ftdi
Debug: 23 57 command.c:143 script_debug(): command - interface ocd_interface ftdi
Debug: 25 57 command.c:364 register_command_handler(): registering 'ocd_ftdi_device_desc'...
Debug: 26 57 command.c:364 register_command_handler(): registering 'ocd_ftdi_serial'...
Debug: 27 57 command.c:364 register_command_handler(): registering 'ocd_ftdi_location'...
Debug: 28 57 command.c:364 register_command_handler(): registering 'ocd_ftdi_channel'...
Debug: 29 57 command.c:364 register_command_handler(): registering 'ocd_ftdi_layout_init'...
Debug: 30 57 command.c:364 register_command_handler(): registering 'ocd_ftdi_layout_signal'...
Debug: 31 57 command.c:364 register_command_handler(): registering 'ocd_ftdi_set_signal'...
Debug: 32 57 command.c:364 register_command_handler(): registering 'ocd_ftdi_get_signal'...
Debug: 33 57 command.c:364 register_command_handler(): registering 'ocd_ftdi_vid_pid'...
Debug: 34 58 command.c:364 register_command_handler(): registering 'ocd_ftdi_tdo_sample_edge'...
Debug: 35 58 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_ftdi_serial FTLT3SE
Debug: 36 58 command.c:143 script_debug(): command - ftdi_serial ocd_ftdi_serial FTLT3SE
Debug: 38 58 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_ftdi_vid_pid 0x0403 0x6010
Debug: 39 58 command.c:143 script_debug(): command - ftdi_vid_pid ocd_ftdi_vid_pid 0x0403 0x6010
Debug: 41 58 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_ftdi_channel 0
Debug: 42 58 command.c:143 script_debug(): command - ftdi_channel ocd_ftdi_channel 0
Debug: 44 58 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_ftdi_layout_init 0x08 0x0b
Debug: 45 58 command.c:143 script_debug(): command - ftdi_layout_init ocd_ftdi_layout_init 0x08 0x0b
Debug: 49 58 configuration.c:82 find_file(): found tcl/target/murax_xip.cfg
Debug: 50 240 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_adapter_khz 800
Debug: 51 240 command.c:143 script_debug(): command - adapter_khz ocd_adapter_khz 800
Debug: 53 240 core.c:1631 jtag_config_khz(): handle jtag khz
Debug: 54 240 core.c:1598 adapter_khz_to_speed(): convert khz to interface specific speed value
Debug: 55 240 core.c:1598 adapter_khz_to_speed(): convert khz to interface specific speed value
User : 56 240 command.c:544 command_print(): adapter speed: 800 kHz
Debug: 57 240 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_adapter_nsrst_delay 260
Debug: 58 240 command.c:143 script_debug(): command - adapter_nsrst_delay ocd_adapter_nsrst_delay 260
User : 60 240 command.c:544 command_print(): adapter_nsrst_delay: 260
Debug: 61 240 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_transport select
Debug: 62 240 command.c:143 script_debug(): command - ocd_transport ocd_transport select
Info : 63 240 transport.c:286 jim_transport_select(): auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Debug: 64 240 command.c:364 register_command_handler(): registering 'ocd_jtag_flush_queue_sleep'...
Debug: 65 241 command.c:364 register_command_handler(): registering 'ocd_jtag_rclk'...
Debug: 66 241 command.c:364 register_command_handler(): registering 'ocd_jtag_ntrst_delay'...
Debug: 67 241 command.c:364 register_command_handler(): registering 'ocd_jtag_ntrst_assert_width'...
Debug: 68 241 command.c:364 register_command_handler(): registering 'ocd_scan_chain'...
Debug: 69 241 command.c:364 register_command_handler(): registering 'ocd_jtag_reset'...
Debug: 70 241 command.c:364 register_command_handler(): registering 'ocd_runtest'...
Debug: 71 241 command.c:364 register_command_handler(): registering 'ocd_irscan'...
Debug: 72 241 command.c:364 register_command_handler(): registering 'ocd_verify_ircapture'...
Debug: 73 241 command.c:364 register_command_handler(): registering 'ocd_verify_jtag'...
Debug: 74 241 command.c:364 register_command_handler(): registering 'ocd_tms_sequence'...
Debug: 75 241 command.c:364 register_command_handler(): registering 'ocd_wait_srst_deassert'...
Debug: 76 241 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 77 241 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 78 241 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 79 241 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 80 241 command.c:364 register_command_handler(): registering 'ocd_jtag'...
[... cut ...]
Debug: 288 811 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_init
Debug: 289 811 command.c:143 script_debug(): command - init ocd_init
Info : 291 33321 server.c:96 add_connection(): accepting 'gdb' connection on tcp/3333
Debug: 292 33322 breakpoints.c:357 breakpoint_clear_target_internal(): Delete all breakpoints for target: fpga_spinal.cpu0
Debug: 293 33322 breakpoints.c:497 watchpoint_clear_target(): Delete all watchpoints for target: fpga_spinal.cpu0
Debug: 294 33322 target.c:1524 target_call_event_callbacks(): target event 23 (gdb-attach)
```

roman3017, 2018-09-23:
Looking at Makefile I think I am not running vexriscv.demo.MuraxWithXip but a stale version of Murax. I have cleaned it up and tried to regenerate but got some errors at the end:

```sh
VexRiscv$ sbt "run-main vexriscv.demo.MuraxWithXip"
[info] Loading project definition from VexRiscv/project
[info] Updating {file:VexRiscv/project/}vexriscv-build...
[info] Resolving org.fusesource.jansi#jansi;1.4 ...
[info] Done updating.
[info] Loading project definition from SpinalHDL/project
[info] Set current project to VexRiscv (in build file:VexRiscv/)
[info] Updating {file:VexRiscv/}root...
[info] Resolving jline#jline;2.12.1 ...
[info] Done updating.
[info] Compiling 64 Scala sources to VexRiscv/target/scala-2.11/classes...
[warn] there was one feature warning; re-run with -feature for details
[warn] one warning found
[info] Running vexriscv.demo.MuraxWithXip 
[info] [Runtime] SpinalHDL v1.2.1    git head : 5024cc5616f2314795c6df3db7095c0f58e30209
[info] [Runtime] JVM max memory : 3543.5MiB
[info] [Runtime] Current date : 2018.09.23 09:27:20
[info] [Progress] at 0.000 : Elaborate components
[info] 
[info] **********************************************************************************************
[info] [Warning] Elaboration failed (0 error).
[info]           Spinal will restart with scala trace to help you to find the problem.
[info] **********************************************************************************************
[info] 
[info] [Progress] at 4.450 : Elaborate components
[error] Exception in thread "main" java.nio.file.NoSuchFileException: src/main/c/murax/xipBootloader/crt.bin
[error] 	at sun.nio.fs.UnixException.translateToIOException(UnixException.java:86)
[error] 	at sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:102)
[error] 	at sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:107)
[error] 	at sun.nio.fs.UnixFileSystemProvider.newByteChannel(UnixFileSystemProvider.java:214)
[error] 	at java.nio.file.Files.newByteChannel(Files.java:361)
[error] 	at java.nio.file.Files.newByteChannel(Files.java:407)
[error] 	at java.nio.file.Files.readAllBytes(Files.java:3152)
[error] 	at vexriscv.demo.Apb3Rom.delayedEndpoint$vexriscv$demo$Apb3Rom$1(MuraxUtiles.scala:104)
[error] 	at vexriscv.demo.Apb3Rom$delayedInit$body.apply(MuraxUtiles.scala:102)
[error] 	at scala.Function0$class.apply$mcV$sp(Function0.scala:40)
[error] 	at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
[error] 	at spinal.core.Component.delayedInit(Component.scala:131)
[error] 	at vexriscv.demo.Apb3Rom.<init>(MuraxUtiles.scala:102)
[error] 	at vexriscv.demo.Murax$$anon$3$$anonfun$8$$anon$4.<init>(Murax.scala:305)
[error] 	at vexriscv.demo.Murax$$anon$3$$anonfun$8.apply(Murax.scala:289)
[error] 	at vexriscv.demo.Murax$$anon$3$$anonfun$8.apply(Murax.scala:289)
[error] 	at spinal.core.ifGen$.apply(Misc.scala:440)
[error] 	at vexriscv.demo.Murax$$anon$3.delayedEndpoint$vexriscv$demo$Murax$$anon$3$1(Murax.scala:289)
[error] 	at vexriscv.demo.Murax$$anon$3$delayedInit$body.apply(Murax.scala:208)
[error] 	at scala.Function0$class.apply$mcV$sp(Function0.scala:40)
[error] 	at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
[error] 	at spinal.core.ClockingArea.delayedInit(Area.scala:108)
[error] 	at vexriscv.demo.Murax$$anon$3.<init>(Murax.scala:208)
[error] 	at vexriscv.demo.Murax.delayedEndpoint$vexriscv$demo$Murax$1(Murax.scala:208)
[error] 	at vexriscv.demo.Murax$delayedInit$body.apply(Murax.scala:150)
[error] 	at scala.Function0$class.apply$mcV$sp(Function0.scala:40)
[error] 	at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
[error] 	at spinal.core.Component.delayedInit(Component.scala:131)
[error] 	at vexriscv.demo.Murax.<init>(Murax.scala:150)
[error] 	at vexriscv.demo.MuraxWithXip$$anonfun$main$2.apply(Murax.scala:337)
[error] 	at vexriscv.demo.MuraxWithXip$$anonfun$main$2.apply(Murax.scala:337)
[error] 	at spinal.core.internals.PhaseCreateComponent.impl(Phase.scala:1900)
[error] 	at spinal.core.internals.PhaseContext.doPhase(Phase.scala:194)
[error] 	at spinal.core.internals.SpinalVerilogBoot$$anonfun$singleShot$10.apply(Phase.scala:2142)
[error] 	at spinal.core.internals.SpinalVerilogBoot$$anonfun$singleShot$10.apply(Phase.scala:2140)
[error] 	at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
[error] 	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
[error] 	at spinal.core.internals.SpinalVerilogBoot$.singleShot(Phase.scala:2140)
[error] 	at spinal.core.internals.SpinalVerilogBoot$.apply(Phase.scala:2076)
[error] 	at spinal.core.Spinal$.apply(Spinal.scala:287)
[error] 	at spinal.core.SpinalConfig.generate(Spinal.scala:134)
[error] 	at spinal.core.SpinalVerilog$.apply(Spinal.scala:305)
[error] 	at vexriscv.demo.MuraxWithXip$.main(Murax.scala:337)
[error] 	at vexriscv.demo.MuraxWithXip.main(Murax.scala)
java.lang.RuntimeException: Nonzero exit code returned from runner: 1
	at scala.sys.package$.error(package.scala:27)
[trace] Stack trace suppressed: run last root/compile:runMain for the full output.
[error] (root/compile:runMain) Nonzero exit code returned from runner: 1
[error] Total time: 114 s, completed 23-Sep-2018 9:27:29 AM
```

roman3017, 2018-09-23:
The telnet session followed by gdb seems more promising:

```sh
telnet localhost 4444
Trying 127.0.0.1...
Connected to localhost.
Escape character is '^]'.
Open On-Chip Debugger
> reset halt
JTAG tap: fpga_spinal.bridge tap/device found: 0x10001fff (mfg: 0x7ff (<invalid>), part: 0x0001, ver: 0x1)
> load_image demo_xip.elf
1072 bytes written at address 0xe0040000
downloaded 1072 bytes in 0.056960s (18.379 KiB/s)
> reg pc
pc (/32): 0xF001E000
> exit
Connection closed by foreign host.
```
```sh
/opt/riscv/bin/riscv64-unknown-elf-gdb demo_xip.elf 
GNU gdb (GDB) 8.0.50.20170724-git
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-pc-linux-gnu --target=riscv64-unknown-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from demo_xip.elf...done.
(gdb) target remote :3333
Remote debugging using :3333
0xf001e000 in ?? ()
(gdb) list
1	#include <stdint.h>
2	
3	#include <murax.h>
4	
5	
6	void main() {
7		volatile uint32_t a = 1, b = 2, c = 3;
8		uint32_t result = 0;
9	
10		interruptCtrl_init(TIMER_INTERRUPT);
(gdb) step
Cannot find bounds of current function
(gdb) si
0xf001e004 in ?? ()
(gdb) disas
No function contains program counter for selected frame.
(gdb) break main
Breakpoint 1 at 0xe0040254: file src/main.c, line 7.
(gdb) cont
Continuing.
Note: automatically using hardware breakpoints for read-only addresses.

Program stopped.
main () at src/main.c:7
7		volatile uint32_t a = 1, b = 2, c = 3;
(gdb) step
8		uint32_t result = 0;
(gdb) step
10		interruptCtrl_init(TIMER_INTERRUPT);
```

Dolu1990, 2018-09-23:
Great :D
Sound good, so yes, don't worry about the 0xf001e000 in ??  stuff, directly go to main ^^

roman3017, 2018-09-23:
I do not mind to load through telnet session. I just wonder why blinking is not working. Also uart does not seem to work.

Dolu1990, 2018-09-23:
If you load via telnet : 

```
telnet 127.0.0.1 4444
flash erase_sector 0 4 4
flash write_bank 0 VexRiscv/src/main/c/murax/xipBootloader/demo.bin 0x40000
flash verify_bank 0 VexRiscv/src/main/c/murax/xipBootloader/demo.bin 0x40000
flash erase_sector 0 4 4
reset
resume
```

should make the leds blinking.
Even this one isn't working ?

Dolu1990, 2018-09-23:
(on my board it work, i just can't rerun the FPGA generation/programming flow now, i'm in some IBus rework)

roman3017, 2018-09-23:
No resume does not make it blink either. Since it does not work I am trying to debug. But debugger is behaving bit strange and it just passes through.

No worry, I will play some more. Thank you very much for your help and great work.

I will let you know if I get some news.

roman3017, 2018-09-23:
Finally got it working. It is rather subtle and your uploaded picture helped a lot:

1. program fpga with j6 jumpers horizontal
2. switch j6 jumpers to vertical before running openocd and gdb

Thank you very much for making XIP, it brings Murax to a new level. I think we may close this now.

Dolu1990, 2018-09-23:
So basicaly the double jumper is changing the flash into master or into slave mode (something like this) ^^

Great that it work for you too, also helped to figure out there was missing files.
Any way,  it's still WIP, i have to improve performance of the thing , and why not adding a Murax with an instruction cache.

wip

Dolu1990, 2018-09-24:
So the mmu actualy need i$ and d$, the d$ is currently a bit fat it could be improved, the mmu is a software refilled one

roman3017, 2018-09-25:
For a second I thought we could run Linux there too :).

Dolu1990, 2018-12-01:
Hi @roman3017 
I'm back on this issue XD

So, with SpinalHDL dev i got the spi "Xdr" controller working on all configs in simulation, i'm now creating a SoC dedicated for the ICE40 UP5K : 
https://github.com/SpinalHDL/SaxonSoc

That's WIP, i'm currently writing the simulation environnement :)

Dolu1990, 2018-12-02:
@roman3017 Got SaxonSoc working on the Ice40 UP5K Evn board with instruction cache + Spi Xip. 12 Mhz core with 12 Mhz SPI (using DDR IO to generate the clock out) and dual SPI

roman3017, 2018-12-08:
@Dolu1990: Sounds great, I will try the SaxonSoc as soon as I get a chance.

Dolu1990, 2018-12-08:
Just as info, it use the VexRiscv and SpinalHDL dev branch, which should be cloned in the directoy which contain the SaxonSoc directory

roman3017, 2018-12-16:
@Dolu1990:  I was away and learned this only now:
https://riscv.org/2018/12/risc-v-softcpu-contest-winners-demonstrate-cutting-edge-risc-v-implementations-for-fpgas/
Congratulations!

Dolu1990, 2018-12-16:
Thanks :)
More info there : 
https://riscv.org/2018/12/risc-v-softcpu-contest-highlights/

Only win by one point over 130 XD

roman3017, 2018-12-26:
The best is that openocd with gdb work with SaxonSoc too:

```sh
$ src/openocd -f tcl/interface/ftdi/ft2232h_breakout.cfg -c "set MURAX_CPU0_YAML $PWD/../SaxonSoc/cpu0.yaml" -f tcl/target/murax_xip.cfg
Open On-Chip Debugger 0.10.0+dev-00315-g2fbab1f3 (2018-09-22-18:10)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
/local/home/roman/projects/riscv/openocd_riscv/../SaxonSoc/cpu0.yaml
adapter speed: 800 kHz
adapter_nsrst_delay: 260
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
jtag_ntrst_delay: 250
Info : set servers polling period to 50ms
Info : clock speed 800 kHz
Info : JTAG tap: fpga_spinal.bridge tap/device found: 0x10001fff (mfg: 0x7ff (<invalid>), part: 0x0001, ver: 0x1)
requesting target halt and executing a soft reset
Info : accepting 'gdb' connection on tcp/3333
32 186 22
0
0
80
ffff
fb

```

```sh
$ /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/riscv32-zephyr-elf/riscv32-zephyr-elf-gdb software/standalone/blinkAndEcho/build/blinkAndEcho.elf 
GNU gdb (GDB) 7.12.50.20161011-git
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-pokysdk-linux --target=riscv32-zephyr-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from software/standalone/blinkAndEcho/build/blinkAndEcho.elf...done.
(gdb) target remote :3333
Remote debugging using :3333
_start () at src/start.S:12
12		la sp, _sp
(gdb) monitor reset halt
JTAG tap: fpga_spinal.bridge tap/device found: 0x10001fff (mfg: 0x7ff (<invalid>), part: 0x0001, ver: 0x1)
(gdb) c
Continuing.
^C
Program received signal SIGINT, Interrupt.
main () at src/main.c:21
21	        while(UART_A->STATUS >> 24){ //UART RX interrupt
(gdb) l
16	    while(1){
17	        if(counter++ == 1000){
18	            GPIO_A->OUTPUT = GPIO_A->OUTPUT + 1;
19	            counter = 0;
20	        }
21	        while(UART_A->STATUS >> 24){ //UART RX interrupt
22	            UART_A->DATA = (UART_A->DATA) & 0xFF;
23	        }
24	    }
25	}
(gdb) 

```

roman3017, 2018-12-27:
Got also zephyr hello_world working on SaxonSoc running on hx8k breakout board:

```sh
***** Booting Zephyr OS zephyr-v1.13.0-17-gb614f68c60 *****
Hello World! riscv32
```

Although zephyr synchronization is not working for me yet and gets stuck on threadA like this:

```sh
***** Booting Zephyr OS zephyr-v1.13.0-17-gb614f68c60 *****
threadA: Hello World
```

roman3017, 2018-12-27:
Sounds good. Once you have more time, please let me know if it is possible to make RAM choice to be configurable instead of hard-coding SPRAM in SaxonSoc.scala. Then the necessary changes for supporting hx8k breakout board could be cleaner.

Dolu1990, 2018-12-27:
Sure :)

roman3017, 2018-12-27:
The changes to Saxon in order to support hx8k breakout board are here:
https://github.com/SpinalHDL/SaxonSoc/pull/1

roman3017, 2018-12-28:
Got zephyr synchronization working:
```sh
***** Booting Zephyr OS zephyr-v1.13.0-17-gb614f68c60 ***** 
threadA: Hello World from riscv32!                                              
threadB: Hello World from riscv32!                                              
threadA: Hello World from riscv32!                                              
threadB: Hello World from riscv32!                                              
...
```
The trick it to use RISCV_MACHINE_TIMER instead of VEXRISCV_CONTEST_AREA_TIMER.

Dolu1990, 2019-01-04:
Nice :D

Also, i just found out the bug with Saxon, it was the MachineTimer implementation. I fix is pushed.

Now it work just fine with the RISCV_MACHINE_TIMER.
But i will have to do some zephyr changes in the future to suport 32 bits machine timer to use less FPGA area :)

Dolu1990, 2019-01-06:
Just note that the icecube2 synthesis produce a bugy bitstream, the icestorm one is fine, SpinalSim via verilator too. Will have to give a try on nextpnr.

closed 2019-01-12 by Dolu1990

Amegodvivado, 2023-10-16:
> Right, it would be great, would be realy usefull for SoC like Murax. Also, i'm thinking about implementing a better SPI controller with selectable SPI mode. Which SPI mode make sense ?
> 
> * half duplex (of course)
> * quad SPI
> * is there some others ?

can be implemented quad spi  in which lib now? and how to use?

Dolu1990, 2023-10-16:
currently there is only mosi/miso support with xip.

Dolu1990, 2023-10-16:
In the xipConfig, there is a few parameters as : 
```scala
                                     xipInstructionModInit: Int = 0,
                                     xipAddressModInit : Int = 0,
                                     xipDummyModInit : Int = 0,
                                     xipPayloadModInit : Int = 0,
                                     xipInstructionEnableInit : Boolean = true,
                                     xipInstructionDataInit : Int = 0x0B,
                                     xipDummyCountInit : Int = 0,
                                     xipDummyDataInit : Int = 0xFF,
                                     xipSsId : Int = 0,
```

which specify how the controller to do different sequances of the access,
especialy the "Mod" which allow to switch between let's say 1 bit to 4 bits spi.

You can define new modes.
The default is : 

```scala
SpiXdrMasterCtrl.Parameters(8, 12, SpiXdrParameter(2, 2, 1)).addFullDuplex(0,1,false),
```

meaning only regular SPI (addFullDuplex(0,1,false))

but for instance you can specify a quad spi mode by doing 
```scala
SpiXdrMasterCtrl.Parameters(8, 12, SpiXdrParameter(2, 2, 1)).addFullDuplex(0,1,false).addHalfDuplex(id=1, rate = 1, ddr = false, spiWidth = 4)
```

Meaning Mod should be 1 to do quad SPI

There is no full example out there.

Amegodvivado, 2023-10-16:
for your example :

`SpiXdrMasterCtrl.Parameters(8, 12, SpiXdrParameter(2, 2, 1)).addFullDuplex(0,1,false).addHalfDuplex(id=1, rate = 1, ddr = false, spiWidth = 4)`
will  run false，because the parameter- spiwidth out of Vec(2),must be inside the range (1 down to  0).

 By the way， this lib has many parameter,such as io.rate、rate、different datawidth、spiwidth etc.For my training,some mutual restraint can be see in some sim ,can you explain these parameters simply. Thank you so much.

Dolu1990, 2023-10-16:
Hooo yes you are right

```scala
SpiXdrMasterCtrl.Parameters(8, 12, SpiXdrParameter(4, 2, 1)).addFullDuplex(0,1,false).addHalfDuplex(id=1, rate = 1, ddr = false, spiWidth = 4)
```

> some mutual restraint can be see in some sim

What do you mean by restraint ? 

rate is how many values per controller cycle you can have on a single pin.
spiwidth is the physical number of data bits on the spi
datawidth would refer to spi transfer width, generaly 8, to have bytes transfer (can be done over multiple cycles)

Amegodvivado, 2023-10-17:
The restraint means that the SpixdrParameter can effect the parameters of def of addfullduplex and addhalfduplex.
by the way,about the def -addfullduplex and addhalfduplex ,Isn't it a binary choice? For your example,you use the both of it,why?
`SpiXdrMasterCtrl.Parameters(8, 12, SpiXdrParameter(4, 2, 1)).addFullDuplex(0,1,false).addHalfDuplex(id=1, rate = 1, ddr = false, spiWidth = 4)`
I want to implement the Quad spi (halfduplex, four Bidirectional data ports and other registers configuration) and compatible dual spi(halfduplex ，Bidirectional data ports) ，standard spi(fullduplex，two Unidirectional ports )。Quad spi can backward compatible dual spi ,standard spi.

Dolu1990, 2023-10-23:
It isn't a binary choices, you can setup the XDR controller to support all sorts of SPI modes. See its testbench :) :
https://github.com/SpinalHDL/SpinalHDL/blob/eeab80336e532ddd4c03b379567dec75a273c33c/tester/src/test/scala/spinal/lib/com/spi/xdr/SpinalSimSpiXdr.scala#L45

Amegodvivado, 2023-10-24:
Thank you fou your answer,I will  try to sim this testbench.by the way , if i want to implement a qspi slave interface (like spixdrmasterctrl )to create some ip for emulator ,how to implement it base on lib of we have.Looking forward to your favourable reply!!!

Dolu1990, 2023-10-24:
Ahh, you want a slave SPI controller ? for simulation purpose only ? or it should be synthetisable ?

Amegodvivado, 2023-10-24:
haha,yes,it shoulb be synthetisable,it likes a slave mode of qspi interface flash or eeprom,use to do vip or transactor ,it can be implemented?

Dolu1990, 2023-10-24:
There is currently no support for slave mode.