SpinalHDL/SpinalHDL #141: RFE: Better error reporting on "val data = RegNext(data_old, enable)"

tomverbeure, 2018-08-24:
This code: 
```scala
    val data_old = B("32'd0")
    val enable = True
    val data = RegNext(data_old, enable)
```

Gives the following error:
```
[info] [Progress] at 0.119 : Elaborate components
[error] Exception in thread "main" java.lang.Exception: Undefined assignment (toplevel/??? :  Bits[? bits]) := (toplevel/enable :  Bool)
[error] 	at spinal.core.BaseType.assignFromImpl(BaseType.scala:193)
[error] 	at spinal.core.Assignable$class.compositAssignFrom(Trait.scala:233)
[error] 	at spinal.core.BaseType.compositAssignFrom(BaseType.scala:52)
[error] 	at spinal.core.Data$class.initFrom(Data.scala:295)
[error] 	at spinal.core.BaseType.initFrom(BaseType.scala:52)
[error] 	at spinal.core.DataPrimitives$class.init(Data.scala:65)
[error] 	at spinal.core.DataPimper.init(Data.scala:92)
[error] 	at spinal.core.Reg$.apply(Reg.scala:36)
[error] 	at spinal.core.RegNext$.apply(Reg.scala:49)
[error] 	at mr1.Fetch.delayedEndpoint$mr1$Fetch$1(Fetch.scala:136)
```

The last usable line in the code `Fetch.scala:136` which points to the start of the Component.

Is there a way to be more precise with these class of errors? So not just this specific error but anything where an error points at a general class without telling the exact line in the code should be avoided.

Tom

Dolu1990, 2018-08-24:
I submited a fix to avoid having all the spinal.core.X lines.
Then i know that sometime the scala trace aren't pointing the place were the issue is. Can you but i can't reproduce it. Can you share the complette code on a git branch ?

So now with this example : 
```scala
  class MyTopLevel extends Component{
    val data_old = B("32'd0")
    val enable = True
    val data = RegNext(data_old, enable) //PlayDev.scala:1165
  }

```

It produce the following error message :

```
Register initialisation type missmatch
Reg((toplevel/data :  Bits[32 bits])) init((toplevel/enable :  Bool))
    spinal.tester.PlayErrorImprovment$MyTopLevel.delayedEndpoint$spinal$tester$PlayErrorImprovment$MyTopLevel$1(PlayDev.scala:1165)
    spinal.tester.PlayErrorImprovment$MyTopLevel$delayedInit$body.apply(PlayDev.scala:1162)
    spinal.tester.PlayErrorImprovment$MyTopLevel.<init>(PlayDev.scala:1162)
    spinal.tester.PlayErrorImprovment$$anonfun$72.apply(PlayDev.scala:1168)
    spinal.tester.PlayErrorImprovment$$anonfun$72.apply(PlayDev.scala:1168)
    spinal.tester.PlayErrorImprovment$.delayedEndpoint$spinal$tester$PlayErrorImprovment$1(PlayDev.scala:1168)
    spinal.tester.PlayErrorImprovment$delayedInit$body.apply(PlayDev.scala:1161)
    spinal.tester.PlayErrorImprovment$.main(PlayDev.scala:1161)
    spinal.tester.PlayErrorImprovment.main(PlayDev.scala)
```

tomverbeure, 2018-08-24:
This will be much better. I'll recreate my initial error to verify.

Tom

tomverbeure, 2018-08-30:
I tried with the latest `dev` branch and the error is now fine. Closing.

commit 56c219d5 (2018-08-24): Fix #141
closed 2018-08-30 by tomverbeure