SpinalHDL/SpinalHDL #143: RFE: Option to generate ascii version of an enum

tomverbeure, 2018-09-10:
When you have an Enum and you're using Verilog, there is no way to see the textual value of the Enum in the waveforms.

There should be an option to generate an additional helper signal that has the ascii version of the signal.

Something like this:
```scala
object MyState extends SpinalEnum(binaryOneHot) {
    val Idle = newElement()
    val Active = newElement()
}
...
   val cur_state = MyState().decodeAsText
```

Should generate something like this:
```verilog
`define MyState_binary_one_hot_Idle 4'b0001
...
reg [3:0] cur_state;
...
`ifndef SYNTHESIS
reg [0:255]  cur_state_txt;

case(1){
   ((cur_state & MyState_binary_one_hot_Idle) != 4'b0000): cur_state_txt = "Idle";
   ...
}
```
You can then add cur_state_txt to `gtkwave` and use the ASCII format to display the signal as "Idle" etc.

asicdruide, 2018-09-10:
I'd like to support the request (having symbolic waves) but would like to propose a different implementation. 
Use the enums defined with SystemVerilog. They show-up in the waves as symbolically.

tomverbeure, 2018-09-10:
While we're talking about how to do things best: right now, these kind of states are created with `define. 

The cleaner way of doing this is to use `localparam` instead.
So instead of this:
```verilog
`define PcState_defaultEncoding_type [4:0]
`define PcState_defaultEncoding_Idle 5'b00001
`define PcState_defaultEncoding_WaitReqReady 5'b00010
`define PcState_defaultEncoding_WaitRsp 5'b00100
`define PcState_defaultEncoding_WaitJumpDone 5'b01000
`define PcState_defaultEncoding_WaitStallDone 5'b10000
```
You do this:
```
    localparam [4:0] PcState_Idle = 5'b00001;
...
```
The only disadvantage of using localparam is it may not be supported by older tools, so this is definitely not a high priority item.

zhutmost, 2018-11-23:
@asicdruide Using `enum` keyword in SystemVerilog to implement Enum is a good idea, which is well supported by most EDA tools.

Dolu1990, 2019-01-11:
Going to implement it the most portable way ( using signal to carry the string name + ifdef SYNTHESIS)
Is that good enough ?
ref #178

tomverbeure, 2019-01-11:
Should be good enough!

Don't worry, we'll review the change. ;-)

Tom

Dolu1990, 2019-01-11:
Example of the actual generation : 

```
  `ifndef SYNTHESIS
  reg [23:0] io_configFrame_stop_string;
  reg [31:0] io_configFrame_parity_string;
  reg [47:0] stateMachine_state_string;
  `endif

  `ifndef SYNTHESIS
  always @(*) begin
    case(io_configFrame_stop)
      `UartStopType_defaultEncoding_ONE : io_configFrame_stop_string = "ONE";
      `UartStopType_defaultEncoding_TWO : io_configFrame_stop_string = "TWO";
      default : io_configFrame_stop_string = "  ?";
    endcase
  end
  always @(*) begin
    case(io_configFrame_parity)
      `UartParityType_defaultEncoding_NONE : io_configFrame_parity_string = "NONE";
      `UartParityType_defaultEncoding_EVEN : io_configFrame_parity_string = "EVEN";
      `UartParityType_defaultEncoding_ODD : io_configFrame_parity_string = " ODD";
      default : io_configFrame_parity_string = "   ?";
    endcase
  end
  always @(*) begin
    case(stateMachine_state)
      `UartCtrlTxState_defaultEncoding_IDLE : stateMachine_state_string = "  IDLE";
      `UartCtrlTxState_defaultEncoding_START : stateMachine_state_string = " START";
      `UartCtrlTxState_defaultEncoding_DATA : stateMachine_state_string = "  DATA";
      `UartCtrlTxState_defaultEncoding_PARITY : stateMachine_state_string = "PARITY";
      `UartCtrlTxState_defaultEncoding_STOP : stateMachine_state_string = "  STOP";
      default : stateMachine_state_string = "     ?";
    endcase
  end
  `endif
```

tomverbeure, 2019-01-11:
Looks good to me!

Tom

closed 2019-06-28 by Dolu1990