SpinalHDL/SpinalHDL #148: Verilog generated in SpinalTemplateSbt

dvc94ch, 2018-09-16:
```verilog
module MyTopLevel (
      input   io_cond0,
      input   io_cond1,
      output  io_flag,
      output [7:0] io_state,
      input   clk,
      input   reset);
  reg [7:0] counter;
  assign io_state = counter;
  assign io_flag = ((counter == (8'b00000000)) || io_cond1);
  always @ (posedge clk or posedge reset) begin
    if (reset) begin
      counter <= (8'b00000000);
    end else begin
      if(io_cond0)begin
        counter <= (counter + (8'b00000001));
      end
    end
  end

endmodule
```

According to clifford wolf the correct way to reset a register is

```verilog
always @(posedge clk) begin
        [non-reset code here]

        if (reset) begin
            [reset code here]
        end
    end
```

and see [here][1] for why those are not equivalent and why the case above can lead to unnecessary logic.

I could not find a way to disable the reset generation, from a quick lock at the spinal hdl src. When targeting an fpga using an initial block to initialize values is IMO preferable.

After trying a bunch of other verilog generators, spinal seems the most promising so far.

- [0] https://twitter.com/oe1cxw/status/1040917788762812416
- [1] http://olofkindgren.blogspot.com/2017/11/resetting-reset-handling.html

Dolu1990, 2018-09-16:
All of your proposals are already supported by SpinalHDL, see  http://spinalhdl.github.io/SpinalDoc/spinal/core/clock_domain/#configuration

```
ResetKind => ASYNC, SYNC, BOOT which is supported by some FPGA (FF values loaded by the bitstream)
```

The SpinalHDL default reset kind is asyncronus
You can define your own defualt SpinalConfig and then using it to generate RTL :

```scala
object MySpinalConfig extends SpinalConfig(defaultConfigForClockDomains = ClockDomainConfig(resetKind = SYNC))

object MyTopLevelVerilogWithCustomConfig {
  def main(args: Array[String]) {
    MySpinalConfig.generateVerilog(new MyTopLevel)
  }
}
```

Dolu1990, 2018-09-16:
This example is now added on the bottom of the SpinalTemplateSbt example

dvc94ch, 2018-09-16:
I see thanks! It looks like the problem can't be triggered with spinal because it generates a separate always block for each reg. I searched the verilog gen and the source for the `initial` keyword. Directly assigning the reg works too.

closed 2018-09-16 by dvc94ch