SpinalHDL/SpinalHDL #154: Scala 2.12.7: error during elaboration (dslScope stack is not empty)

dronox, 2018-10-07:
When compiling the template SBT project (SpinalTemplateSbt) for Scala 2.12.6, SpinalHDL 2.12_1.2.1 and SBT 1.2.3, everything is working fine (build.sbt and project/build.properties updated). But when changing the Scala version to 2.12.7, running **MyTopLevelVerilog** leads to an Exception during elaboration:

`[error] Exception in thread "main" spinal.core.SpinalExit: 
[error]  dslScope stack is not empty :(
[error] ********************************************************************************
[error] ********************************************************************************
[error] Design's errors are listed above.
[error] SpinalHDL compiler exit stack : 
[error]         at spinal.core.SpinalExit$.apply(Misc.scala:366)
[error]         at spinal.core.SpinalError$.apply(Misc.scala:421)
[error]         at spinal.core.internals.PhaseContext.checkGlobalData(Phase.scala:183)
[error]         at spinal.core.internals.PhaseCreateComponent.impl(Phase.scala:1902)
[error]         at spinal.core.internals.PhaseContext.doPhase(Phase.scala:194)
[error]         at spinal.core.internals.SpinalVerilogBoot$.$anonfun$singleShot$10(Phase.scala:2142)
[error]         at spinal.core.internals.SpinalVerilogBoot$.$anonfun$singleShot$10$adapted(Phase.scala:2140)
[error]         at scala.collection.mutable.ResizableArray.foreach(ResizableArray.scala:58)
[error]         at scala.collection.mutable.ResizableArray.foreach$(ResizableArray.scala:51)
[error]         at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:47)
[error]         at spinal.core.internals.SpinalVerilogBoot$.singleShot(Phase.scala:2140)
[error]         at spinal.core.internals.SpinalVerilogBoot$.apply(Phase.scala:2076)
[error]         at spinal.core.Spinal$.apply(Spinal.scala:287)
[error]         at spinal.core.SpinalConfig.generate(Spinal.scala:134)
[error]         at spinal.core.SpinalVerilog$.apply(Spinal.scala:305)
[error]         at mylib.MyTopLevelVerilog$.main(MyTopLevel.scala:47)
[error]         at mylib.MyTopLevelVerilog.main(MyTopLevel.scala)
[error] Nonzero exit code returned from runner: 1
[error] (Compile / runMain) Nonzero exit code returned from runner: 1
[error] Total time: 6 s, completed 07.10.2018, 11:33:59
`

This issue seems to affect Scala 2.12.7 only.

Dolu1990, 2018-10-08:
Hi,

I successfuly reproduced the same issue.
It is very likely an issue with the Scala DelayedInit support. I have to explore it more :)

Dolu1990, 2018-10-08:
Fixed in the scala_2.12 branch with 7ab2e872849d5778276b71428ba4f4b21ae8ea10
Do you know the sbt publish-local stuff to use it ?

dronox, 2018-10-08:
You are really fast, thanks a lot! :)

Yes, I am familiar with that.

dronox, 2018-10-08:
Tested it with several designs, works like a charm. Therefore, I can confirm that this bug is fixed - thank you!

closed 2018-10-08 by Dolu1990