SpinalHDL/SpinalHDL #156: No used logic is not pruning

postoroniy, 2018-10-30:
1) try this val v_B1 = Vec(SFix(0 exp,8 bits),4)
2) somewhere  in your test modules and not use it anyhow
3)check generated verilog 
there are not pruned wires such as 
  wire [7:0] v_B1_0;
  wire [7:0] v_B1_1;
  wire [7:0] v_B1_2;
  wire [7:0] v_B1_3;

Dolu1990, 2018-10-31:
You mean they shoudn't be in the verilog file ?

Dolu1990, 2018-11-27:
So, the reason why SpinalHDL keep all named signals, is becaus they can be usefull for simulation purpose, for instance, a pipeline where you have some context signals going on it's side.

The pruned signal report from SpinalHDL is only a hint which could help the user to do the diagnostic of a broken design.

Does that make sense ?

postoroniy, 2018-12-02:
Case 1.
So when you have a design  with inputs and nothing drive outputs  it leads error in SpinalHDL, right?
or even one single output is not driven  =  error.
Case 2.
Now, mentioned above signals are not used in design anywhere but design has inputs and drives all outputs. Result of rtl generation is not pruned wires ..because they are useful?? Then what was wrong in case 1)

Do you see my point?

tomverbeure, 2018-12-02:
case 1: correct. That's totally an error, and SpinalHDL should complain about it. After all, what's the alternative? Connecting those outputs to 0 would be wrong and connecting to 1 would also be wrong.

case 2: the wires are correct. There is nothing inherently wrong about the design. The only issue is that some signals may not be used. It's a completely different situation than case 1.

But those not-used signals could be helper signals for easier debugging.

Either way, the synthesis tool will prune them for you, so what's the big deal?

postoroniy, 2018-12-02:
Consistent approach is the big deal
but yeah..whatever

closed 2018-12-02 by postoroniy

tomverbeure, 2018-12-02:
> Consistent approach is the big deal

Agreed! Luckily, there's nothing inconsistent about the current behavior. ;-)

Dolu1990, 2018-12-02:
So, initialy, SpinalHDL was removing everything useless for the toplevel outputs behaviour. It was mainly the result of how the netlist was stored into the PC memory. 
There is some argument against pruning signals : 

- It is surprising for the user, because some signal that you define can desapear from the netlist
- It isn't friendly with debug utilities signals 
- Can make issues when you are implementing a lonnnng pipeline, and you want to do some simulation without having finishing the implementation, as the whole pipeline part which isn't connected yet to the output will be removed. (the same if you forget to interconnect two parts of the pipeline)