SpinalHDL/SpinalHDL #157: Intermmitent verilator error

wifasoi, 2018-11-07:
I'm using:
- SpinalHDL bbe3df4d4d896727b7659fa4d70e2214da4fcb1c (dev branch)
- Verilator 4.004 2018-10-06 rev UNKNOWN_REV
- GNU Make 4.2.1

I found this intermittent problem (this could be a pebcac, can someone reproduce this?):
sometimes `make` will throw this error at me (when using verilator and spinalSim), and fail to create a trace. But if i re run it ( I use `~;run` in the sbt terminal, and save the file without modification), it will run just fine. 
the output log [here](https://pastebin.com/F77BNB7z)
the code here:
```scala
import spinal.core._
import spinal.lib._

import spinal.sim._
import spinal.core._
import spinal.core.sim._
import scala.util.Random
import scala.math._
import spinal.core.internals._
import spinal.core.GenerationFlags._
import spinal.core.Formal._

class rle[T <: BaseType](val dataType: HardType[T], val depth: Int) extends Component{
  val io = new Bundle{
    val input = in(dataType())
    val output = out(dataType())
    val lenght = out(Bits(depth bits))
    val valid = out(Bool)
  }

  io.valid := False

  val input = RegNext(io.input)
  val isNew = io.input =/= input
  val count = Reg(UInt(depth bits)) init(0)
  io.lenght := count.asBits
  io.output := input

  when(isNew){
    count := 0
    io.valid := True
  } otherwise {
    count := count + 1
  }

  /////////////////////////////
  // FORMAL
  /////////////////////////////
  //GenerationFlags.formal{
  //  assume(io.input.asUInt < pow(2,size).toInt)
  //  assert(io.output.asUInt >= pow(2,depth).toInt,"")

  //  assert(changed(io.input) || isNew)
  //  }

}

object DutTests {
  def main(args: Array[String]): Unit = {
    //SpinalConfig().includeFormal.generateSystemVerilog(new rle(8,10))
    SimConfig.withWave.compile(new rle(Bits(8 bits),10)).doSim{ dut =>
      dut.clockDomain.forkStimulus(period = 10)
      dut.clockDomain.waitSampling(10)

      Suspendable.repeat(10){
        dut.io.input #= Random.nextInt(256)
        dut.clockDomain.waitSampling(Random.nextInt(100))
      }
    }
  }
}
```

wifasoi, 2018-11-11:
it could be, the 4.0 update was a big one (multi-threading support.. hurray!). I think only I can troubleshoot this. Can you give me some pointer where to look?

Dolu1990, 2018-11-11:
hmm the first thing to do is to identify if that's related to the new verilator version. I have to give some try :D

zeroday-engineer, 2018-11-15:
@wifasoi I tested the code you posted with Verilator version 4.006 and got it to work... I did have to remove two import lines though:
`import spinal.core.GenerationFlags._
import spinal.core.Formal._`

wifasoi, 2018-11-15:
@zeroday-engineer maibe you are not using bbe3df4d4d896727b7659fa4d70e2214da4fcb1c but it's a start. Can you give me the commit share so I can bysect the problem? (Unfortunately I can't do it this week, I'm away from my computer... And forgot the shuco adapter for my laptop... Silly me)

Dolu1990, 2018-12-15:
I just tested now (sorry that's late XD) with verilator 4.009 devel with the following main : 
```
  def main(args: Array[String]): Unit = {
    //SpinalConfig().includeFormal.generateSystemVerilog(new rle(8,10))
    for(i <- 0 to 100) {
      FileUtils.deleteDirectory(new File("simWorkspace"))
      SimConfig.withWave.compile(new rle(Bits(8 bits), 10)).doSim { dut =>
        dut.clockDomain.forkStimulus(period = 10)
        dut.clockDomain.waitSampling(10)

        Suspendable.repeat(10) {
          dut.io.input #= Random.nextInt(256)
          dut.clockDomain.waitSampling(Random.nextInt(100))
        }
      }
      val vcd = new File(s"simWorkspace/rle${if(i > 0) "_" + i else ""}/test.vcd")
      assert(vcd.exists())
      assert(vcd.length() > 10000)
    }
  }
```

And it pass.

Can you try with the same main ? to see if it catch your issue ?

Dolu1990, 2018-12-15:
Anyway, i have seen that there is a new file format suported to replace VCD. I will try to add it's suport :)

wifasoi, 2018-12-15:
ok, on bbe3df4d4d896727b7659fa4d70e2214da4fcb1c works (even with my unchanged code, and @Dolu1990 ).. we got an [Heisenbug](https://en.wikipedia.org/wiki/Heisenbug) over here. I updated my computer... maybe is just archlinux being strange. I think is not wort find the issue, if there was any from the start. I'll close this issue (and make some sacrifice to the gods of programming)

closed 2018-12-15 by wifasoi