SpinalHDL/SpinalHDL #159: Documentation update request

zhutmost, 2018-11-22:
Thanks for your cool work. 
The SpinalHDL wiki is stuck at v1.1.1. Maybe we need to keep it updated to the latest version?

zhutmost, 2018-11-22:
And is there a roadmap of the future development? I am interested in this project.

wifasoi, 2018-11-22:
I can't answer for the roadmap, but i can answer (partially)  for the documentation.
I think that version number is not updated atm (i think because @Dolu1990 update that number manually, every "major" feature update). The documentation is updated constantly... but yes, the documentation is not complete atm(for example... I discovered there is a in hardware scope in the Spinalhdl libraries... but is not documented), but it's a start.
P.S: I'm porting the documentation from jekyll to sphinx (you can check it out [here](https://wifasoi.github.io/SpinalDoc-RTD/html/index.html)) (is not official... yet)

zhutmost, 2018-11-23:
@wifasoi Thanks a lot.

Dolu1990, 2018-11-23:
@zhutmost 
The road map is mainly to improve the SpinalHDL lib and fill it with more useful things. 
Basicaly, the very most of the SpinalHDL core is done and as far i know bug free, sometime there improvement and new features added. For instance, yesterday i released the 1.2.2, i need to write down the tag and the patch note now ^^
The developpement is open :)

About the doc, wifasoi is right ^^

zhutmost, 2019-01-05:
@Dolu1990 
Can we add more comments into the source code? It takes much time to read the code to understand the function of methods. Much of them which dont have comments are not included in the documentation wiki.

Dolu1990, 2019-01-05:
Which part of the code ? the lib, the core or both ? Sure there is missing comments.

wifasoi, 2019-01-05:
Yes, the documentation is not complete, and yes some the SpinalHDL libs lacks in docstring comments in some area. I'll try to add some slowly over time. https://github.com/SpinalHDL/SpinalDoc-RTD/issues/6 we're trying (or at-least is the first step) :)
In the mean time you can ask question on [gitter](https://gitter.im/SpinalHDL/SpinalHDL?utm_source=badge&utm_medium=badge&utm_campaign=pr-badge&utm_content=badge) if you're getting stuck

closed 2020-02-05 by Dolu1990