SpinalHDL/SpinalHDL #160: Create equivalent to readmemb, readmemh, etc to load contents of a Mem

tomverbeure, 2018-11-22:
initialContents is not sufficient: it still requires way too much knowledge of Scala to simply load a binary file.

zhutmost, 2018-11-23:
I think combining spinalHDL with Scala to construct hardware is great. It is not a must to do everything in native SpinalHDL.

For your question, you can read the file with `FileInputStream`.
Here is an example.
```Scala
val file = new File(filename)
val in = new FileInputSteam(file)
val bytes = new Array[Byte](file.length.toInt)
in.read(bytes)
in.close()
```

tomverbeure, 2018-11-23:
> I think combining spinalHDL with Scala to construct hardware is great. It is not a must to do everything in native SpinalHDL.

See my discussion with Dolu about this on gitter.

It took me ages to get it working. The question to ask is: are you targeting Scala experts who want to start doing RTL, or are you targetting Verilog experts who want something better?

> ```scala
> val file = new File(filename)
> val in = new FileInputSteam(file)
> val bytes = new Array[Byte](file.length.toInt)
> in.read(bytes)
> in.close()
> ```

You're simply reading in bytes. Add code to take care of the fact that Java treats bytes as signed, or that there's no support for unsigned integers (important when dealing with 32 bits) etc etc and you have the prefect situation where somebody might throw in the towel on SpinalHDL and say: "this is way more complicated than it needs to be. In verilog, I just do $readmemb. So let's revert to that."

Scala is a very rich language, with a very steep learning curve. You have ease people into it.

Tom

zhutmost, 2018-11-23:
I guess you need a text file in binary encoding instead of a binary file? 
A temporary solution is 
```Scala
Source.fromFile("abc.txt" ).foreach{ 
  .. // convert String to bits
}
```

Abt your comment on the relationship between Scala and SpinalHDL, I think you are right. 

A practical point is that `$readmemh` is used with simulation timesteps. for eg, `initial $readmemh(xxx, xxx);` or `always @(some trigger from UVM) $readmemh(xxx, xxx);`.

Dolu1990, 2018-11-27:
Ok so, my plan is to run all the regression of VexRiscv dev, merge it back to master, then bring things from VexRiscv dev to SpinalHDL lib.

Dolu1990, 2018-11-30:
@tomverbeure 
I moved some stuff from VexRiscv into SpinalHDL.lib, including HexTools : 
dea51faa86146139b40fd8125104d2dea7e2ca75

So, pull requests are welcome :D

ZiCog, 2019-03-29:
I have a memory initialization problem that is somewhat related to this issue so I thought I'd post it here for now.

I want a 32 bit wide RAM with separate byte write enables and initialized with some boot code. So I define my RAM with an initial RAM file attribute like so:

    val memory = Mem (SInt (32 bits), 1024)
    memory.addAttribute ("ram_init_file", "memory.mif")

I write to it using byte enables like so:

    memory.write(
        io.mem_addr,
        io.mem_wdata,
        io.enable & io.mem_valid,
        io.mem_wstrb.asBits
    )

This results in my memory looking like this in Verilog:

    (* ram_init_file = "memory.mif" *) reg [7:0] memory_symbol0 [0:1023];
    (* ram_init_file = "memory.mif" *) reg [7:0] memory_symbol1 [0:1023];
    (* ram_init_file = "memory.mif" *) reg [7:0] memory_symbol2 [0:1023];
    (* ram_init_file = "memory.mif" *) reg [7:0] memory_symbol3 [0:1023];

The issue is of course that the same file is used to initialize all four bytes position of my RAM.

Alternatively I can initialize with a generator function:

    memory.init (generateRamContent)

but the the Verilog then uses an initial section and $readmemb which is not going to work in Quartus:

    initial begin
        $readmemb("Ram32.v_toplevel_memory_symbol0.bin",memory_symbol0);
        $readmemb("Ram32.v_toplevel_memory_symbol1.bin",memory_symbol1);
        $readmemb("Ram32.v_toplevel_memory_symbol2.bin",memory_symbol2);
        $readmemb("Ram32.v_toplevel_memory_symbol3.bin",memory_symbol3);
    end

Am I going about this in entirely the wrong way?

tomverbeure, 2019-03-29:
Quartus definitely supports $readmemh. Are you sure it doesn't support $readmemb as well?

Tom

ZiCog, 2019-03-30:
Well blow me down, $readmemb does indeed synthesize in Quartus and puts it 4 nice byte wide SYNC_RAMs for me.

Not sure if it actually works as a replacement for my Quartus megafunction RAM yet, but it looks like it will.

I'm a neophyte tinkerer with FPGA/Verilog/Spinal/Scala, I tried this kind of thing a year ago with no luck. 

Thanks.

ZiCog, 2019-03-30:
@sergei-romanenko

What do I need to import to get your readmemh to compile?

I have:

    import scala.collection.mutable.ArrayBuffer
    import scala.io.Source

Which is what a google search told me I need. But it fails with:

    ....reference to Source is ambiguous;
    [error] it is imported twice in the same scope by
    [error] import scala.io.Source
    [error] and import javax.xml.transform.Source
    [error]     for (line <- Source.fromFile(path).getLines) {
    [error]                  ^
    [error] one error found
    ....

Or... anyone have an example of how to use the initRam that Dolu1990 refers to above?

Sorry for the newbie questions but the combination of Scala, SBT, IntelliJ and Quartus is fraying my nerves.

sergei-romanenko, 2019-03-30:
On Sat, Mar 30, 2019 at 5:08 PM zicog <notifications@github.com> wrote:

> @sergei-romanenko <https://github.com/sergei-romanenko>
>
> What do I need to import to get your readmemh to compile?
>
> I have:
>
> import scala.collection.mutable.ArrayBuffer
> import scala.io.Source
>
> Try putting the stuff into a separate file as in
https://github.com/sergei-romanenko/spinalhdl-samples/blob/master/src/main/scala/shs/lib/Tools.scala

ZiCog, 2019-03-30:
Actually, Sergi, I did eventually get that readmemh to compile and run. But it did not generate the correct files used in the Verilog readmemb statements. But I adapted it a bit and ...

Yay, I finally ended up with something that initializes my 32 bit wide memory with 4 byte write strobes from a single hex file correctly. Seems to make Quartus happy as well. Four Sync RAM blocks and all.

    class Memory (width : Int, depth : Int, initFile: String) extends Component {
        ...
        ...
        def generateInitialContent = {
            val buffer = new ArrayBuffer[BigInt]
            for (line <- Source.fromFile(initFile).getLines) {
                val tokens: Array[String] = line.split("(//)").map(_.trim)
                if (tokens.length > 0 && tokens(0) != "") {
                    val i = Integer.parseInt(tokens(0), 16)
                    buffer.append(i)
                }
            }

            for(address <- 0 until depth) yield {
                S(buffer(address), width bits)
            }
        }

        val memory = Mem (SInt (width bits), depth)
        memory.init (generateInitialContent)

        memory.write(
            io.mem_addr,
            io.mem_wdata,
            io.enable & io.mem_valid,
            io.mem_wstrb.asBits
        )
        ...
        ...
    }

Thanks all. Let me know if there is a better way to do this.

saahm, 2020-04-02:
Hello, is there already a recommended way of initializing memory with content? As currently I follow the  proposed by @sergei-romanenko with small adaptions to how i generate my hexdumps.

numero-744, 2022-11-22:
Hi! What is the state of this issue?

Dolu1990, 2022-11-23:
There is spinal.lib.misc.BinTools which can load a mem from a binary file, but nothing else i think ?