SpinalHDL/SpinalHDL #161: Feature Proposal: Allow using `Bits` to construct `SpinalEnumEncoding`

zhutmost, 2018-11-23:
Currently, the type of Enum Encoding values is `BigInt`. I think it better to use `Bits`? Or I have to convert a lot of hex numbers to `BigInt` manually.

```Scala
object Mnemonic extends SpinalEnum{
  val BRK, ORA, COP = newElement()
  defaultEncoding = SpinalEnumEncoding("staticEncoding")(
    BRK -> B"x00", // type error here
    ORA -> B"x01",
    COP -> B"x02"
  )
}
```

Error message from IDEA: `Type mismatch, expected: (SpinalEnumElement[NotInferedX], BigInt), actual: (SpinalEnumElement[Mnemonic.this.type], Bits)`

zhutmost, 2018-11-23:
Here is a relevant issue. -> #143

dronox, 2018-11-23:
Ah sorry, I meant `BigInt("f", 16)`.

Dolu1990, 2018-11-23:
So, in general, i would say it's not the best to use Hardware literals to do elaboration time stuff, i realy try to stick with the scala world for everything which isn't infering directly hardware.

Just to understand what happen behind the scene when you  define 

```scala
object Mnemonic extends SpinalEnum{
  val BRK, ORA, COP = newElement()
  defaultEncoding = SpinalEnumEncoding("staticEncoding")(
    BRK -> B"x00", // type error here
    ORA -> B"x01",
    COP -> B"x02"
  )
}
```

So basicaly each B"xxx" will add to the netlist a signal which is driven by the corresponding literal expression. Then durring the https://github.com/SpinalHDL/SpinalHDL/blob/3159d9865a8de00378e0b0405c338a97c2f5a601/core/src/main/scala/spinal/core/internals/Phase.scala#L2114 phase, all signals which have no name and which are useless to the hardware behaviour are removed, so it will remove from the netlist those 3 useless parasitic signals at that time, which is good.

But anyway, in the present case, it could make the syntax more natural to have your proposal, i agree.

Dolu1990, 2022-10-27:
I think String is better yes, Using things as Bits literal has some side effect (it add stuff unecessarly to the netlist durring elaboration)

numero-744, 2022-11-21:
This might be the function to use: https://github.com/SpinalHDL/SpinalHDL/blob/c77cbe95ceefa39c63fb4543f0848b803acb566f/core/src/main/scala/spinal/core/core.scala#L305

numero-744, 2022-11-25:
Hmm I can't do that because of "same type after erasure".

```scala
  def apply[X <: SpinalEnum](name: String)(
    first: (SpinalEnumElement[X], BigInt),
    spec: (SpinalEnumElement[X], BigInt)*
    ): SpinalEnumEncoding = ???

  def apply[X <: SpinalEnum](name: String)(
      first: (SpinalEnumElement[X], String),
      spec: (SpinalEnumElement[X], String)*
  ): SpinalEnumEncoding = ???
```

I think I also can't make an implicit conversion.

Is it okay if I add something like `I"thestring"` and/or `I("thestring")`?

numero-744, 2022-11-25:
Ah, this compiles but I can't mix BigInt and String. Is it okay?

```scala
def apply[X <: SpinalEnum](name: String)(
    spec: (SpinalEnumElement[X], BigInt)*
    ): SpinalEnumEncoding = ???

  def apply[X <: SpinalEnum](name: String)(
      spec: (SpinalEnumElement[X], String)*
  )(implicit dummy1: DummyImplicit): SpinalEnumEncoding = ???
```

Dolu1990, 2022-11-28:
So, I"xxx" to get a BigInt ?
or should it be I for Int, and BI for BigInt ? That maybe safer i think.

> but I can't mix BigInt and String

What do you mean ?

numero-744, 2022-11-28:
> So, I"xxx" to get a BigInt ?

Maybe keeping just `BI`, would be the only one with two letters, also the only one which does not produce a hardware signal?

---

By "can't mix" I mean I cannot do something like:

```scala
.setEncoding(
  A -> 10,
  B -> "o10",
  C -> "x10"
)
```

where I use in the same function call both "BigInt" and "String". But maybe it is not an issue, just tell me.

numero-744, 2022-11-28:
As it is okay to not support mixing syntaxes, maybe it is better to not implement `BI""`? I'm not sure it is an interesting feature actually… I'll PR with just "accepting `String` in `.setEncoding`", and if someone really needs a `BI` prefix another issue can be opened.

numero-744, 2022-11-28:
Err, bad news, `DummyImplicit` removes compilation error of the function definition but leaves errors on function calls… So maybe we don't have other choice than defining `BI`. So it would look like:

```scala
.setEncoding(
  A -> BI"b10",
  B -> BI"o10",
  C -> BI"x10"
)
```

In this case how to manage signed-ness? `BN` (Big Natural) for unsigned?

Dolu1990, 2022-11-29:
Big Natural ? i don't get it.  why not BI everywhere ?

numero-744, 2022-11-29:
Yes big natural.

There are two items because there is a difference between signed and unsigned in the initial algorithm: https://github.com/SpinalHDL/SpinalHDL/blob/0aadf22d77dd3c61d012e14d0bfbfb79f484c7d5/core/src/main/scala/spinal/core/core.scala#L348 but maybe we can just always use one of them?

Dolu1990, 2022-11-29:
Ahhh i see what you mean.
What's about

BS / BU instead then ?

numero-744, 2022-11-29:
Ok I'll update the PR

numero-744, 2022-11-29:
@Dolu1990 if you confirm "adding BS/BU string prefix" is the change set, I can start documenting that

Btw PR updated

Dolu1990, 2022-11-30:
Good to me ^^

PR #981 (2022-12-01): feat: BU and BS string prefixes
Closes #161

# Context, Motivation & Description

This PR splits `bitVectorStringParser` in two: the logic part processing data, and the part creating the actual `BitVector`. It uses the first part on its own to add `BU""` and `BS""`

# Checklist

- [ ] Unit tests were added
- [x] API changes are or will be documented:
  - using Scaladoc comments: `/** */`?
  - on [RTD](https://github.com/SpinalHDL/SpinalDoc-RTD)?
  - thanks to a [new tracking issue on RTD](https://github.com/SpinalHDL/SpinalDoc-RTD/issues/new?title=Document%20XXX&body=Do%20not%20merge%20until%2 [... cut]
closed 2022-12-01 by numero-744