SpinalHDL/SpinalHDL #163: register structure/union

zhutmost, 2018-11-25:
As described in the Wiki, Bundle can be used to implement `structure` with `AnyType`. so I tried a register bundle named `regBank` as follows.
```Scala
package w65c816s

import spinal.core._

class Cpu extends Component{
  val io = new Bundle{
    val addr          = out Bits(24 bits)
    val dataIn        = in  Bits(8 bits)
    val dataOut       = out Bits(8 bits)
    val ready         = in  Bool
//    val emulation     = out Bool
//    val memLockB      = out Bool
//    val memSelect     = out Bool
//    val validDataAddr = out Bool
//    val validProgAddr = out Bool
//    val vecPullB      = out Bool
//    val intReq        = in  Bool
//    val abortB        = in  Bool
  }

  val regBank = new Bundle{
    val X = Reg(UInt(2 bits))
    val Y = Reg(UInt(3 bits))
  }

  when(io.ready === False){
    regBank.X := io.dataIn.asUInt
    regBank.Y := regBank.X + 3
  }

  io.dataOut := regBank.Y.asBits

}

//Generate the MyTopLevel's Verilog
object CpuVerilog {
  def main(args: Array[String]) {
    SpinalVerilog(new Cpu)
  }
}
```
The elaboration fails when I want to generate Verilog HDL. What should I do to fix it?
Thanks. 

Haozhe

Ps. the generated log is attached. 
```log
/home/zhutmost/Software/java/openjdk-11/bin/java -javaagent:/home/hdd/Software/JetBrains/Toolbox/apps/IDEA-U/ch-0/183.4284.148/lib/idea_rt.jar=43793:/home/hdd/Software/JetBrains/Toolbox/apps/IDEA-U/ch-0/183.4284.148/bin -Dfile.encoding=UTF-8 -classpath /home/hdd/Workspace/GitHub/opensnes/target/scala-2.12/classes:/home/zhutmost/.ivy2/cache/com.github.scopt/scopt_2.12/jars/scopt_2.12-3.7.0.jar:/home/zhutmost/.ivy2/cache/org.scala-lang.plugins/scala-continuations-library_2.12/bundles/scala-continuations-library_2.12-1.0.3.jar:/home/zhutmost/.sbt/boot/scala-2.12.7/lib/scala-reflect.jar:/home/zhutmost/.sbt/boot/scala-2.12.7/lib/scala-library.jar:/home/zhutmost/.ivy2/cache/commons-io/commons-io/jars/commons-io-2.4.jar:/home/zhutmost/.ivy2/cache/com.github.spinalhdl/spinalhdl-sim_2.12/jars/spinalhdl-sim_2.12-1.2.1.jar:/home/zhutmost/.ivy2/cache/com.github.spinalhdl/spinalhdl-lib_2.12/jars/spinalhdl-lib_2.12-1.2.1.jar:/home/zhutmost/.ivy2/cache/com.github.spinalhdl/spinalhdl-core_2.12/jars/spinalhdl-core_2.12-1.2.1.jar w65c816s.CpuVerilog
[Runtime] SpinalHDL v1.2.1    git head : 91006ba5f7ae3f3b70abb93182909e7e11eca3d5
[Runtime] JVM max memory : 5994.0MiB
[Runtime] Current date : 2018.11.25 18:21:30
[Progress] at 0.000 : Elaborate components

**********************************************************************************************
[Warning] Elaboration failed (1 error).
          Spinal will restart with scala trace to help you to find the problem.
**********************************************************************************************

[Progress] at 0.088 : Elaborate components
Exception in thread "main" spinal.core.SpinalExit: 

 dslScope stack is not empty :(
********************************************************************************
********************************************************************************

Design's errors are listed above.
SpinalHDL compiler exit stack : 

	at spinal.core.SpinalExit$.apply(Misc.scala:366)
	at spinal.core.SpinalError$.apply(Misc.scala:421)
	at spinal.core.internals.PhaseContext.checkGlobalData(Phase.scala:183)
	at spinal.core.internals.PhaseCreateComponent.impl(Phase.scala:1902)
	at spinal.core.internals.PhaseContext.doPhase(Phase.scala:194)
	at spinal.core.internals.SpinalVerilogBoot$.$anonfun$singleShot$10(Phase.scala:2142)
	at spinal.core.internals.SpinalVerilogBoot$.$anonfun$singleShot$10$adapted(Phase.scala:2140)
	at scala.collection.mutable.ResizableArray.foreach(ResizableArray.scala:58)
	at scala.collection.mutable.ResizableArray.foreach$(ResizableArray.scala:51)
	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:47)
	at spinal.core.internals.SpinalVerilogBoot$.singleShot(Phase.scala:2140)
	at spinal.core.internals.SpinalVerilogBoot$.apply(Phase.scala:2076)
	at spinal.core.Spinal$.apply(Spinal.scala:287)
	at spinal.core.SpinalConfig.generate(Spinal.scala:134)
	at spinal.core.SpinalVerilog$.apply(Spinal.scala:305)
	at w65c816s.CpuVerilog$.main(Cpu.scala:38)
	at w65c816s.CpuVerilog.main(Cpu.scala)

Process finished with exit code 1

```

Dolu1990, 2018-11-25:
I think that's a issue related to Scala 2.12, i remember having fix this issue a time ago
I have to check it again, and release it :)

Dolu1990, 2018-11-25:
Hooo yes i remember now
between something like scala 2.12.4  ans the next version, there was this DelayedInit delta which broke things

dronox, 2018-11-25:
Yep, it was this issue: https://github.com/SpinalHDL/SpinalHDL/issues/154

With Scala 2.12.6 everything was woken and with 2.12.7 there was this dslScope error.

Dolu1990, 2018-11-25:
Fix is 7ab2e872849d5778276b71428ba4f4b21ae8ea10

Dolu1990, 2018-11-25:
Ok
i have to release the 1.2.2 on scala 2.12 anyway ^^

Dolu1990, 2018-11-25:
I'm merging 1.2.2 into the scala 2.12 branch now to do the release

zhutmost, 2018-11-25:
Thx a lot. :P

Dolu1990, 2018-11-25:
It is released now => 1.2.2
Let's me know if you have issues, regressions are passing on scala 2.12, but i'm personnaly on 2.11 ^^

closed 2018-11-25 by Dolu1990