SpinalHDL/SpinalHDL #166: SlowArea doesn't scale frequencies properly

hansihe, 2018-12-17:
## Version
SpinalHDL version: 
```
  "com.github.spinalhdl" % "spinalhdl-core_2.11" % "1.2.2",
  "com.github.spinalhdl" % "spinalhdl-lib_2.11" % "1.2.2"
```
Scala version: `2.11.6`

## Issue
When inside a `SlowArea`, the frequency of `ClockDomain.current` is the same as outside. `clockDomain.readClockWire` appears to be affected as well. Curiously, it seems like the logic itself is unaffected by this issue. Keep in mind that I am not very familiar with SpinalHDL, so I may very well be doing something stupid.

## Reproducing
This is run in the environment of `SpinalTemplateSbt`. Changes are below, and involves instantiating a `SlowArea`, logging frequencies, and actually configuring a clock frequency.

```scala
class MyTopLevel extends Component {
  val io = new Bundle {
    val cond0 = in  Bool
    val cond1 = in  Bool
    val flag  = out Bool
    val state = out UInt(8 bits)
  }
  val counter = Reg(UInt(8 bits)) init(0)

  when(io.cond0){
    counter := counter + 1
  }

  io.state := counter
  io.flag  := (counter === 0) | io.cond1

  println(GlobalData.get.dslClockDomain)
  println(GlobalData.get.dslClockDomain.head.frequency)
  println(clockDomain.frequency)
  println(ClockDomain.current.frequency)
  val a = new SlowArea(8) {
    println(GlobalData.get.dslClockDomain)
    println(GlobalData.get.dslClockDomain.head.frequency)
    println(clockDomain.frequency)
    println(ClockDomain.current.frequency)
  }
  println(GlobalData.get.dslClockDomain)
  println(GlobalData.get.dslClockDomain.head.frequency)
  println(clockDomain.frequency)
  println(ClockDomain.current.frequency)

}

object MySpinalConfig extends SpinalConfig(defaultConfigForClockDomains = ClockDomainConfig(resetKind = SYNC), defaultClockDomainFrequency = FixedFrequency(100 MHz))

object MyTopLevelVerilog {
  def main(args: Array[String]) {
    SpinalVerilog(MySpinalConfig)(new MyTopLevel)
  }
}
```

Output:
```
[Runtime] SpinalHDL v1.2.2    git head : 3159d9865a8de00378e0b0405c338a97c2f5a601
[Runtime] JVM max memory : 3502.5MiB
[Runtime] Current date : 2018.12.17 18:16:48
[Progress] at 0.000 : Elaborate components
Stack(clk)
FixedFrequency(HertzNumber(100000000.0))
FixedFrequency(HertzNumber(100000000.0))
FixedFrequency(HertzNumber(100000000.0))
Stack(clk, clk)
FixedFrequency(HertzNumber(100000000.0))
FixedFrequency(HertzNumber(100000000.0))
FixedFrequency(HertzNumber(100000000.0))
Stack(clk)
FixedFrequency(HertzNumber(100000000.0))
FixedFrequency(HertzNumber(100000000.0))//
FixedFrequency(HertzNumber(100000000.0))
[Progress] at 0.103 : Checks and transforms
[Progress] at 0.158 : Generate Verilog
[Progress] at 0.162 :   emit MyTopLevel
[Info] Number of registers : 8
[Done] at 0.200

Process finished with exit code 0
```

tomverbeure, 2018-12-19:
I'm not an expert either, but I have been playing with this recently: the manual doesn't say it, but a SlowArea creates a clock domain that is slower than its parent by adding a clock enable that is asserted every so many clock cycles.

It does NOT create a new clock.

In other words, FFs in this new clock domain will result in this kind of Verilog:
```
always @(posedge original_fast_clk) begin
    if (slow_clk_ena) begin
        ...
    end
end
```
I initially expected this:
```
always @(posedge slow_clk) begin
    ...
end
```
I suspect that you expected the same thing as well?

Tom

Dolu1990, 2018-12-19:
hmm, but maybe the ClockDomain.curent.frequancy could be change acordingly to the enable rate ?

hansihe, 2018-12-19:
That sounds like a really good solution to me. 

Just scaling the frequency number itself might be misleading, as you might expect a signal with that frequency when calling readClockWire. Having two separate numbers (along with possibly a method on the ClockDomain class to generate the effective frequency?) sounds like the best of both worlds.

wifasoi, 2018-12-19:
@hansihe if you need any help with how to redact the documentation don't hesitate to contact me via gitter: https://gitter.im/SpinalHDL/SpinalHDL

closed 2019-01-11 by Dolu1990