SpinalHDL/SpinalHDL #167: Issue with JTAG - off by 1

cutephoton, 2018-12-17:
I'm testing VecRiscv on an MAX10 and have encountered difficulty with JTAG. (Posting here because the JtagBridge is part of SpinalHDL.) I'm using the avalon variant. OpenOCD produces the following output. Using Versaloon adapter.

```
Open On-Chip Debugger 0.10.0+dev-00308-g94655121 (2018-12-16-16:22)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 100 kHz
adapter_nsrst_delay: 260
jtag_ntrst_delay: 250
Info : set servers polling period to 5ms
START
Info : Versaloon(0x15)by Simon(compiled on Dec 16 2018)
Info : USB_TO_XXX abilities: 0x0000176E:0x010001EF:0xC0000007
Info : clock speed 100 kHz
Info : JTAG tap: fpga_spinal.bridge tap/device found: 0x88000fff (mfg: 0x7ff (<invalid>), part: 0x8000, ver: 0x8)
Warn : JTAG tap: fpga_spinal.bridge       UNEXPECTED: 0x88000fff (mfg: 0x7ff (<invalid>), part: 0x8000, ver: 0x8)
Error: JTAG tap: fpga_spinal.bridge  expected 1 of 1: 0x10001fff (mfg: 0x7ff (<invalid>), part: 0x0001, ver: 0x1)
Error: Trying to use configured scan chain anyway...
Error: fpga_spinal.bridge: IR capture error; saw 0x08 not 0x01
Warn : Bypassing JTAG setup events due to errors
```
Interestingly the ID decoded is.....

```
00010000000000000001111111111111 <-- Expected
10001000000000000000111111111111 <-- detected
```
Off by one error? I used SignalTap to look at the internal JTAG state. It looks like it's working as expected.

The configuration is:
```
interface vsllink

set  _ENDIAN little
set _TAP_TYPE 1234

if { [info exists CPUTAPID] } {
   set _CPUTAPID $CPUTAPID
} else {
  # set useful default
   set _CPUTAPID 0x10001FFF
   #0x10001fff
}

#adapter_khz 4000
adapter_khz 100
transport select jtag
adapter_nsrst_delay 260
jtag_ntrst_delay 250

set _CHIPNAME fpga_spinal
jtag newtap $_CHIPNAME bridge -expected-id $_CPUTAPID -irlen 4 -ircapture 0x1 -irmask 0xF

target create $_CHIPNAME.cpu0 vexriscv -endian $_ENDIAN -chain-position $_CHIPNAME.bridge -coreid 0 -dbgbase 0xF00F0000
vexriscv readWaitCycles 12
vexriscv cpuConfigFile ../cpu0.yaml

poll_period 5
echo "START"
init
```
Thoughts? Anybody had this problem before? I'm getting a new JTAG adapter because I'm not so sure about this Versaloon adapter.

As well, is it possible to use the JTAG interface via the Max10 virtual JTAG & byteblaster?

cutephoton, 2018-12-17:
Back ported FT232R driver to the riscv openocd and was successful. Guess this is a Versaloon issue.

closed 2018-12-17 by cutephoton

tomverbeure, 2018-12-17:
> As well, is it possible to use the JTAG interface via the Max10 virtual JTAG & byteblaster?

I would love to be able to control the VexRiscv through virtual JTAG (or the Xilinx equivalent). The absence of that is essentially preventing me from using VexRiscv with a JTAG debugger.

That said, it's probably quite a bit of effort to make that work in OpenOCD. Through, from the OpenOCD source code, it seems that or1k (Open Risc) was able to do it.

Tom

cutephoton, 2018-12-17:
OpenOCD has some kind of support for the virtual pathway. I saw it in passing so can't say whether it does what I think it does. The part that confused me was figuring out how to connect the virtual jtag IP. It has a variety of signals that I didn't know what to do with.

Dolu1990, 2018-12-18:
So, at a low level, the only requirements to do the JTAG debug is to provide a 32 bits memory read / write via jtag or something else. there is currently some support to debug the target via a TCP socket that do those memory request in the VexRiscv openocd port.