SpinalHDL/SpinalHDL #169: Axi4.incr Alignment?

peteut, 2018-12-19:
`Axi4.burst.INCR` shall align the address [1], A3-49 in `Axi4.incr`, how is this maintain? 
I'm new to _SpinalHDL_ and currently aiming to port some `Axi4` implementation to [migen-axi](https://github.com/peteut/migen-axi). 
I align `base` [here](https://github.com/peteut/migen-axi/commit/8ed145cd40d242a73d7f3231f1c2e1c528bfdf42#diff-342a6b8dd31b6d00b421599becbc6168R257).

[1] [ARM IHI 0022E, A3-48](http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.ihi0022e/index.html)

Dolu1990, 2018-12-19:
Which part of the spinal lib Axi4 are you interrested into ?
Because for instance, the axi interconnect has no data width adapation. it only route things and keep the payload of transactions untouched.

peteut, 2018-12-19:
@Dolu1990: I wonder if `Axi4.burst.INCR` is handled correctly by [Axi4.incr](https://github.com/SpinalHDL/SpinalHDL/blob/369ec039630c441c429b64ffc0a9ec31d21b7196/lib/src/main/scala/spinal/lib/bus/amba4/axi/Axi4.scala#L236) in case of an unaligned access.
In this case the first beat is unaligned but any subsequent beats are aligned to _2 ^ AxSIZE_. 
My understanding is that [result](https://github.com/SpinalHDL/SpinalHDL/blob/369ec039630c441c429b64ffc0a9ec31d21b7196/lib/src/main/scala/spinal/lib/bus/amba4/axi/Axi4.scala#L238) is the address of the next beat.

Dolu1990, 2018-12-21:
So, as a practical case, having  on a 32 data width axi bus an access of 8 beats, size of each beat is 4 bytes, but the address of the access isn't alligned on 4 bytes ?

Then yes, you are right, the incr function doesn't follow the spec, and forget to set the LSB to zero in the case of an access where the address is unaligned on the access size

Thanks :D

peteut, 2018-12-21:
> So, as a practical case, having on a 32 data width axi bus an access of 8 beats, size of each beat is 4 bytes, but the address of the access isn't alligned on 4 bytes ?
>
> Then yes, you are right, the [...]

@Dolu1990 exactly, it's probably an edge case w/o practical relevance but my unit test failed.

closed 2019-01-11 by Dolu1990