SpinalHDL/SpinalHDL #171: Make SpinalHDL compile without warnings

typingArtist, 2018-12-23:
Current compilation attempts on the dev branch lead to some warnings. It’d be cool to get these sorted out before release of 1.3

```text
[warn] There may be incompatibilities among your library dependencies; run 'evicted' to see detailed eviction warnings.
[…]
[warn] bootstrap class path not set in conjunction with -source 1.7
[warn] /home/spinalhdl/SpinalHDL/core/src/main/scala/spinal/core/sim/SimBootstraps.scala:444:5: match may not be exhaustive.
[warn] It would fail on the following inputs: (None, None), (Some(_), Some(_))
[warn]     (_rtlGen, _spinalReport) match {
[warn]     ^
[warn] /home/spinalhdl/SpinalHDL/core/src/main/scala/spinal/core/internals/Modifier.scala:1:1: 
[warn] Found names but no class, trait or object is defined in the compilation unit.
[warn] The incremental compiler cannot record the dependency information in such case.
[warn] Some errors like unused import referring to a non-existent class might not be reported.
[warn]     
[warn] /*                                                                           *\
[warn] ^
[warn] there were 26 deprecation warnings; re-run with -deprecation for details
[warn] three warnings found
[warn] /home/spinalhdl/SpinalHDL/lib/src/main/scala/spinal/lib/bus/simple/PipelinedMemoryBus.scala:243:116: match may not be exhaustive.
[warn] It would fail on the following input: (None, None)
[warn]   def setConnector(bus : PipelinedMemoryBus)( connector : (PipelinedMemoryBus,PipelinedMemoryBus) => Unit): Unit = (masters.get(bus), slaves.get(bus)) match {
[warn]                                                                                                                    ^
[warn] /home/spinalhdl/SpinalHDL/lib/src/main/scala/spinal/lib/bus/simple/PipelinedMemoryBus.scala:248:153: match may not be exhaustive.
[warn] It would fail on the following input: None
[warn]   def setConnector(m : PipelinedMemoryBus, s : PipelinedMemoryBus)(connector : (PipelinedMemoryBus,PipelinedMemoryBus) => Unit): Unit = connections.find(e => e.m == m && e.s == s) match {
[warn]                                                                                                                                                         ^
[warn] /home/spinalhdl/SpinalHDL/lib/src/main/scala/spinal/lib/bus/wishbone/WishboneIntercon.scala:25:86: match may not be exhaustive.
[warn] It would fail on the following input: (None, None)
[warn]   def setConnector(bus : Wishbone)( connector : (Wishbone,Wishbone) => Unit): Unit = (masters.get(bus), slaves.get(bus)) match {
[warn]                                                                                      ^
[warn] /home/spinalhdl/SpinalHDL/lib/src/main/scala/spinal/lib/bus/wishbone/WishboneIntercon.scala:40:113: match may not be exhaustive.
[warn] It would fail on the following input: None
[warn]   def setConnector(m : Wishbone, s : Wishbone)(connector : (Wishbone,Wishbone) => Unit): Unit = connections.find(e => e.m == m && e.s == s) match {
[warn]                                                                                                                 ^
[warn] /home/spinalhdl/SpinalHDL/lib/src/main/scala/spinal/lib/bus/amba4/axi/wip/Axi4WidthAdapter.scala:1:1: 
[warn] Found names but no class, trait or object is defined in the compilation unit.
[warn] The incremental compiler cannot record the dependency information in such case.
[warn] Some errors like unused import referring to a non-existent class might not be reported.
[warn]     
[warn] package spinal.lib.bus.amba4.axi.wip
[warn] ^
[warn] there were 8 deprecation warnings; re-run with -deprecation for details
[warn] there were four feature warnings; re-run with -feature for details
[warn] 7 warnings found
[info] Done compiling.
[info] Compiling 7 Scala sources to /home/spinalhdl/SpinalHDL/debugger/target/scala-2.11/classes ...
[warn] there was one deprecation warning; re-run with -deprecation for details
[warn] one warning found
[info] Done compiling.
[info] Compiling 6 Scala sources to /home/spinalhdl/SpinalHDL/demo/target/scala-2.11/classes ...
[warn] there was one deprecation warning; re-run with -deprecation for details
[warn] one warning found
[info] Done compiling.
[info] Compiling 27 Scala sources and 4 Java sources to /home/spinalhdl/SpinalHDL/tester/target/scala-2.11/classes ...
[warn] /home/spinalhdl/SpinalHDL/tester/src/main/scala/spinal/tester/PlayDev.scala:1092:37: a pure expression does nothing in statement position; you may be omitting necessary parentheses
[warn]           case s : MemReadSync => s.readUnderWrite
[warn]                                     ^
[warn] /home/spinalhdl/SpinalHDL/tester/src/main/scala/spinal/tester/bug/bug.scala:1:1: 
[warn] Found names but no class, trait or object is defined in the compilation unit.
[warn] The incremental compiler cannot record the dependency information in such case.
[warn] Some errors like unused import referring to a non-existent class might not be reported.
[warn]     
[warn] /*
[warn] ^
[warn] there were 8 deprecation warnings; re-run with -deprecation for details
[warn] there was one feature warning; re-run with -feature for details
[warn] four warnings found
[warn] bootstrap class path not set in conjunction with -source 1.7
```

typingArtist, 2018-12-23:
```
$ sbt evicted
[…]
[info] Done updating.
[info] Here are other dependency conflicts that were resolved:
[info] 	* net.java.dev.jna:jna:4.2.2 is selected over 4.2.1
[info] 	    +- com.github.purejavacomm:purejavacomm:1.0.2.RELEASE (depends on 4.2.2)
[info] 	    +- net.java.dev.jna:jna-platform:4.2.1                (depends on 4.2.1)
[info] 	    +- net.openhft:affinity:3.1.11                        (depends on 4.2.1)
[info] Done updating.
[info] Updating tester...
[info] Done updating.
[info] Here are other dependency conflicts that were resolved:
[info] 	* net.java.dev.jna:jna:4.2.2 is selected over 4.2.1
[info] 	    +- com.github.purejavacomm:purejavacomm:1.0.2.RELEASE (depends on 4.2.2)
[info] 	    +- net.java.dev.jna:jna-platform:4.2.1                (depends on 4.2.1)
[info] 	    +- net.openhft:affinity:3.1.11                        (depends on 4.2.1)
[info] 	* org.scala-lang.modules:scala-xml_2.11:1.0.5 is selected over 1.0.2
[info] 	    +- org.scala-lang:scala-compiler:2.11.12              (depends on 1.0.5)
[info] 	    +- org.scalatest:scalatest_2.11:2.2.1                 (depends on 1.0.2)
```

Dolu1990, 2018-12-24:
Yes, there is some cleaning to do, also i have to port other sub project, and there i don't know why SBT was hanging after compilation

closed 2020-02-05 by Dolu1990