SpinalHDL/SpinalHDL #174: RAM contents are included as 'assign's in generated Verilog even when inlineRom is set to false

tomverbeure, 2018-12-29:
The default configuration value of inlineRom is false. I don't change it my project.

However, when I create the verilog, it still creates wires and assignments for the the values of the ROM.

https://github.com/tomverbeure/panologic-g2/blob/5f6d0923da917684a3779ec1da253d33fcbdf8c6/spinal/Pano.v#L2035-L2051

and

https://github.com/tomverbeure/panologic-g2/blob/5f6d0923da917684a3779ec1da253d33fcbdf8c6/spinal/Pano.v#L4237-L4253

Those assignments aren't used in the rest of the Verilog file, so that's fine, but they are a nuisance that can make a small file very big.

Tom

Dolu1990, 2018-12-30:
This isn't related to the inlineRom option
It is probably because you have keep the referance of the List[Bits] used to initialise the rom inside your component (via a val)

 ex : 
```scala
val romValues = (0 to 255).map(i => B(i*i,16 bits)) //Will generate 128 useless hardware literals
val rom = Mem(Bits(16 bits), romValues)
```

There is multiple ways to avoid that : 

```scala
def romValues = (0 to 255).map(i => B(i*i,16 bits))
val rom = Mem(Bits(16 bits), romValues)
```

```scala
val rom = Mem(Bits(16 bits), (0 to 255).map(i => B(i*i,16 bits)) )
```

```scala
@dontName val romValues = (0 to 255).map(i => B(i*i,16 bits)) //Wil not keep the ramValues name, to the anonymus literal signals will be removed.
val rom = Mem(Bits(16 bits), romValues)
```

```scala
val romInit = (0 to 255).map(i => BigInt(i*i))
val rom = Mem(Bits(16 bits), 256) initBigInt(romInit)
```

Dolu1990, 2018-12-30:
The doc need to be more filled :)

tomverbeure, 2018-12-31:
Ok, that made it work!

Thanks!
Tom

closed 2018-12-31 by tomverbeure