SpinalHDL/SpinalHDL #176: InOutWrapper doesn't always work

tomverbeure, 2018-12-30:
I have an interface that is defined here:
https://github.com/tomverbeure/panologic-g2/blob/ulpi/spinal/src/main/scala/pano/Ulpi.scala

According to the ULPI specification, the PHY is the master and the controller (LINK) is the slave. So I have defined the signals that way.

Since I'm implementing a controller, I have defined the bundle as slave in the controller:
https://github.com/tomverbeure/panologic-g2/blob/e2f5c389d0b8558d5af04383c2016f1600b352e2/spinal/src/main/scala/pano/UlpiCtrl.scala#L17

Within the Ulpi bundle declaration, I have a TriStateArray `data`:
https://github.com/tomverbeure/panologic-g2/blob/e2f5c389d0b8558d5af04383c2016f1600b352e2/spinal/src/main/scala/pano/Ulpi.scala#L11

Since `data` is inout, I don't care whether it's 'master' or 'slave': it shouldn't matter, since both master and slave are supposed to access them. 

But it does matter: because when it ends up being a slave, then InOutWrapper won't convert it to an inout at the toplevel. As can be seen here:
https://github.com/SpinalHDL/SpinalHDL/blob/8c7ff12e96527df4a268daf0f0128eb1b2cc288f/lib/src/main/scala/spinal/lib/io/InOutWrapper.scala#L56

I don't agree with this decision (why should a slave interface not be converted to an inout port???), but that's a different issue.

So, I declare it 'slave', which means it becomes 'master' when I make `Ulpi` a slave.

After that, I can successfully create a UlpiCtrl stand-alone verilog file by running
```
sbt "run-main pano.UlpiCtrlVerilog"
```
So far so good.

Now I need to bring the `ulpi` interface all the way to the toplevel of Pano.scala. This is a super straight forward connection 2 levels up. `ulpi` is declared as `slave` all along (which means, again, `data` is master) The code elaborates fine after that.

```
sbt "run-main pano.PanoVerilog"
```
However, after elaboration, the resulting Pano.v file still has writeEnable/write/read ports instead of one inout port!

That's the bug. :-)

Note: if you want to repro this, use commit e2f5c389d0b8558d5af04383c2016f1600b352e2.

Tom

tomverbeure, 2018-12-30:
Note: with this commit, I can work around the issue:
https://github.com/tomverbeure/panologic-g2/commit/d83d3409c7bfa65a48ee124ec3b0ab7eb7eca1e1

Dolu1990, 2018-12-30:
Hi,

About the master/slave on tristate bundles, basicaly, the slave(tristate) should practicaly never be used. 

The idea of  master/slave is sometime "Who is in charge of the bus -> master" In the case of a tristate, the master is always the chip who contains the IO buffer, because he controle the write/writeEnable signals of the io buffer.

A slave tristate bundle could be use inside sub components of a toplevel to do maybe some wrapping and stuff, but only in a digital manner

so : 
```scala
case class Ulpi() extends Bundle with IMasterSlave
{
    val clk         = Bool
    val data        = TriStateArray(8)
    val direction   = Bool
    val stp         = Bool
    val nxt         = Bool
    val reset       = Bool

    // Master = Phy, since that's how it's defined in the spec.
    override def asMaster: Unit = {
        out(clk)
        master(data)
        out(direction)
        in(stp)
        out(nxt)
        in(reset)
    }
    override def asSlave: Unit = {
        super.asSlave(this)
        master(data) //Override the super.asSlave slave(data)
    }
}
```

does that solve the issue ?

tomverbeure, 2018-12-31:
You solution solved the issue.

That said: it still shouldn't have been necessary, since `data` *was* already a master.

But it's good enough for me. :-)

Thanks!
Tom

closed 2018-12-31 by tomverbeure