SpinalHDL/SpinalHDL #177: Global prefix for all but the top-level entity

cutephoton, 2018-12-31:
Is there a global prefix that renames all but the top level components/modules?

When using QSysify, certain modules are multiply defined resulting in errors like...
```
Error (10228): Verilog HDL error at VexRiscvAvalon.v(75): module "BufferCC" cannot be declared more than once
```

In addition, how does one apply SpinalConfig to the SpinalVerilog style method.... I looked at the Scala source but the syntax is a bit above my head in this case.

```
    val report = SpinalConfig(mode = Verilog, globalPrefix = "interrupt").generate(new AvalonInterruptCtrl()).printPruned()
```
I did a quick modification to add `globalPrefixExcludeToplevel` to the SpinalConfig and could generate. I think a better option may be to prefix based on the toplevel component name for all inner modules. If you get a couple self-contained verilog files, that you plan to include in a project it almost makes sense to always do some kind of prefixing. (I.e. a number of qsys slaves)

What's the best strategy here?

Dolu1990, 2019-01-04:
hmm, i see the issue. So basicaly, before the global prefix feature was added, they way i was solving this kind of issues was by importing each file generated from spinalhdl into a different VHDL/Verilog library. That's a safe way of sandbox the thing.

The privateNamespace  is a good idea. I will implement it :)

About VHDL, that's the same storry ^^

Thanks for that feedback :)

Dolu1990, 2019-01-04:
Added in 438ea37dbc011e37d1d9cf57223177812e8009df, there is an example of usage there : 
https://github.com/SpinalHDL/SpinalHDL/blob/438ea37dbc011e37d1d9cf57223177812e8009df/tester/src/main/scala/spinal/tester/code/Play2.scala#L1960

Let's me know if you see some pitfalls ^^

cutephoton, 2019-01-05:
:+1: 
Just briefly reviewing, I believe it is as discussed.

I'll test it when I get a chance to merge recent changes.

closed 2019-01-11 by Dolu1990