SpinalHDL/SpinalHDL #178: have ascii string to make the state machine more easy to debug

linuxbest, 2019-01-01:
https://www.veripool.org/wiki/verilog-mode/verilog-mode_veritedium#SEC23

tomverbeure, 2019-01-01:
This is a duplicated of issue #143.

But, yes, I very much would like this feature.

Tom

Dolu1990, 2019-01-01:
Is there also some fancy pragma to point out that a signal should be displayed as a string in the wave ?

linuxbest, 2019-01-02:
Just select data format to ASCII in gtkwave, Here is a snapshot with gtkwave.

[image]

tomverbeure, 2019-01-02:
Fancy pragmas are never portable between tools. ASCII works with all waveform viewers and with $display.

It’s really the best solution, even if it feels like a hack (which it is!)

Tom

wifasoi, 2019-01-02:
Well, gtkwave can import a file to decode/transalte a value to ascii:
[Alias Files and Attaching External Disassemblers](www.ic.unicamp.br/~ducatte/mc542/Docs/gtkwave.pdf#page=53)

But i think this is specific to gtkwave. I think for the time being is a feasible workaround.

Dolu1990, 2019-01-02:
So, the univerasl solution for the verilog backend would be to use "large" vector of bits that contain the asci string ?

tomverbeure, 2019-01-02:
Yup, that’s guaranteed to work under all circumstances irrespective of which too you use.

You don’t have to be very selective about the size of the vector: just make it 64 characters or something. And surround all the code with “ifdef SYNTHESIS”, as mentioned in my issue #143.

Tom

Tom

Dolu1990, 2019-01-12:
In dev :)

commit a988f898 (2019-01-11): For each enum signal in the generated verilog there is now a string signal to get a waveform friendly state value.
Fix #178 #143
closed 2019-01-12 by Dolu1990