SpinalHDL/SpinalHDL #180: add splitAt method for Bits, UInt, SInt and Vec

typingArtist, 2019-01-03:
From a discussion on Gitter in October:

how about a ```splitAt(pos : UInt)``` method for ```Bits```, ```UInt```, ```SInt``` and ```Vec``` that combines a ```resizeLeft``` and ```resize``` such that the resulting tuple builds a ```(left, right)``` result, sliced at the specified pos position. Example:

```scala
val s = U"110101"
val (l,r) = s.splitAt(2) // l === U"1101", r === U"01"
```

Dolu1990, 2019-01-03:
Yes, it could be quite usefull, +1 :)

typingArtist, 2019-01-03:
@Dolu1990 is this the right approach? Do we need a copy as ```splitAt(bitId: UInt)``` as well?

Although it’s now as described, it actually doesn’t match the standard Scala collection’s ```splitAt``` function in what it returns first and what last. But if we make it matching, we get into even bigger mess, creating big surprise at the hardware design engineer, don’t you think?

Can you guide me how to and where to put tests for that?

numero-744, 2022-06-04:
Hi, could you open a pull request for https://github.com/typingArtist/SpinalHDL/commit/9234ca65bb22d42554a0194a5d59a05b3c670681 please?

Dolu1990, 2022-06-06:
Seems like it was added to SpinalHDL upstream already :

```scala
  /**
    * Split at n st bits
    * @example {{{ val res = data10bits.splitAt(4) }}}
    * @return (data10bits(8 downto 4), data10bits(3 downto 0))
    */
  def splitAt(n: Int): (Bits,Bits) = {
    require(n>=0)
    if(n==0){
      (toBitsInstance(this), Bits(0 bits))
    } else {
      (toBitsInstance(this(this.high downto n)), toBitsInstance(this(n - 1 downto 0)))
    }
  }
  ```

closed 2022-06-06 by Dolu1990